gnss-sdr/gnss-sdr #232: GPS PSEUDORANGE PROBLEM

matteocutugno, 2018-12-18:
Hello,
i work with Nuand bladerf x40 and GNSS-SDR 0.0.9 and i'm experiencing a particular issue.
The RINEX observation file has strange values of Pseudorange.
To investigate this i get PVT also with my smartphone running rinex ON (an application on android usefull to compute position and save RINEX files).  
If i compare the RINEX observation file obtained with my smartphone and the RINEX observation file obtained with Nuand (same place and same instant of time) i get Pseudoranges that differs of 3000 meters. In particular the Pseudorange obtained with smartphone is  about 21000000 meters, the one obtained with GNSS-SDR is about 18000000 meters. How is this possible? Is there an explanation for this difference?
Below i post the parts of my two RINEX observation files:

-------------------------------------------SMARTPHONE RINEX----------------------------------------------------------
     3.03           OBSERVATION DATA    M: Mixed            RINEX VERSION / TYPE
rinex ON            NSL                 20181217 132624 UTC PGM / RUN BY / DATE
                                                            MARKER NAME
                                                            MARKER TYPE
rinex ON User                                               OBSERVER / AGENCY
XXXXXXXX            HUAWEI              WAS-LX1A            REC # / TYPE / VERS
XXXXXXXX            WAS-LX1A                                ANT # / TYPE
        0.0000        0.0000        0.0000                  APPROX POSITION XYZ
        0.0000        0.0000        0.0000                  ANTENNA: DELTA H/E/N
G    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES
R    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES
E    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES
  2018    12    17    12    34   14.0003625     GPS         TIME OF FIRST OBS
                                                            END OF HEADER
> 2018 12 17 12 39 18.0003625  0  4
G09  21360114.301           0.000         748.990          36.000
G13  24941238.640           0.000       -4038.639          25.000
R143232383296671.654           0.000       -4848.217          32.000
G05  22527513.627           0.000       -4033.625          30.000
> 2018 12 17 12 39 19.0003625  0  4
G09  21360256.402           0.000         748.714          36.000
G13  24940470.272           0.000       -4037.550          23.000
R143232383295761.784           0.000       -4849.416          32.000
G05  22526748.257           0.000       -4033.095          30.000
> 2018 12 17 12 39 20.0003625  0  3
G09  21360398.804           0.000         748.962          35.000
R143232383294851.914           0.000       -4849.938          32.000
G05  22525979.889           0.000       -4033.597          29.000
> 2018 12 17 12 39 21.0003625  0  5
G09  21360541.805           0.000         748.714          36.000
G13  24938931.737           0.000       -4037.573          22.000
R143232383293942.942           0.000       -4846.757          31.000
G05  22525211.820           0.000       -4033.297          29.000

-------------------------------------------GNSS-SDR RINEX---------------------------------------------------------------
     3.02           OBSERVATION DATA    G                   RINEX VERSION / TYPE
G = GPS  R = GLONASS  E = GALILEO  S = GEO  M = MIXED       COMMENT             
GNSS-SDR            matteo              20181217 133900 UTC PGM / RUN BY / DATE 
GPS OBSERVATION DATA FILE GENERATED BY GNSS-SDR             COMMENT             
GNSS-SDR VERSION 0.0.9.git-next-f3e32e30                    COMMENT             
See https://gnss-sdr.org                                    COMMENT             
DEFAULT MARKER NAME                                         MARKER NAME         
matteo              CTTC                                    OBSERVER / AGENCY   
GNSS-SDR            Software Receiver   0.0.9.git           REC # / TYPE / VERS 
Antenna number      Antenna type                            ANT # / TYPE        
        0.0000        0.0000        0.0000                  APPROX POSITION XYZ 
        0.0000        0.0000        0.0000                  ANTENNA: DELTA H/E/N
G    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES 
DBHZ                                                        SIGNAL STRENGTH UNIT
  2018    12    17    12    39   18.0600000     GPS         TIME OF FIRST OBS   
                                                            END OF HEADER       
> 2018 12 17 12 39 18.0600000  0  4                                             
G30  18094651.411 6   -149325.266 6      1468.122 6        38.511               
G07  17904803.888 5     70315.979 5      -766.581 5        35.606               
G09  18734083.841 6    204060.014 6     -2056.220 6        38.456               
G05  19901410.793 6   -246497.297 6      2701.097 6        36.142        
> 2018 12 17 12 39 19.0000000  0  4                                             
G30  18094389.357 6   -150716.304 6      1480.019 6        36.552               
G07  17904939.974 6     71002.011 6      -705.821 6        38.998               
G09  18734454.758 6    206001.621 6     -2072.113 6        40.798               
G05  19900934.910 6   -249051.723 6      2726.999 6        36.905        
> 2018 12 17 12 39 20.0000000  0  4                                             
G30  18094099.043 6   -152199.276 6      1481.440 6        36.979               
G07  17905075.502 6     71728.630 6      -733.603 6        38.476               
G09  18734851.387 7    208063.661 7     -2070.583 7        42.081               
G05  19900402.390 5   -251772.164 5      2765.955 5        35.054 
> 2018 12 17 12 39 21.0000000  0  4                                             
G30  18093825.920 6   -153681.001 6      1481.563 6        37.096               
G07  17905217.650 6     72456.545 6      -729.253 6        37.942               
G09  18735232.118 6    210126.629 6     -2055.842 6        38.885               
G05  19899900.423 6   -254491.482 6      2729.723 6        36.472   
----------------------------------------------------------------------------------------------------------------------------------

As you can see both receiver tracks the Gps satellite G30 and the pseudoranges measures, as i previously said, differs a lot from each other.
Thanks in advance for any help

odrisci, 2019-02-19:
I know this is a bit late, but this is perfectly normal. The pseudorange measurement is just that: "pseudo" range, not true range. The pseudorange consists of the true distance to the satellite plus the receiver clock bias (ignoring other errors). In the example you give, the difference in clock bias is about 3 km (or 10 microseconds), which is not unusual. Typically low cost receivers will try to keep their clock bias to less than a few milliseconds, while some high end receivers can steer the clock to keep it within a few 10s of nanoseconds of GPS time.

shuwang1, 2019-07-06:
Have you be able to make a fix successfully from the observables which you obtained from Nuand bladerf x40?     You can download ephemeris data from CCDIS from NASA

closed 2020-09-18 by matteocutugno

jt3232, 2021-09-08:
Am I missing something here? He said his pseudoranges were 21000000 meters and 18000000 meters. That's not a difference of 3000m, but 3000km! That's not a 10 microsecond difference, but a 10,000 microsecond difference (0.01s). 

I bring this up because I am running multiple HRF1s on the same antenna and they all output different pseudoranges, by a large amount. This is extremely unsettling to me. They each have the NooElec TCXO (10MHz at 0.5ppm). However, I get differences of 189km between two and 380km on another. I am not sure, but this seems like out of the realm of what should be acceptable. They each have the same configuration and are running on 3 identical computers.

Thanks,

Jaxon