gnss-sdr/gnss-sdr #233: Loss of lock in channel while using bladeRF for real time acquisiton

BushraAnsarii, 2018-12-25:
This is the issue that I am facing continuously: Can someone suggest some solution to this. Thanks in advance

Initializing GNSS-SDR v0.0.10 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: bladerf=0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfsp                                                                                                             ace airspy redpitaya
Opening nuand bladeRF with device identifier string: "*:instance=0"
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src                                                                                                             /version_compat.c":96] Firmware version (v2.3.1) is newer than entries in libbla                                                                                                             deRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src                                                                                                             /version_compat.c":127] FPGA version (v0.9.0) is newer than entries in libbladeR                                                                                                             F's compatibility table. Please update libbladeRF if problems arise.
 Serial # 8ea3...8318
 FW v2.3.1 FPGA v0.9.0
Actual RX Rate: 3000000.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...
Actual LNA Gain: 6.000000 dB...
Actual VGA1 Gain: 30.000000 dB...
Actual VGA2 Gain: 18.000000 dB...
Actual Bandwidth: 28000000.000000 [Hz]...
Using Volk machine: avx2_64_mmx_orc
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)                                                                                                             
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)                                                                                                             
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)                                                                                                             
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)                                                                                                             
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)                                                                                                            
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block IIR)                                                                                                             
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)                                                                                                            
Current receiver time: 2 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)                                                                                                             
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)                                                                                                             
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)                                                                                                             
Current receiver time: 3 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Current receiver time: 4 s
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Current receiver time: 5 s
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)                                                                                                             
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Current receiver time: 6 s
Current receiver time: 7 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)                                                                                                             
Current receiver time: 8 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)                                                                                                             
Current receiver time: 9 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)                                                                                                             
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)                                                                                                             
Current receiver time: 10 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 11 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 12 s
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Loss of lock in channel 2!
Current receiver time: 13 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 14 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 15 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 16 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 17 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 18 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 19 s
Current receiver time: 20 s
Current receiver time: 21 s
Current receiver time: 22 s
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
Current receiver time: 27 s
Current receiver time: 28 s

matteocutugno, 2019-01-08:
Hi, 
so you can't receive any navigation message?
Can you post your configuration file?
MC

BushraAnsarii, 2019-01-09:
No, I am not getting any navigation message.
this is the configuration file that I am using.

; This is a GNSS-SDR configuration file
; The configuration API is described at https://gnss-sdr.org/docs/sp-blocks/

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
;# RF Gain: LNA Gain {0, 3, 6}
SignalSource.gain=6
;# IF Gain: N/A
SignalSource.rf_gain=40
;# BB Gain: RX VGA1 + VGA2 [5, 60]
SignalSource.if_gain=48
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.osmosdr_args=bladerf=0  ; This line enables the bladeRF
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition_Fine_Doppler
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.threshold=0.015
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=500
Acquisition_1C.max_dwells=15
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=2.0;
Tracking_1C.order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.dump=false
Tracking_1C.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
;PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

matteocutugno, 2019-01-09:
Hi
Your configuration file seems good. 
The only thing that I don't understand is why  you commented the line
;PVT.implementation=RTKLIB_PVT
Uncomment it to get fixed position once you will be able to receive a complete nav message from 4 different satellite.
Maybe your problem is an hardware issue. 
What kind of antenna are you using(active antenna, noise figure.... ) ?  Are you feeding the antenna with an external bias tee providing 5 Volt?
I don't know what model of bladeRf are you using. 
I have bladeRF x40 and it doesn't have an integrated bias tee so you need an external one.
MC

BushraAnsarii, 2019-01-09:
That was written wrong here by mistake. I have not commented that line in my config file.
I am using bladeRF X115 and I m not feeding the antenna with any external 5v as it doesn't need that.
I have installed GNSS-SDR from source, as given in this link: https://gnss-sdr.org/build-and-install/
And I m using an active antenna

matteocutugno, 2019-01-09:
I knew that the only difference between x40 e x115 is the fpga size. Maybe is my fault but I was able to acquire gnss signal once I get an external bias tee to feed the antenna. Are you sure that x115 doesn't need it?

oscarmezyk, 2019-01-09:
"Active antenna" needs power supply (it has LNA inside). If you have one single wire (SMA connector) to the antenna then you need bias-t in-line with that wire or a power source inside the sdr. The rtl-sdr v3 has it already inside for example. If you have active antenna hooked directly to bladeRF - it will not work. 

With my bladeRF I managed to get PVT - see discussion with matteocutugno here: https://github.com/gnss-sdr/gnss-sdr/issues/226

BushraAnsarii, 2019-01-24:
I am using bias tee now, but still, there is the loss of lock. 
Bias tee is-- ZFBT-4R2GW+  by mini circuits.

BushraAnsarii, 2019-01-24:
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 03 (Block IIF)
Current receiver time: 42 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 03 (Block IIF)
Current receiver time: 43 s
Current receiver time: 44 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 03 (Block IIF)
Current receiver time: 45 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 03 (Block IIF)
Current receiver time: 46 s
Current receiver time: 47 s
Current receiver time: 48 s
Current receiver time: 49 s
Current receiver time: 50 s
Current receiver time: 51 s
Current receiver time: 52 s
Current receiver time: 53 s
Current receiver time: 54 s
Current receiver time: 55 s
Current receiver time: 56 s
Current receiver time: 57 s
Current receiver time: 58 s
Current receiver time: 59 s
.......
it just keep on showing this only

oscarmezyk, 2019-01-27:
1) Take your antenna outside - clear view of the sky
2) check if gains are set correctly - 
I would try to do:
;#SignalSource.gain=6
;#SignalSource.rf_gain=40
;#SignalSource.if_gain=48
;#SignalSource.AGC_enabled=false

before starting gnss-sdr go to balderf-cli -i
set frequency 1575.42e6
set agc off
set gain rx 48
cal lms
cal dc rx
quit

3) just for sake of experiment modify some lines:
GNSS-SDR.internal_fs_sps=4000000
InputFilter.implementation=Pass_Through
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
;#Acquisition_1C.max_dwells=15
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
;#Tracking_1C.pll_bw_hz=40.0;
;#Tracking_1C.dll_bw_hz=2.0;
PVT.output_rate_ms=500
PVT.display_rate_ms=1000

4) paste here output of the gnss-sdr during startup

b.r.

BushraAnsarii, 2019-01-28:
Initializing GNSS-SDR v0.0.9.git-next-70be265 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: bladerf=0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Opening nuand bladeRF with device identifier string: "*:instance=0"
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/version_compat.c":96] Firmware version (v2.3.1) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/version_compat.c":127] FPGA version (v0.10.2) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
 Serial # 8ea3...8318
 FW v2.3.1 FPGA v0.10.2
Actual RX Rate: 4000000.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...
Actual LNA Gain: 6.000000 dB...
Actual VGA1 Gain: 30.000000 dB...
Actual VGA2 Gain: 18.000000 dB...
Actual Bandwidth: 28000000.000000 [Hz]...
Using Volk machine: avx2_64_mmx_orc
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 s
Loss of lock in channel 0!
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Current receiver time: 3 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 4 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Current receiver time: 5 s
Current receiver time: 6 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 7 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 8 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 9 s
Current receiver time: 10 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 11 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 12 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 13 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Current receiver time: 14 s
Current receiver time: 15 s
Current receiver time: 16 s
Current receiver time: 17 s
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
Current receiver time: 21 s
Current receiver time: 22 s
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
Current receiver time: 27 s
Current receiver time: 28 s
Current receiver time: 29 s
Current receiver time: 30 s
Current receiver time: 31 s
Current receiver time: 32 s
Current receiver time: 33 s
Current receiver time: 34 s
Current receiver time: 35 s
Current receiver time: 36 s
Current receiver time: 37 s
Current receiver time: 38 s
Current receiver time: 39 s
Current receiver time: 40 s
Current receiver time: 41 s
Current receiver time: 42 s
Current receiver time: 43 s
Current receiver time: 44 s
Current receiver time: 45 s
Current receiver time: 46 s
Current receiver time: 47 s
Current receiver time: 48 s
Current receiver time: 49 s
Current receiver time: 50 s
Current receiver time: 51 s
Current receiver time: 52 s
Current receiver time: 53 s
Current receiver time: 54 s

This is the message that I am getting.After making the changes that you suggested.

oscarmezyk, 2019-01-28:
Well... tough mtfckr.  There are some other things I may suggest:
1)  the bandwidth is set to 28 000 000 -> kinda weird; should be rather 2 800 000 or 3 000 000
start gnss-sdr 
stop gnss-sdr
go to bladerf-cli -i
print -> check if frequency / gains / bandwidth / agc are set correctly as you'd expect;
if not - set them manually and run gnss-sdr again

2) show us the antenna setup: what is the antenna, what is its gain, where is it placed, how long is the cable? 
In order to work "inside" the room I had to put additional LNA inline (+10dB). When placed antenna outside it works without it

3) did you do calibrate your sdr? ( cal table dc rx )
4) pay attention to [INFO] saying you're running not compatibile firmware&libbladerf - you'd need to upate bladerf to branch master 0.10.x (I guess)

If all that doesn't work - I am out of ideas.