gnss-sdr/gnss-sdr #234: No PVT while tracking 4 or more different satellties

valcesch, 2018-12-30:
Hello,

Thank you for your work.

I am currently using the rtl-sdr V3 with gnss-sdr 0.0.10 and acquisition/tracking is working fine. I can successfully get a fix after a few seconds. Nevertheless, the PVT solution stops being generated after a few minutes, even if more than 4 satellites are tracked. No PVT will be generated even after 1 hour of operation.

At the end, I am not really interested in the PVT solution, but I need more to generate a RINEX file with all satellites in view (even if in bad tracking conditions).

Here is my configuration file:

	[GNSS-SDR]

	;######### GLOBAL OPTIONS ##################
	GNSS-SDR.internal_fs_sps=2000000

	;######### CONTROL_THREAD CONFIG ############
	ControlThread.wait_for_flowgraph=false

	;######### INITIAL RECEIVER POSITIION ######
	GNSS-SDR.init_latitude_deg=XX
	GNSS-SDR.init_longitude_deg=XX
	GNSS-SDR.init_altitude_m=XX

	;######### SUPL RRLP GPS assistance configuration #####
	GNSS-SDR.SUPL_gps_enabled=true
	GNSS-SDR.SUPL_read_gps_assistance_xml=false
	GNSS-SDR.SUPL_gps_ephemeris_server=supl.google.com
	GNSS-SDR.SUPL_gps_ephemeris_port=7275
	GNSS-SDR.SUPL_gps_acquisition_server=supl.google.com
	GNSS-SDR.SUPL_gps_acquisition_port=7275
	GNSS-SDR.SUPL_MCC=217
	GNSS-SDR.SUPL_MNC=7
	GNSS-SDR.SUPL_LAC=861
	GNSS-SDR.SUPL_CI=40184

	;######### SIGNAL_SOURCE CONFIG ############
	SignalSource.implementation=RtlTcp_Signal_Source
	SignalSource.item_type=gr_complex
	SignalSource.sampling_frequency=2000000
	SignalSource.freq=1575420000
	SignalSource.gain=60
	SignalSource.rf_gain=40
	SignalSource.if_gain=30
	SignalSource.AGC_enabled=true
	SignalSource.samples=0
	SignalSource.repeat=false
	SignalSource.dump=false
	SignalSource.dump_filename=signal_source.dat
	SignalSource.enable_throttle_control=false
	SignalSource.address=127.0.0.1 ;
	SignalSource.port=1234 
	SignalSource.swap_iq=false

	;######### INPUT_FILTER CONFIG ############
	;InputFilter.implementation=Pass_Through
	;InputFilter.input_item_type=gr_complex
	;InputFilter.output_item_type=gr_complex

	InputFilter.implementation=Freq_Xlating_Fir_Filter
	InputFilter.input_item_type=gr_complex
	InputFilter.output_item_type=gr_complex
	InputFilter.taps_item_type=float
	InputFilter.number_of_taps=5
	InputFilter.number_of_bands=2
	InputFilter.band1_begin=0.0
	InputFilter.band1_end=0.45
	InputFilter.band2_begin=0.55
	InputFilter.band2_end=1.0
	InputFilter.ampl1_begin=1.0
	InputFilter.ampl1_end=1.0
	InputFilter.ampl2_begin=0.0
	InputFilter.ampl2_end=0.0
	InputFilter.band1_error=1.0
	InputFilter.band2_error=1.0
	InputFilter.filter_type=bandpass
	InputFilter.grid_density=16
	InputFilter.sampling_frequency=2000010.02 
	InputFilter.IF=-7894.97  ; 

	;######### RESAMPLER CONFIG ############
	Resampler.implementation=Pass_Through
	Resampler.dump=false
	Resampler.item_type=gr_complex

	;######### CHANNELS GLOBAL CONFIG ############
	Channels_1C.count=8
	Channels.in_acquisition=1;
	Channel.signal=1C

	;######### ACQUISITION GLOBAL CONFIG ############
	Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
	Acquisition_1C.item_type=gr_complex
	Acquisition_1C.coherent_integration_time_ms=4
	Acquisition_1C.threshold=0.001
	;Acquisition_1C.pfa=0.01
	Acquisition_1C.doppler_max=10000
	Acquisition_1C.doppler_min=-10000
	Acquisition_1C.doppler_step=250

	;######### TRACKING GLOBAL CONFIG ############
	Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
	Tracking_1C.item_type=gr_complex
	Tracking_1C.dump=true
	Tracking_1C.dump_filename=epl_tracking_ch_
	Tracking_1C.pll_bw_hz=40.0;
	Tracking_1C.dll_bw_hz=4.0;
	Tracking_1C.order=3;
	Tracking_1C.max_lock_fail=5;

	;######### TELEMETRY DECODER GPS CONFIG ############
	TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

	;######### OBSERVABLES CONFIG ############
	Observables.implementation=Hybrid_Observables
	Observables.dump=false

	;######### PVT CONFIG ############
	PVT.implementation=RTKLIB_PVT;
	PVT.positioning_mode=Static;
	PVT.iono_model=OFF;
	PVT.trop_model=OFF;
	PVT.elevation_mask=0;
	PVT.threshold_GDOP=1000;

	PVT.output_rate_ms=100
	PVT.display_rate_ms=1000
	PVT.rinex_version=3

Here is the log:

	New GPS NAV message received in channel 2: subframe 2 from satellite GPS PRN 30 (Block IIF)
	New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 05 (Block IIR-M)
	New GPS NAV message received in channel 4: subframe 2 from satellite GPS PRN 13 (Block IIR)
	New GPS NAV message received in channel 5: subframe 2 from satellite GPS PRN 02 (Block IIR)
	Current receiver time: 1 min 21 s
	Position at 2018-Dec-30 12:25:25.000000 UTC using 4 observations is Lat = XX [deg], Long = XX [deg], Height = XX [m]
	Current receiver time: 1 min 22 s
	Position at 2018-Dec-30 12:25:25.000000 UTC using 4 observations is Lat = XX [deg], Long = XX [deg], Height = XX [m]
	Current receiver time: 1 min 23 s
	Position at 2018-Dec-30 12:25:25.000000 UTC using 4 observations is Lat = XX [deg], Long = XX [deg], Height = XX [m]
	Current receiver time: 1 min 24 s
	Position at 2018-Dec-30 12:25:25.000000 UTC using 4 observations is Lat = XX [deg], Long = XX [deg], Height = XX [m]
	Current receiver time: 1 min 25 s
	Position at 2018-Dec-30 12:25:25.000000 UTC using 4 observations is Lat = XX [deg], Long = XX [deg], Height = XX [m]
	Current receiver time: 1 min 26 s
	Position at 2018-Dec-30 12:25:25.000000 UTC using 4 observations is Lat = XX [deg], Long = XX [deg], Height = XX [m]
	New GPS NAV message received in channel 7: subframe 3 from satellite GPS PRN 05 (Block IIR-M)
	New GPS NAV message received in channel 2: subframe 3 from satellite GPS PRN 30 (Block IIF)
	New GPS NAV message received in channel 4: subframe 3 from satellite GPS PRN 13 (Block IIR)
	New GPS NAV message received in channel 5: subframe 3 from satellite GPS PRN 02 (Block IIR)
	Current receiver time: 1 min 27 s
	Current receiver time: 1 min 28 s
	Current receiver time: 1 min 29 s
	Current receiver time: 1 min 30 s
	Current receiver time: 1 min 31 s
	Current receiver time: 1 min 32 s
	New GPS NAV message received in channel 2: subframe 4 from satellite GPS PRN 30 (Block IIF)
	New GPS NAV message received in channel 7: subframe 4 from satellite GPS PRN 05 (Block IIR-M)
	New GPS NAV message received in channel 4: subframe 4 from satellite GPS PRN 13 (Block IIR)
	New GPS NAV message received in channel 5: subframe 4 from satellite GPS PRN 02 (Block IIR)
	Current receiver time: 1 min 33 s
	Current receiver time: 1 min 34 s
	Current receiver time: 1 min 35 s
	Current receiver time: 1 min 36 s
	Current receiver time: 1 min 37 s
	Current receiver time: 1 min 38 s
	New GPS NAV message received in channel 2: subframe 5 from satellite GPS PRN 30 (Block IIF)
	New GPS NAV message received in channel 7: subframe 5 from satellite GPS PRN 05 (Block IIR-M)
	New GPS NAV message received in channel 5: subframe 5 from satellite GPS PRN 02 (Block IIR)
	New GPS NAV message received in channel 4: subframe 5 from satellite GPS PRN 13 (Block IIR)
	Current receiver time: 1 min 39 s
	Current receiver time: 1 min 40 s
	Current receiver time: 1 min 41 s
	Current receiver time: 1 min 42 s
	Current receiver time: 1 min 43 s
	Current receiver time: 1 min 44 s
	New GPS NAV message received in channel 5: subframe 1 from satellite GPS PRN 02 (Block IIR)
	New GPS NAV message received in channel 7: subframe 1 from satellite GPS PRN 05 (Block IIR-M)
	New GPS NAV message received in channel 4: subframe 1 from satellite GPS PRN 13 (Block IIR)
	New GPS NAV message received in channel 2: subframe 1 from satellite GPS PRN 30 (Block IIF)
	Current receiver time: 1 min 45 s
	Current receiver time: 1 min 46 s
	Current receiver time: 1 min 47 s
	Current receiver time: 1 min 48 s
	Current receiver time: 1 min 49 s
	Current receiver time: 1 min 50 s
	New GPS NAV message received in channel 2: subframe 2 from satellite GPS PRN 30 (Block IIF)
	New GPS NAV message received in channel 5: subframe 2 from satellite GPS PRN 02 (Block IIR)
	New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 05 (Block IIR-M)
	New GPS NAV message received in channel 4: subframe 2 from satellite GPS PRN 13 (Block IIR)

valcesch, 2019-01-19:
Hi, 

Thank you for your answer. 

Pseudorange measurements are present and look correct. During my last run, no PVT solution was generated while 7 satellites were tracked (no more PVT solution after around 1 hour, but navigation messages are still decoded).

One thing that I also observed is that once the first set of satellites are not visible anymore, gnss-sdr does not try to look for new ones. In this run, it started tracking 7 satellites, and  finished with only 3 channels effectively tracking the same satellites after 2 hours. It seems that gnsss-sdr assigns PRN to channels at the beginning and that the configuration is not changed afterward. 

Thank you for your help

matteocutugno, 2019-01-19:
Hi
It's quite strange that with 7 satellites locked you wasn't able to compute PVT. 
Try to add in the acquisition block:

Acquisition.repeat_satellite=true

MC

noelpedro, 2021-05-20:
This get solved.  Having similar issue on v 13....

jnunezortuno, 2023-10-16:
Has it been solved yet? Having similar issue on v0.0.16

jnunezortuno, 2023-10-23:
> Has it been solved yet? Having similar issue on v0.0.16

Also on 0.0.18

ZiglioUK, 2023-10-25:
A while ago I had looked into it. It was to do, if I remember correctly, with errors from rtk-lib not being handled correctly. That's where I'd be looking. Sorry, I should have submitted a patch request.

ZiglioUK, 2023-10-25:
Yes, for navigation. If you guys have a snippet of radio signal showing that problem I could take another look

jnunezortuno, 2023-10-25:
Thanks for your offer, Emanuele.

I think that GNSS-SDR is somehow not getting into the PVT (navigation) phase as no PVT internal binary data file is being logged, but anyway, here is a link to download a fake GPS signal of 5 satellites emulating the (0°,0°,0) position. It's working with an Ublox receiver but doesn't works with GNSS-SDR v0.0.18.

[https://www.dropbox.com/scl/fi/olvwdtxpzplpyjszsvcnp/3_4_8_14_21.bin?rlkey=6s9xdqrvnw8g6l510fwkijrdv&dl=0](url)

Thanks in advance.

ZiglioUK, 2023-10-25:
Downloading now @jnunezortuno , do you have a conf file I can use?

jnunezortuno, 2023-10-25:
> Downloading now @jnunezortuno , do you have a conf file I can use?

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2500000

; enabled [3 8 14 21]
;GNSS-SDR.GPS_banned_prns=1,2,4,5,6,7,9,10,11,12,13,15,16,17,18,19,20,22,23,24,25,26,27,28,29,30,31,32

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/3_4_8_14_21.bin
SignalSource.item_type=ishort
SignalSource.sampling_frequency=2500000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=2500000
Resampler.sample_freq_out=2500000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=5 ; Number of channels targeting GPS L1 C/A signals
Channels.in_acquisition=1 ;number of channels performing acquisition on different satellite signals at the same time
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250
Acquisition_1C.blocking=true

Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump
Acquisition_1C.dump_channel=0;
; G_1C_ch_N_K_sat_P.mat 
;  N is the channel number defined by dump_channel
;  K is the dump number
;  P is the targeted satellite’s PRN number

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

Tracking_1C.dump=true
Tracking_1C.dump_filename=./trk_dump
;trk_dump_chX.mat where X is the channel number

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

PVT.output_enabled=true
PVT.rtcm_output_file_enabled=false
PVT.gpx_output_enabled=true
PVT.geojson_output_enabled=true
PVT.kml_output_enabled=true
PVT.xml_output_enabled=true
PVT.rinex_output_enabled=true
PVT.nmea_output_file_enabled=false

PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1234
PVT.enable_protobuf=false

PVT.dump=true
PVT.dump_filename=./PVT
PVT.threshold_reject_GDOP=10000
PVT.threshold_reject_innovation=10000
PVT.elevation_mask=0

ZiglioUK, 2023-10-25:
@jnunezortuno thank you for the conf.
With those acquisition settings I see 5 GPS sats. Only 3 of them are healthy so no PVT solution is computed.
GNSS-SDR starts produces Rinex files only after a first PVT solution is solved.
Here I've added some debug info to src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc:
```
Ch 0 PRN 3 Valid
Ch 1 PRN 4 Health = 0
Ch 2 PRN 8 Valid
Ch 3 PRN 14 Health = 0
Ch 4 PRN 21 Valid

```

So basically we need more satellites. It may help adjusting the acquisition settings or adding Galileo

jnunezortuno, 2023-10-26:
Thanks for debugging, Emanuele.

I don't see any unhealthy SV warning in the Log file. I have tested also with PVT.use_unhealthy_sats=true but it doesn't works

Can you tell me where did you "touch" the _rtklib_pvt_gs.cc_ code to obtain this debug info?

ZiglioUK, 2023-10-26:
> I don't see any unhealthy SV warning in the Log file. I have tested also with PVT.use_unhealthy_sats=true but it doesn't works

what does uBlox say about those sats? I remember PRN 4 being unhealthy for years but I thought it had been replaced by now. Other sats can be set unhealthy from time to time

> Can you tell me where did you "touch" the _rtklib_pvt_gs.cc_ code to obtain this debug info?

See: 
https://github.com/ZiglioUK/gnss-sdr/tree/next.234

https://github.com/ZiglioUK/gnss-sdr/commit/ac21b6e5918eae4bcbbd07fa703054579e7bfa72

ZiglioUK, 2023-10-30:
@jnunezortuno 

I'm having a closer look. Going through one satellite at the time. The CN0 is very strong. Navigation messages are being decoded. For some reason PRN4 for example is failing this test src/core/system_parameters/gps_navigation_message.cc:

```
bool Gps_Navigation_Message::satellite_validation()
{
    bool flag_data_valid = false;
    b_valid_ephemeris_set_flag = false;

    // First Step:
    // check Issue Of Ephemeris Data (IODE IODC..) to find a possible interrupted reception
    // and check if the data have been filled (!=0)
    if (d_TOW_SF1 != 0.0 and d_TOW_SF2 != 0.0 and d_TOW_SF3 != 0.0)
        {
            if (d_IODE_SF2 == d_IODE_SF3 and d_IODC == d_IODE_SF2 and d_IODC != -1.0)
                {
                    flag_data_valid = true;
                    b_valid_ephemeris_set_flag = true;
                }
        }
    return flag_data_valid;
}

```
I'll carry on investigating...

jnunezortuno, 2023-10-30:
@ZiglioUK 

Thanks for your code. CN0 is strong because we are generating GPS signal with very low noise (or even no noise, I don't remember). 
The signal was generated using ephemeris file BRDC2630.23N (20-SEP-23 13:26). In this file, SV_HEALTH is OK for all SVs. However, _tmp_eph_iter_gps->second.SV_health_ brings up a not healthy SV.

I have no problem at all with my uBlox EVK-8MT

[image]

ZiglioUK, 2023-10-30:
@jnunezortuno basilcally what I'm seeing is that gnss-sdr takes longer than the uBlox to track and decode messages so it starts with subframe 2, 3, 4... (for PRN4, that's what I'm looking at). For a satellite ephemeris to be declared valid it needs to decode d_TOW_SF1, 2, 3...
Now I'm tricking it into attempting a PVT solution just with d_TOW_SF2 and 3. 
I think I'm getting there but without a reference location it thinks satellites are below the horizon.

We should be able to speed up tracking convergence by for example avoid or reducing the FLL pullin.
A longer data sample would eventually compute a PVT and start emitting Rinexes.

jnunezortuno, 2023-10-30:
> @jnunezortuno basically what I'm seeing is that gnss-sdr takes longer than the uBlox to track and decode messages so it starts with subframe 2, 3, 4... (for PRN4, that's what I'm looking at). For a satellite ephemeris to be declared valid it needs to decode d_TOW_SF1, 2, 3... Now I'm tricking it into attempting a PVT solution just with d_TOW_SF2 and 3. I think I'm getting there but without a reference location it thinks satellites are below the horizon.
> 
> We should be able to speed up tracking convergence by for example avoid or reducing the FLL pullin. A longer data sample would eventually compute a PVT and start emitting Rinexes.

@ZiglioUK I don't think it's a signal duration issue. Here is a 120 second signal with identical results:

https://www.dropbox.com/scl/fi/9kbzm3bcyhcr5h0lka1i9/3_4_8_14_21_120s.bin?rlkey=cyoy0je1cd6ptndzuuqdilgp3&dl=0

ZiglioUK, 2023-10-30:
@jnunezortuno 

`Now all subframes are decoded at least twice:`
So, what I see now is that yes, we get enough subframes so we can decode d_TOW_SF1.
Still, some ephemeris are discarded because d_IODC != d_IODE_SF2. I don't know what that means exactly but according to a comment it looks like a way to check we haven't missed any other frame (I'm guessing!).

So it'd be useful to have a file with 8 satellites, rather than 5.
Also, are you sure you've set up lat, lon correctly in the simulator?
Even uBlox thinks it's 0,0 with negative altitude, that can't be right:
[image]

`What does it means (prn_aux == in[i][epoch].PRN)`

I think that code is pulling ephemeris from a map. The problem is not there, the problem is in the RTK-LIB solver, it's either:
1. not enough valid pseudo-ranges are available for a single point solution
2. the computed satellite positions are not correct (even if force the ephemeris into the system by ignoring the errors above)

jwmelto, 2023-10-30:
I wonder if you could be seeing the effects that #737 was trying to fix?

ZiglioUK, 2023-10-30:
> I wonder if you could be seeing the effects that #737 was trying to fix?

Spot on! thank you @jwmelto 

Here we go @jnunezortuno : [GSDR303q29.23O.txt](https://github.com/gnss-sdr/gnss-sdr/files/13207528/GSDR303q29.23O.txt)

jnunezortuno, 2023-10-30:
> @jnunezortuno
>
> `Now all subframes are decoded at least twice:` So, what I see now is that yes, we get enough subframes so we can decode d_TOW_SF1. Still, some ephemeris are discarded because d_IODC [...]

Here you have a file with the signal corresponding to the whole visible constellation. As you can test, assuming that you use a 8 channel receiver a clear fix is obtained. Altitude is still negative (I don't know why).

https://www.dropbox.com/scl/fi/xi9gwtrofmm9n5azwa3dz/all.bin?rlkey=wcqq6req5ryrnzspfjsmdwso6&dl=0

Anyway, my goal is to obtain a fix with only 4 satellites with enough DOP.

jnunezortuno, 2023-10-30:
Now is working with the 120 seconds file with a TTFF in 48 s and using a 4 cannel receiver! Thanks @jwmelto for the catch and @ZiglioUK for debugging and for your patience.