gnss-sdr/gnss-sdr #236: Setting gains for bladeRF not working

oscarmezyk, 2019-01-06:
Hi,

please have a look on the output: 

> Initializing GNSS-SDR v0.0.10 ... Please wait.
> Logging will be written at ./logs
> OsmoSdr arguments: bladerf=0
> gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
> [bladeRF common] init: DEBUG: entering initialization
> [bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
> [bladeRF source] Device: Nuand bladeRF Serial # 5c77...5e15 FW v1.8.0 FPGA v0.9.0
> [bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
> [bladeRF source] bladerf_source_c: DEBUG: initialization complete
> Actual RX Rate: 4000000.000000 [SPS]...
> Actual RX Freq: 1575420000.000000 [Hz]...
> PLL Frequency tune error 0.000000 [Hz]...
> [WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1803] bladerf1_get_gain_stage: gain stage 'LNA' invalid
> Actual LNA Gain: 0.000000 dB...
> [WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1803] bladerf1_get_gain_stage: gain stage 'VGA1' invalid
> Actual VGA1 Gain: 0.000000 dB...
> [WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1803] bladerf1_get_gain_stage: gain stage 'VGA2' invalid
> Actual VGA2 Gain: 0.000000 dB...
> Actual Bandwidth: 28000000.000000 [Hz]...

Clearly gains stages are not recognized.  I could not tell which library is really responsible for this - it is most probably somewhere in osmo-sdr implementation that these values are not transfered properly to libbladeRF. I have noticed that others don't see this issue when using older setups: 
[/issues/233]

Any hints where to look for?
thanks

BushraAnsarii, 2019-03-11:
Hey, I m facing the same problem. 
your problem is resolved or still facing it?

oscarmezyk, 2019-03-11:
Hi, 
I haven't been playing with it too much recently, but quick check shows issue is still there.

I'm using:
v0.0.10.git-next-8934473
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
bladeRF: FW v2.3.2 FPGA v0.10.2

In the end using bladeRF-cli was fine for me.
best regards

BushraAnsarii, 2019-03-12:
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.10.git-next-8525782c ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: bladerf=0
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl_tcp uhd bladerf rfspace redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF Serial # 8ea3...8318 FW v2.3.1 FPGA v0.10.2
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
Actual RX Rate: 4000000.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1870] bladerf1_get_gain_stage: gain stage 'LNA' invalid
Actual LNA Gain: 0.000000 dB...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1870] bladerf1_get_gain_stage: gain stage 'VGA1' invalid
Actual VGA1 Gain: 0.000000 dB...
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1870] bladerf1_get_gain_stage: gain stage 'VGA2' invalid
Actual VGA2 Gain: 0.000000 dB...
Actual Bandwidth: 3000000.000000 [Hz]...
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
[bladeRF source] start: DEBUG: starting source
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 s
Loss of lock in channel 1!
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Loss of lock in channel 3!

BushraAnsarii, 2019-03-12:
Hey, do you have any idea how to solve it?

oscarmezyk, 2019-03-14:
If you mean how to repair it - no, I don't. But to set gains you can just use bladeRF-cli -i before starting gnss-sdr. BladeRF will remember all settings until you reset / reboot.
b.r.

erhankarakoca, 2020-04-24:
Hi, 

Also, I'm facing the same problem. I'm using bladeRF-xA9 with following properties
Firmware version : 2.3.2
FPGA version : 0.10.2 
gr-osmosdr : 0.1.5
gnuradio : 3.7.11

After a quick web search I found [that](https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Verifying-Basic-Device-Operation#gain) :  

**"On the bladeRF x40/x115, RX gain is implemented using three gain stages: LNA, rxvga1, and rxvga2. On the bladeRF xA4/xA9, there is one gain stage, full."**

However the example bladeRF config file designed for the LNA, rxvga1 and rxvga2 gains. That's the why these errors are happening (maybe it is deeper than that idk). When I comment SignalSource.gain and SignalSource.if_gain in the config file, errors are disappeared but I am sure that still GNSS-SDR not working correctly ( a lot of loss of locks happening) with bladeRF-xA9.
So I'm wondering ; 
1. Is there anyone can run GNSS-SDR with bladeRF-xA9/xA4 (not bladeRF x40/x115)   ? 
2. Is it possible to say that GNSS-SDR not working on bladeRF xA4 / xA9 because there is no defined parameter "full" in the config file for those devices ? Is there any way to edit the .conf file to enable bladeRF-xA9?
3. And may be more importantly I am very suspicious about it ; **I hope it is not happening because of  the **lack of bladeRF-xA9 support on GNSS-SDR.****

I'm waiting for answers and solutions,
Thanks.

Here is the copy of the terminal:

[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 6b8b...5a65 FW v2.3.2 FPGA v0.10.2
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
Actual RX Rate: 2000000.000000 [SPS]...
Actual RX Freq: 1575419998.000000 [Hz]...
PLL Frequency tune error -2.000000 [Hz]...
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_fpga.c:553] _rfic_fpga_get_gain_stage: unknown gain stage 'LNA'
[bladeRF source] get_gain: Could not get gain for stage 'LNA': Invalid operation or parameter
Actual LNA Gain: 0.000000 dB...
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_fpga.c:553] _rfic_fpga_get_gain_stage: unknown gain stage 'VGA1'
[bladeRF source] get_gain: Could not get gain for stage 'VGA1': Invalid operation or parameter
Actual VGA1 Gain: 0.000000 dB...
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_fpga.c:553] _rfic_fpga_get_gain_stage: unknown gain stage 'VGA2'
[bladeRF source] get_gain: Could not get gain for stage 'VGA2': Invalid operation or parameter
Actual VGA2 Gain: 0.000000 dB...
Actual Bandwidth: 18000000.000000 [Hz]...
[bladeRF source] start: DEBUG: starting source

lenhart, 2021-02-24:
currently at a related problem and wanted to point out for future onlookers that the last comment by @erhankarakoca uses a different bladeRF than above (path says "..board/bladerf2/..").

This issue/report #431 mentions that setting the gains manually before starting gnss-sdr works. (maybe a solution until a proper fix is found)