gnss-sdr/gnss-sdr #239: GLONASS L2 sample config does not start

gaudima, 2019-01-14:
Hello!
I just tried to launch sample glonass l2 config from this repo and it does not start. When I change this config to work with glonass l1 everything works as expected.

This seems to be some kind of a bug in gnuradio block connection, I tried to debug it but I dont have very much experience with gnuradio.

Here is config for glonass l2:

```
; This is a GNSS-SDR configuration file
; The configuration API is described at https://gnss-sdr.org/docs/sp-blocks/

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=6625000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=NT1065_GLONASS_L2_20160831_fs6625e6_60e3_schar_1m.bin ; <- PUT YOUR FILE HERE
SignalSource.item_type=ibyte
SignalSource.sampling_frequency=6625000
SignalSource.samples=0
SignalSource.dump=false;
SignalSource.dump_filename=/archive/signal_glonass.bin

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ibyte_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Pass_Through
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channel.signal=2G
Channels.in_acquisition=1
Channels_2G.count=5

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_2G.implementation=GLONASS_L2_CA_PCPS_Acquisition
Acquisition_2G.item_type=gr_complex
Acquisition_2G.threshold=0.0
Acquisition_2G.pfa=0.0001
Acquisition_2G.doppler_max=10000
Acquisition_2G.doppler_step=250
Acquisition_2G.dump=false;
Acquisition_2G.dump_filename=/archive/glo_acquisition.dat
;Acquisition_2G.coherent_integration_time_ms=1
;Acquisition_2G.max_dwells = 5

;######### TRACKING GLOBAL CONFIG ############
Tracking_2G.implementation=GLONASS_L2_CA_DLL_PLL_Tracking
Tracking_2G.item_type=gr_complex
Tracking_2G.early_late_space_chips=0.5
Tracking_2G.pll_bw_hz=20.0;
Tracking_2G.dll_bw_hz=2.0;
Tracking_2G.dump=false;
Tracking_2G.dump_filename=/archive/glo_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_2G.implementation=GLONASS_L2_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false;
Observables.dump_filename=/archive/glo_observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
[... 261 lines cut ...]
Current receiver time: 59 s
New GLONASS L1 GNAV almanac received in channel 4 from satellite Glonass PRN 17 (Block 4)
Position at 2019-Apr-09 18:40:28.500000 UTC using 4 observations is Lat = 40.007997585 [deg], Long = -105.262672807 [deg], Height = 1633.038 [m]
New GLONASS L1 GNAV almanac received in channel 1 from satellite Glonass PRN 16 (Block -1)
New GLONASS L1 GNAV almanac received in channel 2 from satellite Glonass PRN 15 (Block 0)
New GLONASS L1 GNAV almanac received in channel 3 from satellite Glonass PRN 02 (Block -4)
Position at 2019-Apr-09 18:40:29.000000 UTC using 4 observations is Lat = 40.007912609 [deg], Long = -105.262634828 [deg], Height = 1682.858 [m]
Current receiver time: 1 min 0 s
Position at 2019-Apr-09 18:40:29.500000 UTC using 4 observations is Lat = 40.007856490 [deg], Long = -105.262548554 [deg], Height = 1698.130 [m]
Position at 2019-Apr-09 18:40:30.000000 UTC using 4 observations is Lat = 40.007953850 [deg], Long = -105.262650501 [deg], Height = 1662.934 [m]
Current receiver time: 1 min 1 s
Position at 2019-Apr-09 18:40:30.500000 UTC using 4 observations is Lat = 40.007889537 [deg], Long = -105.262604262 [deg], Height = 1705.754 [m]
Position at 2019-Apr-09 18:40:31.000000 UTC using 4 observations is Lat = 40.007871577 [deg], Long = -105.262508008 [deg], Height = 1718.888 [m]
Current receiver time: 1 min 2 s
Position at 2019-Apr-09 18:40:31.500000 UTC using 4 observations is Lat = 40.008015871 [deg], Long = -105.262625386 [deg], Height = 1665.707 [m]
Position at 2019-Apr-09 18:40:32.000000 UTC using 4 observations is Lat = 40.007862867 [deg], Long = -105.262580203 [deg], Height = 1726.337 [m]
Current receiver time: 1 min 3 s
Position at 2019-Apr-09 18:40:32.500000 UTC using 4 observations is Lat = 40.007888298 [deg], Long = -105.262627770 [deg], Height = 1723.143 [m]
Position at 2019-Apr-09 18:40:33.000000 UTC using 4 observations is Lat = 40.007979438 [deg], Long = -105.262640538 [deg], Height = 1669.685 [m]
Current receiver time: 1 min 4 s
Position at 2019-Apr-09 18:40:33.500000 UTC using 4 observations is Lat = 40.007985968 [deg], Long = -105.262663266 [deg], Height = 1669.484 [m]
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 8.683296 [seconds]
GNSS-SDR program ended.
```

gaudima, 2019-01-16:
This should now be fixed after #240 merge.

carlesfernandez, 2019-01-28:
Thanks for your contribution! PR https://github.com/gnss-sdr/gnss-sdr/pull/240 has been merged into `next`.

carlesfernandez, 2019-01-28:
Please feel free to open a new issue if the problem persists.

closed 2019-01-28 by carlesfernandez