gnss-sdr/gnss-sdr #250: How we can read and run multiple input data file in GNSS-SDR C++ code

dinesh0612, 2019-02-07:


acebrianjuan, 2019-02-10:
Hi @dinesh0612 

If you mean processing more than one input file at the same time (multiple sources), GNSS-SDR can do it.
Please refer to the documentation page:

- [https://gnss-sdr.org/docs/sp-blocks/signal-source/#multiple-sources](https://gnss-sdr.org/docs/sp-blocks/signal-source/#multiple-sources)

dinesh0612, 2019-02-10:
Thanks for the reply,

No, it is not multiple sources. Source of data is single but it stored the
data in multiple files i.e. for 15 min duration data stored one data file
and next 15 min another file. So I want to process the first file first and
then the second file in GNSS-SDR Code.

Thank you

acebrianjuan, 2019-02-11:
In that case, you can combine both data files into a single file with the `cat` command as explained here:

- [https://stackoverflow.com/questions/10347278/how-can-i-copy-several-binary-files-into-one-file-on-a-linux-system](https://stackoverflow.com/questions/10347278/how-can-i-copy-several-binary-files-into-one-file-on-a-linux-system)

and then process the resulting file with GNSS-SDR as you would normally do.

Cheers,

Álvaro

dinesh0612, 2019-02-11:
Yes, I will do that. Thanks for your suggestion
I have one more question, I want to know whether the input data file is
reading at one go i.e. complete file or reading it by batch-wise during
acquisition and in tracking function (d_current_prn_length_samples). I am
asking this because in file_signal_sourse we are calculation total number
of byte present in a file for the calculation of total duration of data.
So, I think we are reading the whole file and using it in further
processing. If input data is reading at one go then it will take some time
to read the whole file that can be avoided by reading the file in
batch-wise.

Thank you

closed 2019-09-25 by dinesh0612