gnss-sdr/gnss-sdr #251: Decoding latitude and longitude from recored signal BladeRF

wojciechkrukar, 2019-02-12:
I am trying to obtain longitude and latitude from recorded GPS signal with BladeRF. 
I have trouble decoding it with GNS-SDR v0.0.10. on ubuntu 18.04

Antenna 17dB gain. The IQ signal looks pretty good when inspecting signal intensity (python) yet using the the configuration file I have trouble obtaining fix of Longitude and Latitude.

Here is configuration of BladeRF:
Bias Tee (RX1): on
RX1 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
RX1 Bandwidth:   5000000 Hz (Range: [200000, 56000000])
Setting RX1 sample rate 
-
req:   8000000 0/1Hz, actual:   8000000 0/1Hz

State: Running
Channels: RX1
Last error: None
File: /home/research/Desktop/GPS_IQ_recorded_data_1ch(SR8M_BW5M).bin

File format: SC16 Q11, Binary
# Samples: 503316480
# Buff
ers: 32
# Samples per buffer: 32768
# Transfers: 16
Timeout (ms): 1000

I am working with two configuration files but it feels like I am barking at the wrong tree. I have some form of synchronization but no position.
Here are links to configuration files I am fiddling with. 

https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/gnss-sdr_GPS_L1_bladeRF.conf

and of course with the original file for the firs https://gnss-sdr.org/my-first-fix/ 

the bin file is available here:
https://drive.google.com/file/d/1HcN5dvYeI3lt41_sBCEAqFowGH0N4c4k/view?usp=sharing 

Could you please help? I need to admit I am ferly new to the GNSS-SDR technology.
Is there special software configuration or installation outside of GNSS – SDR necessary? Is there any good tutorial how to deal with such file coming from BladeRF? Or maybe obvious way to set parameters in the config file + some additional drivers required? 
Thanks, Mark

oscarmezyk, 2019-02-12:
Hi. You're saying you have 17dB of antenna gain but you're processing a file - which is true? Where from you got the file of samples? What you mean by signal intensity? 

Please see previous issues on bladeRF:
https://github.com/gnss-sdr/gnss-sdr/issues/233
https://github.com/gnss-sdr/gnss-sdr/issues/243
https://github.com/gnss-sdr/gnss-sdr/issues/226

paste you're actuall config file and output of the gnss-sdr
b.r.
oskar

wojciechkrukar, 2019-02-13:
The pipeline in my project is that I will be receiving a raw files from the field and build *.gpx . The antenna in the contraption of my supplier has gain of 17dB. The reason I posted this info is that it could be important clue for troubleshooting. 
I'll follow your suggestion and come back with update.
Thank you,
Mark

Argonghost, 2023-11-15:
> The pipeline in my project is that I will be receiving a raw files from the field and build *.gpx . The antenna in the contraption of my supplier has gain of 17dB. The reason I posted this info is [...]

Any update on this issue?

Im facing similar trouble smh