gnss-sdr/gnss-sdr #256: Randomness in acquisition output for different run

dinesh0612, 2019-03-08:
Dear all,

When I run GNSS-SDR C++ code at different time then getting different acquisition test statistics, although I am running for the same input data file (gr_complex data) without any filtering and resampling. it is happening because input data used for calculating test statistics is changing at the different run for given PRNs. So, I want to know why input data is changing at the different run, when I used the same data file and starting from the beginning of the data file i.e. without skipping any initial sample. Also, I want to know how and where ninput_items and input_items values are changing for a different run. ninput_items and input_items are declared as input parameters of the general_work function. I am running pcps_acquisition_cc.cc acquisition file in GNSS-SDR c++ code.

Thanks, in advance

Rrravi, 2020-10-18:
I am also interested to know this since there is change in PRN's tracked at every time when we give  data file to GNSS-SDR eventhough data file is same it is happening like that

lenhart, 2021-04-16:
Hi @Rrravi and @dinesh0612 
(I know this is a bit older, I randomly stumbled across this issue). I noticed the same behaviour, that the gnss-sdr output is non-deterministic (when working with sample files). 

My explanation is that there are more satellites detectable than channels configured. I ran a few experiments and noticed that the output gets more repeatable when you ramp up the number of channels (like 20 or so. of course then the time required for progressing increases drastically). 

If you need it to be a bit more repeatable you can manually specify the satellite GNSS-SDR is supposed to track in the config file.

I am no expert in this project, so don't take this as an official answer, but I hope this might explain or help a bit :-)
Cheers

acse-mfp120, 2021-08-10:
Has anyone here found out why this is happening, or found a way to make things deterministic? I am seeing the exact same problem when building from source right now. @lenhart  Increasing the number of channels and limiting the satellites definitely does not improve things.

acse-mfp120, 2021-08-11:
@Rrravi, @dinesh0612  Try setting the number of acquisition channels to be equal to the number of tracking channels (and set the number of tracking channels equal to the number of satellites you are interested in)

Rrravi, 2021-08-11:
Thank you sir for your answer, but the query is one year old one and now it
is solved.

acse-mfp120, 2021-08-11:
Then please close this issue.

lenhart, 2021-08-20:
Hi
@Rrravi : could you brifly mention how you solved it? The issue is currently on the mailinglist again, so an answer might be interesting for a wider audience.
@acse-mfp120 : did the increase in acquisition channels made it more deterministic to you then? (I am a bit confused because you ask and also answer a day later :-) Just to make sure it is a founded recommendation). 

And I remembered that when I was working with the non-deterministic-problem, I was more taking the resulting pvt solutions as a way to compare the executions, so there it helped to increase the overall number of channels, but I see that I wouldn't necc. help to the specific question.

Cheers