gnss-sdr/gnss-sdr #257: Cannot read files generated with Spirent GSS6450

valcesch, 2019-03-11:
Hello,

Thank you for your work.

I am trying to process files from a Spirent GSS6450, nevertheless the following error message is displayed:

	gnss-sdr@gnsssdr-VirtualBox:~/work$ sudo sudo gnss-sdr --config_file=./GSS6450_default.conf
	Initializing GNSS-SDR v0.0.10 ... Please wait.
	Logging will be written at "/tmp"
	Use gnss-sdr --log_dir=/path/to/log to change that.
	Processing file 190311095144.A.gns, which contains 3517710708 [bytes]
	GNSS signal recorded time to be processed: 35.1745 [s]
	E0311 14:57:05.281533  5958 gnss_flowgraph.cc:137] vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
	E0311 14:57:05.281750  5958 control_thread.cc:236] Unable to connect flowgraph
	Total GNSS-SDR run time: 0.000366364 [seconds]
	GNSS-SDR program ended.

And the configuration file:

	[GNSS-SDR]

	;######### GLOBAL OPTIONS ##################
	GNSS-SDR.internal_fs_sps=2000000

	;######### SIGNAL_SOURCE CONFIG ############
	SignalSource.implementation=Spir_GSS6450_File_Signal_Source
	SignalSource.filename=190311095144.A.gns
	SignalSource.sampling_frequency=50000000
	SignalSource.adc_bits=8

	;######### SIGNAL_CONDITIONER CONFIG ############
	SignalConditioner.implementation=Signal_Conditioner
	DataTypeAdapter.implementation=Ishort_To_Complex

I am using gnss-sdr 0.0.10 and acquisition/tracking is working fine with the example from "My first position fix".

Thank you in advance for your help,
Raphael

carlesfernandez, 2019-03-11:
Dear Raphael,

please take into account that the `Spir_GSS6450_File_Signal_Source` implementation delivers samples in `gr_complex` format, so you might need to change the line `DataTypeAdapter.implementation=Ishort_To_Complex` (and maybe some other changes in the Signal Conditioner).

Probably, something as
`DataTypeAdapter.implementation=Pass_Through`
can work for you.

Best regards,
Carles

valcesch, 2019-03-11:
Dear Carles,

Thank you for the very quick answer.

Indeed, I have made several tests and forget to change this line. Here is the actual configuration file:

	[GNSS-SDR]

	;######### GLOBAL OPTIONS ##################
	GNSS-SDR.internal_fs_sps=2000000

	;######### SIGNAL_SOURCE CONFIG ############
	SignalSource.implementation=Spir_GSS6450_File_Signal_Source
	SignalSource.filename=190311095144.A.gns
	SignalSource.sampling_frequency=50000000
	SignalSource.adc_bits=8

	;######### SIGNAL_CONDITIONER CONFIG ############
	SignalConditioner.implementation=Signal_Conditioner
	DataTypeAdapter.implementation=Pass_Through

Nevertheless, the error is still present. I had a look at the log files, but there are not more details than in the command line output. I have also made a test with the next branch, without success. Could it be a change in GSS6450 output format after a firmware update that is not compatible anymore with the adapter in GNSS-SDR ?

Best regards,
Raphael

carlesfernandez, 2019-03-11:
Hi,

from what I've seen, parameter `bytes_to_skip` (the lenght of the binary header) varies with the version of the Spirent firmware. Please check your documentation, or open a raw data file with an hex editor and check how many bytes your receiver is putting in the header.

Hope this helps,
Carles

valcesch, 2019-03-12:
Hello,

Yes, indeed, I checked the length of the header and it is much smaller (around 370 bytes). I tried to increase progressively the number of bytes to skip, but without success. I have also tried with a much larger number than the default, but still the same error message.

What would be the best solution to track the error in the code ?

Thank you in advance,
Raphael

yealwang, 2019-11-20:
Hi 
Have you solve the problem, now i am confusing about which front-end is suitable for GNSSSDR, I have the stereo V2 front-end data and the flexiband data, but they all have problems with GNSSSDR.
Thany you 
Yeal

valcesch, 2019-11-27:
Hi,
No, problem is still unsolved. Nevertheless, I found out that in the file spir_gss6450_file_signal_source.cc, there are conditions with:

	if (n_channels_ > 1)

If you set 

	if (n_channels_ > 0)

Then you can run GNSS-SDR with only one channel. But still, tracking only works for a few seconds. I have also been through Spirent format recording documentation and the actual implementation of the code is perfectly fine. So problem should be after file reading.

Bests,
Raphael

SpirentQZ, 2021-10-11:
Dear Raphael,
Not sure if you have solved the issue. I have got similar issue as below
qzhang@debian:~/sdr/work$ gnss-sdr --config_file=./gnss-sdr_GPS_L1_SPIR.conf
Initializing GNSS-SDR v0.0.15.git-next-decade163d ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/qzhang/sdr/work/sim_4b_gps_l1/211005175328.A.gns, which contains 3347841404 [bytes]
GNSS signal recorded time to be processed: 111.591 [s]
RF Channels: 1
E1011 16:18:05.891134  3611 gnss_flowgraph.cc:685] Can't connect signal source block 0 internally: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Unable to connect flowgraph
Total GNSS-SDR run time: 0.00229629 [seconds]
GNSS-SDR program ended.

I have done some tests as above suggestions from Carles. If you have solved the issue, please let me know. 

Thanks in advance. 

Regards

Qiang