gnss-sdr/gnss-sdr #259: SignalSource.Osmosdr_Signal_Source: Undefined implementation for block

wxhahxw, 2019-03-18:
Hi, I am trying to catch the real-time GPS L1 signal by bladerf. 
I used the config file: 
https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/gnss-sdr_GPS_L1_bladeRF.conf 

And the command below:
xx@xx-OptiPlex-9020:~/gnss-sdr$ gnss-sdr --config-file=/home/xx/bladerf_gnsssdr.conf
Then the log shows:
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.10.git-next-b409f1c1 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
E0318 21:05:08.356220 17252 gnss_block_factory.cc:1809] SignalSource.Osmosdr_Signal_Source: Undefined implementation for block
Segmentation fault (core dumped)

I checked the SignalSource is Osmosdr_Signal_Source. And osmosdr_signal_source.h and osmosdr_signal_source.cc exist in the folder.

Seek for your ideas

oscarmezyk, 2019-03-18:
Osmosdr is not turned on by default - you need to pass  -DENABLE_OSMOSDR=ON  to the compiler. I don't know how you installed the gnss-sdr, so won't tell you the exact way to do it - read the installation guidelines 
b.r.

wxhahxw, 2019-03-21:
Hi oscarmezyk, I  installed the gnss-sdr through PyBOMBS. 
I tried to use cmake -DENABLE_OSMOSDR=ON, it said no CMakeList.txt found.  
Then, I tried to install the osmo-sdr under the folder /gnss-sdr/src. 
Then, I  use $cmake -DENABLE_OSMOSDR=ON under /pybombs/gnss-sdr/src/gnss-sdr (because there is a CMakeList.txt). 
Then $make and $sudo make install. It installed successfully.
Then i try to use command $gnss-sdr --config_file=/home/xiao/pybombs/gnss-sdr/mybladerf.conf

the error shows again 
E0321 16:32:35.571974  3823 gnss_block_factory.cc:1933] SignalSource.Osmosdr_Signal_Source: Undefined implementation for block
Segmentation fault (core dumped)

oscarmezyk, 2019-03-21:
what is the output of "pybombs inv"?

wxhahxw, 2019-03-22:
[INFO] Prefix Python version is: 2.7.15
[INFO] PyBOMBS Version 2.3.4a0
Showing package state:
apache-thrift:      installed
uhd:                installed
gnuradio:           installed
gnss-sdr:           installed

wxhahxw, 2019-03-22:
xiao@xiao-Precision-Tower-5810:~/pybombs/gnss-sdr$ pybombs inv
[INFO] Prefix Python version is: 2.7.15
[INFO] PyBOMBS Version 2.3.4a0
Showing package state:
apache-thrift:      installed
uhd:                installed
gnuradio:           installed
rtl-sdr:            installed
osmo-sdr:           installed
libosmo-dsp:        installed
gr-iqbal:           installed
bladeRF:            installed
soapysdr:           installed
hackrf:             installed
airspy:             installed
gr-osmosdr:         installed
gqrx:               installed

wxhahxw, 2019-03-22:
Hi  oscarmezyk ,  I tried again to $cmake -DENABLE_OSMOSDR=ON , 
My device start to track the channel  when i  use $ gnss-sdr --config_file=/home/xiao/Desktop/mybladerf.conf.

I closed it since it keep loss tracking.
But when I want to retry, there is another error:

xiao@xiao-Precision-Tower-5810:~$ gnss-sdr --config_file=/home/xiao/Desktop/mybladerf.conf
gnss-sdr: error while loading shared libraries: libgnuradio-filter-3.7.13.4.so.0.0.0: cannot open shared object file: No such file or directory
I reinstall the gnss-sdr and omso-sdr. And it happened again, work first, then error.
Looking for you help. Thanks.

oscarmezyk, 2019-03-24:
Your pybombs inv doesn't show "gnss-sdr"  -  that means you probably messed up pybombs&apt installation. I suggest you remove everything and install from scratch using **only** pybombs. 
Install uhd, gnuradio, bladeRF, gqrx etc.

Before installing gnss-sdr:
- go to pybombs recipes: cd ~/.pybombs/recipes/gr-recipes/
- edit: gnss-sdr.lwr
- at the end of the file add a line that enables osmocom:

```
vars:
  config_opt: ' -ENABLE_OSMOSDR=ON'

```
Then install or rebuild gnss-sdr using pybombs.

wxhahxw, 2019-03-24:
Dear oscarmezyk，

Thanks a lot，I will try to reinstall all tomorrow. Good night.

Regards
Xiao

carlesfernandez, 2019-07-03:
Please feel free to reopen the issue if you can't install gnss-sdr.

closed 2019-07-03 by carlesfernandez

evso9816, 2020-12-04:
Hi
It seems like I encountered similar problem after installing everything (gnuradio, osmo-sdr, gnss-sdr). Running the program Realtek dongle full configuration file, I have the following error:
gnss-sdr --config_file=./my_rtlsdr.conf
Initializing GNSS-SDR v0.0.13.git-next-30b52cd68e ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
E1203 23:23:29.732079  4307 gnss_block_factory.cc:1290] SignalSource.Osmosdr_Signal_Source: Undefined implementation for block 
Segmentation fault (core dumped)

Running "pybombs inv", here is my installed:
pybombs inv
[INFO] Prefix Python version is: 3.8.5
[INFO] PyBOMBS Version 2.3.4
Showing package state:
libvolk:            installed
uhd:                installed
gnuradio:           installed
airspy:             installed
airspyhf:           installed
gr-fcdproplus:      installed
soapysdr:           installed
osmo-sdr:           installed
bladeRF:            installed
hackrf:             installed
rtl-sdr:            installed
libosmo-dsp:        installed
gr-iqbal:           installed
gr-osmosdr:         installed
gnss-sdr:           installed

Question: what did i do wrong? Should I uninstall gnss-sdr, add to "gnss-sdr.lwr" "vars:   config_opt: ' -ENABLE_OSMOSDR=ON' ", and make again "pybombs install gnss-sdr"? I need support. Thanks.

gongfx, 2022-01-24:
> Your pybombs inv doesn't show "gnss-sdr" - that means you probably messed up pybombs&apt installation. I suggest you remove everything and install from scratch using **only** pybombs. Install uhd, [...]

should be: config_opt: ' -DENABLE_OSMOSDR=ON'