gnss-sdr/gnss-sdr #260: Large pseudoranges discontinuities while continuous operation

valcesch, 2019-03-19:
Dear developers,

I am using GNSS-SDR (v0.0.10) with a USRP N210 operated in common clock and common antenna with a geodetic receiver. 

After running the setup for a few hours, I compared the pseudoranges measured with GNSS-SDR against the one generated with the hardware receiver. I noticed large discontinuities that could probably appears (I suppose) when data packets coming from the USRP are lost.

[image]

[image]

Have you already noted such behaviour and would you have a way to solve it? 

Thank you in advance for the help,
Raphael

carlesfernandez, 2019-03-22:
👍 looking into it

odrisci, 2019-03-22:
Hi Raphael, Carles,

It seems to me that this is most likely due to the way that the hybrid_observables are generated. From what I remember the earliest satellite is taken as a reference and all the other pseudoranges are computed relative to that. When there is a change in the earliest satellite there will be an associated jump in the clock bias. I'm currently working on a solution that uses "receiver synchronous" observables, but it is in a very experimental state right now, I'll push some changes to my fork in the coming weeks.

In essence the current implementation is a "common transmit time" approach, while I'm implementing a "common receiver time" approach, see [here](https://insidegnss.com/code-tracking-and-pseudoranges/) for definitions. The common receiver time approach does not have these jumps, but requires a mechanism to keep track of the receiver clock bias (relating the receiver time in samples since we started, to local time expressed in GPS time, GST, UTC, etc). I'm currently testing out a "GnssTime" class that should help with this, but it is getting out of hand!

Cillian

valcesch, 2019-03-25:
Hi Carles, Cillian,
Thank you for your answers. Unfortunately, I will not have enough expertise to help for the implementation, but I can of course support for the testing.
Raphael

valcesch, 2019-07-17:
Hi Carles, Cillian,
Any update on this topic? I have a limited expertise in GNSS but I would be interested to contribute to the development of the "common receiver time" approach you have proposed Cillian. 
The final goal for me would be to have sort of a SDR based timing receiver that can be used for time transfer.
Would you have any early stage code that you could share as a starting point ?
Thank you in advance,
Raphael

donghuixin, 2019-07-24:
How did you draw this pseudo-range map?

odrisci, 2019-07-24:
Sorry @valcesch I haven't really had much time to put into this lately. I've pushed my version of the observables to my [fork](https://github.com/odrisci/gnss-sdr/tree/rx_synch_obs). This is not really properly tested yet, but gives an idea of the approach that I propose.

You can view a comparison between this fork and the `next` branch [here](https://github.com/gnss-sdr/gnss-sdr/compare/next...odrisci:rx_synch_obs)

valcesch, 2019-07-29:
@donghuixin, this is a plot of the pseudoranges contained in the RINEX generated with the SDR (USRP N210) operated in common clock (10MHz IN) + common antenna with a timing receiver. 

@odrisci, thank you for your answer, I am currently running through your code and hope to be able to conduct some tests very soon.

In parallel, I was also working on another implementation, which should solve the pseudorange discontinuities, but requires an accurate and stable time source locked to UTC for example. My idea is to synchronize the SDR measurements using 10MHz and 1PPS inputs of the USRP N210. In such a way, I know the absolute epoch of every sample. By subtracting the TOW broadcasted in the navigation message and the TOW of the reception, I should be able to obtain directly the “absolute” pseudorange, without the need of relative TOW comparison between the satellite messages.

On my [fork](https://github.com/valcesch/gnss-sdr), I am able to set the time of the first sample sent by the SDR (using set_start_time() from UHD library). The call of the function is done just before the start of the gr flow graph. Nevertheless, when monitoring the `traveltime_ms` variable in `hybrid_observables_gs.cc`, I have noticed that the signal travel time can be up to 500ms. I expect a few ns jitter... This makes me think that after starting the gr flow graph, some samples are lost before starting the processing itself (some samples are indeed lost between the acquisition and the tracking, but the number of samples consumed is known).

If you have some inputs on what could be the cause of this issue it would be appreciated.

Thanks in advance,
Raphael

carlesfernandez, 2019-07-30:
Hi guys,

the solution currently implemented in hybrid_observables_gs.cc works as expected with synthetically-generated signals where the clock offset is not simulated. However, when using real signals with real hardware, the results I'm getting are the same as yours, with those ugly discontinuities in the pseudoranges (and, of course, also in the carrier phase measurements).

The problem is clearly the way we update that clock offset: we correct the clock when the offset reaches 1 ms. This keeps the time solution within 1 ms of error, but it causes those discontinuities when the correction is applied. 

We are trying to change that (that is, applying the correction each time we have a PVT fix) in order to get rid of those discontinuities. We expect to come with a solution soon.

Thanks for your interest,
Carles

carlesfernandez, 2019-08-03:
Hi folks,

it seems that the latests changes in `next` solve this problem. A test with a RTL-SDR provided continuous pseudoranges in a test we did for more than 20 hours. Longer tests will be conducted this weekend, and if everything is ok, we will release 0.0.11 early next week shipping this change.

Any test report using your own hardware (and differences w.r.t your own implementations) will be appreciated!

Best regads,
Carles

valcesch, 2019-08-05:
Hi Carles,

Indeed, I have made a test and all is continuous and clean:

[image]
[image]

Nevertheless, in my initial post, I think that the problem was more related to interruption of the data flow coming from the SDR. Once lock is loss, GNSS-SDR will again start with a default pseudorange assigned to the closest satellite, which will produce a jump. I have changed my setup in order to not have anymore these interruption in the data flow.

I also mention that in the two posts, the hardware receiver and SDR are locked to a stable and accurate reference (always less than +/-5ns away from GPS time). So I do not expect to see any clock correction from the software.

In the meantime, I was also looking directly at the pseudorange difference (as time delay) and observed some repetitive pattern:

[image]

I do not clearly see why I observe such pattern if the two are connected to the same antenna. Once again, the hardware receiver and SDR are operated in common clock and common antenna.

Shall I move this to a new subject ?

Thanks in advance,
Raphael

carlesfernandez, 2019-08-21:
Thanks Raphael!

Pseudoranges and carrier phase observables look fine now, but I'm wondering why are you getting these patters when comparing pseudoranges. We did a very similar test and the results were almost zero.

Could you please move this into a new issue, so we can close this one?

Best regards,
Carles

valcesch, 2019-09-24:
Hi Carles, 
I can finally move this last point to a new issue and close this one.
Best regards,
Raphael

closed 2019-09-24 by valcesch