gnss-sdr/gnss-sdr #262: Segmentation fault at gnss_flowgraph.cc:181

johnwstanford, 2019-03-30:
I'm getting a segmentation fault in the latest commit when I try to run with the example configuration that reads a file into eight GPS L1 C/A channels.  I traced it to the exact line by printing to stdout.

```
// In GNSSFlowgraph::connect():
std::cout << "At gnss_flowgraph:181 " << channels_.at(i) << std::endl;
channels_.at(i)->connect(top_block_);
std::cout << "At gnss_flowgraph:183 " << std::endl;
```

Result (stdout):

```
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
At gnss_flowgraph:181 0
Segmentation fault (core dumped)
```

Running on Ubuntu 18.04.

acebrianjuan, 2019-03-30:
Hi @johnwstanford 

I would like to replicate the issue on my system.

What configuration and what raw signal file are you using? I believe you are referring to the example in the [_"My first position fix"_](https://gnss-sdr.org/my-first-fix/) tutorial. Can you please confirm?

In addition, can you please share your `gnss-sdr.INFO`, `.WARNING` and `.ERROR` logs? They are stored in the "/tmp" folder, by default.

That would be very helpful.

Álvaro

johnwstanford, 2019-03-30:
I was originally using a different signal that I captured myself with an Ettus USRP device, but I tried it with the data from "My first position fix" and got the same results.  There is no `.ERROR` log, but I attached the other two.

It looks like from the `.WARNING` file that the problem is coming from the try/catch block in `GNSSBlockFactory::GetChannels`

Let me know if I can do anything else to help.  Thanks!

[gnss-sdr.zip](https://github.com/gnss-sdr/gnss-sdr/files/3025562/gnss-sdr.zip)

acebrianjuan, 2019-03-30:
Thank you for providing the logs.

I built the [`next`](https://github.com/gnss-sdr/gnss-sdr/tree/next) branch with the latest changes and ran the [_"My first position fix"_](https://gnss-sdr.org/my-first-fix/) example without any errors whatsoever.

What version is printed when you run the command `gnss-sdr --version`?
Can you please share your receiver configuration file?

Álvaro

johnwstanford, 2019-03-30:
Here's the configuration file.

I traced it down further to pcps_acquisition.cc.  I'm not sure why making a new gr::fft::fft_complex would throw an exception but it looks like that's what's happening.  I have the gnuradio shared library installed and I don't think the program would successfully build and load if I didn't.

```
    // Direct FFT
    std::cout << "JWS3: Good so far, before d_fft_if, d_fft_size=" << d_fft_size << std::endl;
    d_fft_if = new gr::fft::fft_complex(d_fft_size, true);
    std::cout << "JWS3: Good so far, after d_fft_if" << std::endl;

    // Inverse FFT
    d_ifft = new gr::fft::fft_complex(d_fft_size, false);
    std::cout << "JWS3: Good so far, after d_ifft" << std::endl;

```

Output:

```
Starting a TCP/IP server of RTCM messages on port 2101
JWS3: Good so far, before GetAcqBlock
JWS3: Good so far, before new GpsL1CaPcpsAcquisition
JWS3: Good so far, before pcps_make_acquisition
JWS3: Good so far, before d_fft_if, d_fft_size=2000
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Segmentation fault (core dumped)
```

[first_position_fix.zip](https://github.com/gnss-sdr/gnss-sdr/files/3025616/first_position_fix.zip)

acebrianjuan, 2019-03-30:
I ran the receiver with the configuration file you sent me and I get no crashes at all.

What GNSS-SDR version are you using?

Álvaro

johnwstanford, 2019-03-30:
This is my version:

```
john@deathStar:~/SDR/GNSS/temp$ gnss-sdr --version
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gnss-sdr version 0.0.10.git-HEAD-c877e745
```

Which gnuradio versions are you building and running with?  If you're using the PPA repository, what do you get when you run this?

```
john@deathStar:~/SDR/GNSS/temp$ dpkg -l gnuradio*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                          Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  gnuradio                                      3.7.11-10                   amd64                       GNU Radio Software Radio Toolkit
ii  gnuradio-dev:amd64                            3.7.11-10                   amd64                       GNU Software Defined Radio toolkit development

```

acebrianjuan, 2019-03-30:
> This is my version:
> 
> ```
> john@deathStar:~/SDR/GNSS/temp$ gnss-sdr --version
> linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown
> 
> gnss-sdr version 0.0.10.git-HEAD-c877e745
> ```

I can see that you are not building the latest development version of GNSS-SDR, as your git HEAD is pointing to commit [`c877e745`](https://github.com/gnss-sdr/gnss-sdr/commit/c877e745652d43e2d35d136130f2f626a75d56b7), which is a few commits behind the current development version.

Are you building at that point in history for any particular reason?

> Which gnuradio versions are you building and running with? If you're using the PPA repository, what do you get when you run this?

I installed all the dependencies (including GNURadio) following the instructions detailed in the section [Install all dependencies using packages](https://gnss-sdr.org/build-and-install/#install-all-dependencies-using-packages) of the the [Build and install GNSS-SDR](https://gnss-sdr.org/build-and-install) guide.

As you can see, I have the same GNURadio versions as you:

```
alvaro@msi:~$ dpkg -l gnuradio*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                Version        Architecture   Description
+++-===================-==============-==============-===========================================
ii  gnuradio            3.7.11-10      amd64          GNU Radio Software Radio Toolkit
ii  gnuradio-dev:amd64  3.7.11-10      amd64          GNU Software Defined Radio toolkit developm

```

Nevertheless, I did a `git checkout c877e745` to point my git HEAD to the same commit as you, re-built the code and then ran the receiver using the configuration file that you provided, and everything is working smoothly. I could not find any signs of an error or malfunction as you can see below:

```
alvaro@msi:~/work$ $HOME/git/gnss-sdr/install/gnss-sdr -c ./first_position_fix.conf 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.10.git-HEAD-c877e745 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/alvaro/signals/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 20 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 32 (Block IIF)
Current receiver time: 2 s
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 3 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 11 (Block IIR)
Current receiver time: 4 s
Current receiver time: 5 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 6 s
Loss of lock in channel 4!
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Current receiver time: 14 s
Current receiver time: 15 s
Current receiver time: 16 s
Current receiver time: 17 s
Loss of lock in channel 2!
[... 171 lines cut ...]
Current receiver time: 1 min 38 s
Position at 2013-Apr-04 06:24:44.000000 UTC using 5 observations is Lat = 41.274899770 [deg], Long = 1.987693575 [deg], Height = 76.572 [m]
New GPS NAV message received in channel 1: subframe 5 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 5 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 7: subframe 5 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel New GPS NAV message received in channel 6: subframe 5 from satellite 3: subframe 5 from satellite GPS PRN 17 (Block IIR-M)GPS
 PRN 11 (Block IIR)
Position at 2013-Apr-04 06:24:44.500000 UTC using 5 observations is Lat = 41.274813864 [deg], Long = 1.987673940 [deg], Height = 93.947 [m]
Current receiver time: 1 min 39 s
Position at 2013-Apr-04 06:24:45.000000 UTC using 5 observations is Lat = 41.274870454 [deg], Long = 1.987676467 [deg], Height = 71.659 [m]
Position at 2013-Apr-04 06:24:45.500000 UTC using 5 observations is Lat = 41.274848398 [deg], Long = 1.987682214 [deg], Height = 77.002 [m]
Current receiver time: 1 min 40 s
Position at 2013-Apr-04 06:24:46.000000 UTC using 5 observations is Lat = 41.274853838 [deg], Long = 1.987634740 [deg], Height = 54.362 [m]
Position at 2013-Apr-04 06:24:46.500000 UTC using 5 observations is Lat = 41.274848088 [deg], Long = 1.987632715 [deg], Height = 65.025 [m]
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 5.105989 [seconds]
GNSS-SDR program ended.
```

At this point, I would suggest you building the latest version of [`next`](https://github.com/gnss-sdr/gnss-sdr/tree/next) and see if that solves the problem.

I hope you find this information helpful.

Álvaro

acebrianjuan, 2019-03-30:
I am also running Ubuntu 18.04 by the way.

Álvaro

johnwstanford, 2019-03-30:
Ok.  I'm still going to keep digging until I figure out why instantiating a gnuradio FFT is causing an exception.  Want me to leave the issue open so that if I figure it out, I can post it?

johnwstanford, 2019-03-30:
I'm getting an exception when I try to instantiate an `fft_complex` even outside gnss-sdr.

Source code:
```
#include <iostream>

#include <boost/thread.hpp>

#include <gnuradio/fft/api.h>
#include <gnuradio/fft/fft.h>
#include <gnuradio/gr_complex.h>

int main() {

	auto fft = new gr::fft::fft_complex(2000, false);

	std::cout << "Success" << std::endl;

}
```

Result:

```
john@deathStar:~/Desktop/gr_test$ g++ ./main.cpp -lboost_system -lgnuradio-fft -o main
john@deathStar:~/Desktop/gr_test$ ./main
terminate called after throwing an instance of 'std::exception'
  what():  std::exception
Aborted (core dumped)
```

acebrianjuan, 2019-03-31:
> I'm getting an exception when I try to instantiate an `fft_complex` even outside gnss-sdr.

I compiled your `main.cpp` program and I get `Success` after running it:

```
alvaro@msi:~/git/fft_complex$ ./main 
Success
```

> Ok. I'm still going to keep digging until I figure out why instantiating a gnuradio FFT is causing an exception. Want me to leave the issue open so that if I figure it out, I can post it?

Since this issue is not being caused by GNSS-SDR, I would recommend closing this issue and getting in touch with the GNURadio community or opening a new issue in their official repository, as you have done already (issue [#2427](https://github.com/gnuradio/gnuradio/issues/2427)).

Thank you.

Álvaro

closed 2019-03-31 by johnwstanford