gnss-sdr/gnss-sdr #266: Duplicate assignment in dll_pll_veml_tracking.cc:1357-1358

johnwstanford, 2019-04-14:
In lines 1357 and 1358 of dll_pll_veml_tracking.cc, you assign different values to the same variable two lines in a row.  Is this a mistake?

acebrianjuan, 2019-04-14:
Hi @johnwstanford 

Lines 1357-1358 in the `master` branch correspond to lines 1581-1582 in the `next` branch:

https://github.com/gnss-sdr/gnss-sdr/blob/7d3a2322676ef33210c1eb4d724216b02cd69247/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L1581-L1582

I can't give an authoritative answer, but this indeed looks like a mistake.

Álvaro

carlesfernandez, 2019-04-14:
Yes, it is. The Doppler correction is small, so that part was disconnected but the corresponding line was not removed. Fixed in 7fc8d53

acebrianjuan, 2019-04-15:
@johnwstanford,

Please consider closing this issue, as recommended in [CONTRIBUTING.md#reporting-an-issue](https://github.com/gnss-sdr/gnss-sdr/blob/master/CONTRIBUTING.md#reporting-an-issue), now that the problem has been fixed by @carlesfernandez 

Thank you,

Álvaro

commit 7fc8d535 (2019-04-14): Remove double assignment. Fixes #266
closed 2019-04-15 by johnwstanford