gnss-sdr/gnss-sdr #271: Rollover in RINEX output

matteocutugno, 2019-05-04:
Dear all,

i'm experiencing that since GPS week Rollover happened last 6th of April RINEX output from my receiver is affected by this issue. I get year 1999 in both nav and obs file.
I use GNSS-SDR 0.0.10 in Ubuntu 16.04 acquiring signal with Nuand bladeRF x40.
How can this problem been solved?

Thank you 

MC

acebrianjuan, 2019-05-05:
Hi @matteocutugno 

Today I have made a live capture using an RTL-SDR V3 dongle with the latest version of GNSS-SDR (version 0.0.10.git-next-cd0e7ada) and I have not found any anomaly in the dates reported in the RINEX NAV and OBS files.

As you can see in the following snippets, today's date (05/05/2019) is reported correctly.

- OBS file:

```
     3.02           OBSERVATION DATA    G                   RINEX VERSION / TYPE
G = GPS  R = GLONASS  E = GALILEO  S = GEO  M = MIXED       COMMENT             
GNSS-SDR            alvaro              20190505 115300 UTC PGM / RUN BY / DATE 
GPS OBSERVATION DATA FILE GENERATED BY GNSS-SDR             COMMENT             
GNSS-SDR VERSION 0.0.10.git-next-cd0e7ada                   COMMENT             
See https://gnss-sdr.org                                    COMMENT             
DEFAULT MARKER NAME                                         MARKER NAME         
alvaro              CTTC                                    OBSERVER / AGENCY   
GNSS-SDR            Software Receiver   0.0.10.gi           REC # / TYPE / VERS 
Antenna number      Antenna type                            ANT # / TYPE        
        0.0000        0.0000        0.0000                  APPROX POSITION XYZ 
        0.0000        0.0000        0.0000                  ANTENNA: DELTA H/E/N
G    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES 
DBHZ                                                        SIGNAL STRENGTH UNIT
  2019    05    05    11    53   17.9000000     GPS         TIME OF FIRST OBS   
                                                            END OF HEADER       
> 2019 05 05 11 53 18.0000000  0  4                                             
G25  15710592.131 7    -44700.981 7       817.978 7        45.262               
G24  16698651.901 7    137806.580 7     -2308.452 7        43.463               
G12  14283310.455 8    145960.510 8     -1084.024 8        49.171               
G14  17378638.582 6   -145864.197 6      1017.366 6        40.722               
...
```

- NAV file:

```
     3.02           N: GNSS NAV DATA    G: GPS              RINEX VERSION / TYPE
GNSS-SDR            alvaro              20190505 115300 UTC PGM / RUN BY / DATE 
GPS NAVIGATION MESSAGE FILE GENERATED BY GNSS-SDR           COMMENT             
GNSS-SDR VERSION 0.0.10.git-next-cd0e7ada                   COMMENT             
See https://gnss-sdr.org                                    COMMENT             
GPSA    .0000D+00   .0000D+00   .0000D+00   .0000D+00       IONOSPHERIC CORR    
GPSB    .0000D+00   .0000D+00   .0000D+00   .0000D+00       IONOSPHERIC CORR    
GPUT   .0000000000D+00  .000000000D+00      0 1024          TIME SYSTEM CORR    
     0     0     0     0                                    LEAP SECONDS        
                                                            END OF HEADER       
G12 2019 05 05 12 00 00  .245233066380D-03 -.341060513165D-11  .000000000000D+00
      .270000000000D+02  .265625000000D+01  .447661504041D-08  .112396095801D+01
      .502914190292D-07  .765871407930D-02  .130943953991D-05  .515364822578D+04
      .432000000000D+05 -.158324837685D-06  .164846251450D+01  .223517417908D-07
      .982716922584D+00  .367812500000D+03  .107654343088D+01 -.845178062202D-08
     -.158935191717D-09  .100000000000D+01  .102800000000D+04  .100000000000D+01
      .000000000000D+00  .000000000000D+00 -.121071934700D-07  .270000000000D+02
      .427980000000D+05  .400000000000D+01                                      
...
```

I am running Ubuntu 18.04.

I would like to learn more details about your issue to see if I can replicate it with my setup.

Can you please share your RINEX files?
What receiver configuration were you using?

Álvaro

matteocutugno, 2019-05-05:
Hi,
first of all thank you for your reply.

I use:

bladeRF-cli version:        1.7.1-git-1da130c
libbladeRF version:         2.2.0-git-1da130c
gnss-sdr version 0.0.10.git-next-8799098
ubuntu 16.04LTS  on a  Intel® Core™ i7-8550U CPU @ 1.80GHz × 8 

My hardware configuration for the acquisition is completed by a Nuand bladerf x40 bias tee and active ublox antenna.
Here my RINEX obs and nav files:

     3.02           OBSERVATION DATA    M                   RINEX VERSION / TYPE
G = GPS  R = GLONASS  E = GALILEO  S = GEO  M = MIXED       COMMENT             
GNSS-SDR            UNKNOWN USER        20190504 113542 UTC PGM / RUN BY / DATE 
MIXED (GPS/GALILEO) OBSERVATION DATA FILE GENERATED BY GNSS-COMMENT             
GNSS-SDR VERSION 0.0.10.git-next-8799098                    COMMENT             
See https://gnss-sdr.org                                    COMMENT             
DEFAULT MARKER NAME                                         MARKER NAME         
UNKNOWN USER        CTTC                                    OBSERVER / AGENCY   
GNSS-SDR            Software Receiver   0.0.10.gi           REC # / TYPE / VERS 
Antenna number      Antenna type                            ANT # / TYPE        
        0.0000        0.0000        0.0000                  APPROX POSITION XYZ 
        0.0000        0.0000        0.0000                  ANTENNA: DELTA H/E/N
G    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES 
E    4 C1B L1B D1B S1B                                      SYS / # / OBS TYPES 
DBHZ                                                        SIGNAL STRENGTH UNIT
  1999    09    18    11    33   59.3600000     GPS         TIME OF FIRST OBS   
                                                            END OF HEADER       
> 1999 09 18 11 34 00.0000000  0  5                                             
G326169692606.780 6     20155.190 6      -786.397 6        36.860               
G256171961609.741 6   -307204.118 6       908.087 6        38.829               
G126171877341.144 7    391181.957 7     -1256.853 7        43.850               
E246170935835.404 6   -175161.310 6       858.275 6        39.222               
E336167399218.321 6     69371.836 6     -1328.531 6        39.075               
> 1999 09 18 11 34 01.0000000  0  5                                             
G326169692458.360 5     20943.136 5      -792.375 5        35.520               
G256171961778.532 6   -308109.753 6       907.362 6        38.650               
G126171877101.091 7    392442.382 7     -1260.415 7        43.823               
E246170935999.263 6   -176021.051 6       860.349 6        39.224               
E336167398965.259 6     70700.859 6     -1329.253 6        39.042               
> 1999 09 18 11 34 02.0000000  0  5                                             
G326169692304.007 6     21731.579 6      -794.623 6        36.172               
G256171961954.363 6   -309015.154 6       910.907 6        38.692               
G126171876861.361 7    393703.019 7     -1262.242 7        43.887               
E246170936159.620 6   -176880.685 6       859.757 6        39.162               
E336167398708.553 6     72030.195 6     -1329.332 6        38.919               
> 1999 09 18 11 34 03.0000000  0  5                                             
G326169692161.198 5     22520.698 5      -786.482 5        34.630               
G256171962129.336 6   -309920.152 6       905.377 6        38.899               
G126171876626.204 7    394964.038 7     -1258.924 7        43.930               
E246170936321.799 6   -177740.017 6       858.456 6        39.135               
E336167398457.600 6     73360.087 6     -1330.777 6        38.772               
> 1999 09 18 11 34 04.0000000  0  5                                             
G326169692010.955 5     23311.186 5      -793.653 5        35.491               
G256171962301.415 6   -310824.089 6       909.052 6        38.731               
G126171876386.616 7    396226.080 7     -1261.334 7        44.059               
E246170936488.088 6   -178598.437 6       857.122 6        39.119               
E336167398205.862 6     74691.128 6     -1332.407 6        38.849               
> 1999 09 18 11 34 05.0000000  0  5                                             
G326169691860.355 5     24102.758 5      -803.184 5        35.842               
G256171962472.028 6   -311727.152 6       908.131 6        38.682               
G126171876145.349 7    397488.925 7     -1262.934 7        44.030               
E246170936652.126 6   -179456.116 6       857.116 6        39.234               
E336167397951.192 6     76023.117 6     -1333.677 6        38.912               
> 1999 09 18 11 34 06.0000000  0  5                                             
G326169691705.538 6     24894.915 6      -787.651 6        39.302               
G256171962639.818 6   -312629.917 6       908.190 6        38.873               
G126171875903.295 7    398752.062 7     -1262.345 7        44.195               
[... 228 lines cut ...]
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.output_rate_ms=100;
PVT.display_rate_ms=500;
PVT.dump=false
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump_filename=./PVT
PVT.rinex_version=2

;######### MONITOR CONFIG ############
Monitor.enable_monitor=true
Monitor.decimation_factor=1
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1234

In my honest opinion could be some issue with PVT block because today, during other tests, gnss-sdr wasn't able to wrote RINEX files at all, obviously i got several fixes.
If you need other infos don't esitate to ask.

Thank you again 

MC

acebrianjuan, 2019-05-06:
> gnss-sdr version 0.0.10.git-next-8799098

Ha! I just figured you are one commit behind https://github.com/gnss-sdr/gnss-sdr/commit/4bb1872e17a310e591ce32ed26031c12e3872582 _"Fix date in RINEX annotations beyond the rollover"_, which @carlesfernandez pushed on April 29th to fix the issue.

I suggest that you pull the latest changes of the [`next`](https://github.com/gnss-sdr/gnss-sdr/tree/next) branch and re-build GNSS-SDR.

The problem with the dates should be fixed.

Álvaro

matteocutugno, 2019-05-07:
Hello, 
I upgraded and rebuilded gnss-sdr and now it writes correct RINEX. 
Thanks a lot for your help

carlesfernandez, 2019-05-10:
There still are some annotations in RINEX files related to time which are not correct (specifically, annotations around the TOW rollover of a Saturday midnight, and also in annotations in the navigation header files related to the week number). Working on it.

carlesfernandez, 2019-08-09:
We believe this issue is solved in new release v0.0.11. Please feel free to re-open this issue or to open a new one if the problem persists.

closed 2019-08-09 by carlesfernandez