gnss-sdr/gnss-sdr #273: Sudden loss while tracking

matteocutugno, 2019-05-09:
Hello,

i'm able to get fixed position in real time with bladeRF x40 for GPS and Galileo in L1 band.
Everything is updated.
bladeRF-cli version: 1.7.1-git-1da130c
libbladeRF version: 2.2.0-git-1da130c
gnss-sdr version 0.0.10.git-next
ubuntu 16.04LTS on a Intel® Core™ i7-8550U CPU @ 1.80GHz × 8

The only concern is about the fact that even if i'm tracking 4, 5 or 6 satellites after some minutes of fixed positioning (8-10 minutes) the front-end losses all the satellites at the same time. After that it began to acquire again. 
Can someone explain why it has this behavior?

Thank you all

MC

shuwang1, 2019-06-18:
I know RTL-SDR has a "maybe questionable" benchmark tool called `rtl_test`, which can be used to show you the instant and cumulative PPM of your RTL-SDR dongle.  

```
shu@sdr-ubuntu-01:~$ rtl_test -p
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
Reporting PPM error measurement every 10 seconds...
Press ^C after a few minutes.
Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 32 bytes
real sample rate: 2047970 current PPM: -15 cumulative PPM: -15
real sample rate: 2047936 current PPM: -31 cumulative PPM: -23
real sample rate: 2047982 current PPM: -8 cumulative PPM: -18
real sample rate: 2047958 current PPM: -20 cumulative PPM: -18
real sample rate: 2047975 current PPM: -12 cumulative PPM: -17
real sample rate: 2047891 current PPM: -53 cumulative PPM: -23
real sample rate: 2048046 current PPM: 23 cumulative PPM: -17
real sample rate: 2047965 current PPM: -17 cumulative PPM: -17
real sample rate: 2047900 current PPM: -49 cumulative PPM: -20
real sample rate: 2047979 current PPM: -10 cumulative PPM: -19
real sample rate: 2047946 current PPM: -26 cumulative PPM: -20
real sample rate: 2048001 current PPM: 1 cumulative PPM: -18
real sample rate: 2048047 current PPM: 23 cumulative PPM: -15
real sample rate: 2047907 current PPM: -45 cumulative PPM: -17
real sample rate: 2047961 current PPM: -19 cumulative PPM: -17
real sample rate: 2047977 current PPM: -11 cumulative PPM: -17
real sample rate: 2047945 current PPM: -27 cumulative PPM: -17
real sample rate: 2048090 current PPM: 44 cumulative PPM: -14
real sample rate: 2047958 current PPM: -20 cumulative PPM: -14
real sample rate: 2047935 current PPM: -32 cumulative PPM: -15
real sample rate: 2048053 current PPM: 26 cumulative PPM: -13

```
I don't have BladeRF.  Do you have a tool similar to the "rtl_test" describe above?

Shu

matteocutugno, 2019-07-11:
Hi @shuwang1,
first of all thank you for your answer.
I don't know if there are such tests for bladerf?
Anyway i agree with you. 
Maybe is a TCXO working temperature issue. 
I don't know how to solve it.
Matteo

closed 2020-09-18 by matteocutugno