gnss-sdr/gnss-sdr #282: compile error, cmake -DENABLE_CUDA=ON -DENABLE_OSMOSDR=ON ../

shuwang1, 2019-06-17:
after I did 
```
git checkout next
cmake -DENABLE_CUDA=ON -DENABLE_OSMOSDR=ON ../
make

```
I received errors like 

```
[ 25%] Built target tracking_libs
[ 25%] Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/gps_l1_ca_dll_pll_tracking_gpu_cc.cc.o
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc: In member function ‘virtual int Gps_L1_Ca_Dll_Pll_Tracking_GPU_cc::general_work(int, gr_vector_int&, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:527:40: error: ‘d_carr_phase_error_secs_Ti’ was not declared in this scope
                     tmp_float = 1.0 / (d_carr_phase_error_secs_Ti * CURRENT_INTEGRATION_TIME_S);
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:527:40: note: suggested alternative: ‘carr_phase_error_secs_Ti’
                     tmp_float = 1.0 / (d_carr_phase_error_secs_Ti * CURRENT_INTEGRATION_TIME_S);
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                        carr_phase_error_secs_Ti
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:529:40: error: ‘d_code_error_filt_chips_Ti’ was not declared in this scope
                     tmp_float = 1.0 / (d_code_error_filt_chips_Ti * CURRENT_INTEGRATION_TIME_S);
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:529:40: note: suggested alternative: ‘code_error_filt_chips’
                     tmp_float = 1.0 / (d_code_error_filt_chips_Ti * CURRENT_INTEGRATION_TIME_S);
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                        code_error_filt_chips
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:532:33: error: ‘d_code_error_chips_Ti’ was not declared in this scope
                     tmp_float = d_code_error_chips_Ti * CURRENT_INTEGRATION_TIME_S;
                                 ^~~~~~~~~~~~~~~~~~~~~
/home/nvidia/Documents/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:532:33: note: suggested alternative: ‘code_error_chips_Ti’
                     tmp_float = d_code_error_chips_Ti * CURRENT_INTEGRATION_TIME_S;
                                 ^~~~~~~~~~~~~~~~~~~~~
                                 code_error_chips_Ti
src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/build.make:302: recipe for target 'src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/gps_l1_ca_dll_pll_tracking_gpu_cc.cc.o' failed
make[2]: *** [src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/gps_l1_ca_dll_pll_tracking_gpu_cc.cc.o] Error 1
CMakeFiles/Makefile2:1990: recipe for target 'src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/all' failed
make[1]: *** [src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/all] Error 2
Makefile:140: recipe for target 
```

What should I do now?

TIA

carlesfernandez, 2019-07-03:
Could you pull from origin/next and try again?

shuwang1, 2019-07-06:
It works for me now. Thx!

closed 2019-08-17 by shuwang1