gnss-sdr/gnss-sdr #284: Does GNSS-SDR support coherent integration period more than 1ms?  And what about the non-coherent integration?

shuwang1, 2019-06-20:
I noticed since v0.0.10 release, GNSS-SDR supports Enabled extended coherent integration times.  How can I do that?  for example, if I am going to use coherent_integration_time_ms=5, besides the following configurations,

```
Acquisition_1C.coherent_integration_time_ms=5
Acquisition_1C.doppler_step=150

```
what else should I do?

And how can I configure the non-coherence integration counts?

AdityaSanthosh, 2020-03-31:
The Paper _Robust Signal Processing for GNSS_ quotes
> N is the number of samples used during the estimation
> process. The product, T c = N T s defines the coherent integration time.

So, I think it's possible to increase Coherent Integration Period by decreasing `InputFilter.sampling_frequency` and increasing the value of `InputFilter.length`.
Or Am I missing Something?

carlesfernandez, 2020-04-14:
Hi,

the integration time can be set in the Acquisition and Tracking parameters:

For instance, for GPS L1:

 * `Acquisition_1C.coherent_integration_time_ms` sets the coherent integration time. 
 * `Acquisition_1C.max_dwells` sets the maximum number of non-coherent integrations (of the duration set by `coherent_integration_time_ms`).

Setting a large integration coherent integration time in Acquisition implies longer FFTs, so it has an impact in the processing load. If you are using a high sample rate and a long coherent integration time, the lenght of the FFT will be huge, and so the computational load.

* `Tracking_1C.extend_correlation_symbols` sets the coherent integration time in the Tracking stage. This is applied after bit synchronization. Here, the limiting factor is the dynamics of your receiver. 

More info on the website.