gnss-sdr/gnss-sdr #289: Glonass telemetry decoder fails

wkazubski, 2019-07-09:
There is a problem wih a Glonass L1 telemetry decoder. It fails after few seconds:

wk@jawor:~/gnss-sdr> gnss-sdr --config_file=./conf/gnss-sdr_GLONASS_L1_CA_ibyte.conf
linux; GNU C++ version 9.1.1 20190520 [gcc-9-branch revision 271396]; Boost_106900; UHD_003.009.007-0-unknown

Initializing GNSS-SDR v0.0.10.git-- ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file ./data/NT1065_GLONASS_L1_20160923_fs6625e6_if0e3_schar.bin, which contains 4144207352 [bytes]
GNSS signal recorded time to be processed: 312.769 [s]
GNSS-SDR cannot create dump file for the Observables block. Wrong permissions?
GNSS-SDR cannot create dump file for the Acquisition block. Wrong permissions?
GNSS-SDR cannot create dump file for the Acquisition block. Wrong permissions?
GNSS-SDR cannot create dump file for the Acquisition block. Wrong permissions?
GNSS-SDR cannot create dump file for the Acquisition block. Wrong permissions?
GNSS-SDR cannot create dump file for the Acquisition block. Wrong permissions?
Current receiver time: 1 s
Tracking of GLONASS L1 C/A signal started on channel 0 for satellite Glonass PRN 24 (Block 2)
Tracking of GLONASS L1 C/A signal started on channel 1 for satellite Glonass PRN 01 (Block 1)
Tracking of GLONASS L1 C/A signal started on channel 2 for satellite Glonass PRN 02 (Block -4)
Tracking of GLONASS L1 C/A signal started on channel 3 for satellite Glonass PRN 20 (Block 2)
Tracking of GLONASS L1 C/A signal started on channel 4 for satellite Glonass PRN 21 (Block 4)
Loss of lock in channel 0!
Tracking of GLONASS L1 C/A signal started on channel 0 for satellite Glonass PRN 24 (Block 2)
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
thread[thread-per-block[21]: <block glonass_l1_ca_telemetry_decoder_gs (13)>]: circular_buffer
thread[thread-per-block[18]: <block glonass_l1_ca_telemetry_decoder_gs (17)>]: circular_buffer
thread[thread-per-block[15]: <block glonass_l1_ca_telemetry_decoder_gs (21)>]: circular_buffer
thread[thread-per-block[24]: <block glonass_l1_ca_telemetry_decoder_gs (9)>]: circular_buffer
thread[thread-per-block[12]: <block glonass_l1_ca_telemetry_decoder_gs (25)>]: circular_buffer

after this, the program does not process antyhing, just wait for "q".
The problem appeared after 0.0.10 release but before June 24.

carlesfernandez, 2019-07-09:
Hi @wkazubski,

thanks for spotting this. Could you pull from upstream next and try again? Commit  faeb99b should fix this bug.

wkazubski, 2019-07-10:
With this commit the  Glonass telemetry decoder works. Currently I am receiving messages from three satellites (half of the sky visible). Thanks!

commit faeb99bf (2019-07-09): Fix GLONASS L1 nav message decoding (fixes #289)
closed 2019-07-16 by carlesfernandez