gnss-sdr/gnss-sdr #295: Rapid Tracking and Loss of Lock

gooroo363, 2019-07-23:
Good Morning, 
Having gone through he previous related issues and coming up empty, I figured I'd post here to see if I'm doing anything wrong. 

**Operating Environment**
Free and clear view of sky

**Machine Specs:**
Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
32GB RAM

**Hackrf Info:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00674757

**Antenna Info:**
Mfg: PulseLarsen Antennas
Model: RAZ62311MM
Notes:
GNSS Beidou-GPS-Glonass
Frequency 1561.098±2.046,1575.42±1.023,1602.5625±4 MHz
VSWR 2:1
Nominal Impedance 50 ohms
Gain (Radiating element) 1 dBic +/- 1dB
Gain (LNA gain) 30 dB +/- 2 dB
Polarization RHCP
**Specs Sheet:**
https://productfinder.pulseeng.com/doc_type/WEB301/doc_num/RAZXXXXXMM/doc_part/RAZXXXXXMM.pdf

**Config File**
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=30
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
[... 155 lines cut ...]
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 10 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 20 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 11 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 20 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 21 (Block IIR)
qCurrent receiver time: 12 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 20 (Block IIR)

Quit keystroke order received, stopping GNSS-SDR !!
Stopping GNSS-SDR, please wait!
OOOOOOOOOOOTotal GNSS-SDR run time: 13.1425 [seconds]
GNSS-SDR program ended.
Stopping TCP/IP server on port 2101

**Notes**
If it runs long enough (10-15min), I am able to receive some sporadic signals, but they are not consistent.

Thoughts?

matteocutugno, 2019-07-23:
Hi,
Did you fed the antenna with a proper bias-t?

gooroo363, 2019-07-23:
I may be wrong as I'm still learning, but it was my understanding that the hackrf is able to do that once the proper setting was set "SignalSource.osmosdr_args=hackrf,bias=1"

gooroo363, 2019-07-23:
UPDATE: 
I've removed my previous install (installed via sudo apt-get install gnss-sdr) and build it via the Clone, build and install GNSS-SDR instructions...with the cmake -DENABLE_OSMOSDR=ON ../ option

The results seem better I'll get a series of tracking and loss of lock messages every 5-15 seconds vs the constant lock and loss of lock I had before. I am also now receiving "New GPS NAV messages"

matteocutugno, 2019-07-23:
Yes, but you have also to phisically connect a bias-tee, that is a simple three way component useful to fed the active antenna.
In the first port you connect the front end with a RF cable, in the second you connect a DC cable with a 5 volt input(USB port of a pc) and you get in output of the third port the sum RF+DC where you will connect the antenna.
Without this hardware configuration you will have a poor gain and it will be difficult to get a suitable number of NAV messages from different satellites.
MC

gooroo363, 2019-07-23:
Ahhhh ok gotcha...that's what I'm missing. Thanks, I'll report back....

odrisci, 2019-07-24:
If you check the documentation [here](https://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-osmosdr_signal_source) you'll see that setting the `bias=1` flag only has an effect if the version of `gr-osmosdr` that you are using is >= 0.1.4-13. If you are using an older version either upgrade it or use an external bias tee as suggested by @matteocutugno

gooroo363, 2019-07-24:
Thanks @matteocutugno and @odrisci. It is my understanding that I'm on a newest version, at least one greater than 0.1.4-13.

When running "sudo apt install gr-osmosdr"
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gr-osmosdr is already the newest version (0.1.4-14build7).

UPDATE:
After updating and running 'next' along with the supplied hackrf config, I'm now getting a few between 1-4 channel locks and New GPS NAV messages. After running long enough, (~20min) I had also started to receive positional information, that positional information was also sporadic.

luisrho, 2019-07-25:
Hi @gooroo363 ,

Since you are receiving NAV messages and obtaining PVT information it looks like is not a bias-tee issue.

It seems that your acquisition configuration is not quite accurate. The channels start tracking and loss the lock very easily. You should try to find a proper threshold or PFA (probability of false alarm) since it looks like you are obtaining many false alarms. A very low threshold (or high PFA) will declare a satellite present even when it is not there (I think it's your case). On the other hand, a very high threshold (or low PFA) will not declare a satellite present even being there.

You could also try to make narrower Doppler steps or apply a Doppler refinement (using "Acquisition_1C.make_two_steps=true") to obtain more accurate Doppler estimations for the tracking phase.

If the problem persists, you should verify if the receiver is trying to track the satellites in view or just random satellites. [Here](http://www.taroz.net/GNSS-Radar.html) you can check the GNSS satellites in view at your location.

Hope it helps!

gooroo363, 2019-07-25:
Thanks @luisrho. I'll give those settings a try. I also have another active antenna arriving in a few days to see if I can learn more about my hardware configuration. I also stumbled upon Trimble's GNSS planning tool. For those that are interested. https://www.gnssplanning.com/#/settings

gooroo363, 2019-07-25:
@luisrho Early testing looks like the hackrf is trying to track some random satellites :-/ I can't tell if the ones that are in view that are being tracked are just dumb luck or if some are actually getting through. I'll continue adjusting the gain and Acquisition settings and see how it goes. I've also verified that the USB3 ports are not providing "Supplemental charging power"

gooroo363, 2019-07-27:
UPDATE: 
Received and tested new active antenna, same results. Will continue to debug.

shuwang1, 2019-08-19:
@gooroo363 

I feel these configuration values below may not make much sense ...

Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_min=-10000
Acquisition_1C.doppler_step=500

If I remember this correctly, the doppler due to the satellites should be +/-2.7 ppm or so.  What is the ppm of your receiver clock?

In general, the search windows should be 

half search window = max satellite doppler + max receiver doppler + clock drifting ...

shuwang1, 2019-08-19:
@gooroo363

Also the step size of 500Hz may be too large ... what is your coherence integration size here? I see ... it is 1 ms.  As such the step size = 1 / 0.001 / 2 = 500 Hz.

I think you should set 

**Acquisition_1C.sampled_ms=2**

and then

Acquisition_1C.doppler_step=250 

to improve your doppler estimation

bpshop, 2019-12-29:
Hello

I have very similar problems using the Hack RF with an ublox multiband active GNSS antenna.
Did you manage to solve the problem?

Thank you,
Daniel.

AgileEngineeringLLC, 2020-05-27:
HI, I'm having the same issue with my HackRF One. It starts tracking on a channel, then loses it about 10 sec later. Has anyone creates a good conf file that works (better than the demo one on the main page)?
Thx!
Don

bpshop, 2020-05-27:
Hi Don,
I never managed to solve that issue. I would be interested as well in a work-around / solution to this.
I tried it on a virtual machine first, and then with a physical linux machine, but it did not work in both cases.

AgileEngineeringLLC, 2020-05-27:
Thx. I just get this repeated:

Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 22 min 31 s
Current receiver time: 22 min 32 s
Current receiver time: 22 min 33 s
Current receiver time: 22 min 34 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 23 (Block IIR)
Current receiver time: 22 min 35 s
Current receiver time: 22 min 36 s
Current receiver time: 22 min 37 s
Current receiver time: 22 min 38 s
Current receiver time: 22 min 39 s
Current receiver time: 22 min 40 s
Current receiver time: 22 min 41 s
Current receiver time: 22 min 42 s
Current receiver time: 22 min 43 s
Current receiver time: 22 min 44 s
Current receiver time: 22 min 45 s
Current receiver time: 22 min 46 s
Loss of lock in channel 0!

And I get this when I run the front-end-cal:
Searching for GPS Satellites in L1 band...
[ 1  2  3  4  5  6  7  8  9  10  11  12  13  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32 ]
Total signal acquisition run time 9.37991 [seconds]
Unable to get Ephemeris SUPL assistance. TOW is unknown!
GNSS-SDR Front-end calibration program ended.

So I'm guessing it's either the signal is too weak for the HackRF, or I need a higher quality external clock for it to lock, or maybe there are some settings I can improve to get lock.

Frustrating that I can't seem to find some good info on others trying and having success witht he HackRF and gnss-sdr.

Don

Abdob, 2020-05-30:
same issue here I've tried several other osmosdr receivers (limesdr, bladerf, and rtlsdr)

AgileEngineeringLLC, 2020-05-31:
Got both the hackrf and an rtl-sdr processing L1. Had to drop the gains to 30, 30, 22 for my particular antenna. Both lock in about 45 sec. Got the rtl-sdr also locking on L2, but its a bit sporatic. It would be nice if the software provided more clues as to whats going on, or whats happening in the acquisition or tracking phases so we’d know what parameters we might try to vary.

Abdob, 2020-05-31:
Can you please share what antenna and configuration file you are using and whether you have a biast, I see the author of this paper https://www.mdpi.com/2076-3263/10/2/79 managed to get it working but with a ridiculously expensive antenna “low cost solution”

AgileEngineeringLLC, 2020-05-31:
I’m enabling the power to the antenna via software (in the conf file). Two antenna work equally well for me, the ublox multiband antenna (about$60 from Sparkfun) or a high-gain base station antenna (probably about $400) that’s on my roof.

AgileEngineeringLLC, 2020-05-31:
The conf file i used is the example conf file for the hackrf that’s provided on the My First Fix page on the gnss-sdr org site. I just had to lower the three gains as i mention above. I think the original gains were too high and were saturating the front end. I also bought an Nooelec LNA for the antenna, but not thinking i need it. I also have A couple Cheap bandpass (L1, L2) filters Coming from etsy, so will try those as well when they arrive. So far, it seems the rtl-sdr v3 acquires quicker for me than the hackrf, and I can get the rtl-sdr v3 to also lock on L2 sporatically. Would be nice to view the fft of the exact gnss-sdr gain settings i’m using, but not seeing how to do that in gnuradio as the gains are named differently.

Abdob, 2020-05-31:
This config file shows the respective gains from gnss-sdr to osmosdr:
https://github.com/gnss-sdr/gnss-sdr/blob/master/conf/gnss-sdr_GPS_L1_bladeRF.conf

;# RF Gain: LNA Gain {0, 3, 6}
SignalSource.gain=6
;# IF Gain: N/A
SignalSource.rf_gain=40
;# BB Gain: RX VGA1 + VGA2 [5, 60]
SignalSource.if_gain=48

That would be nice to see what the spectrum looks like when the system is working, perhaps a waterfall plot. I will also be trying the ublox multiband antenna and two other antennas provided with the products below:

Assuming these are working systems we'd be able to compare the acquired signals from rtl-sdr against the ones here:

https://www.amazon.com/gp/product/B07TZVQDHQ/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

https://anavs.de/multi-sensor-rtk-module/

AgileEngineeringLLC, 2020-05-31:
The conf file you're referring to is for a BladeRF front end, and they likely are using a powered-T with a LNA amplifier, so thus the relatively low (6 db) gain level.

Abdob, 2020-05-31:
Yes the specific values are for bladerf and the ranges shown are the ones supported by the bladerf. But the naming of the gains apply to all osmosdr receivers. So in gnuradio using the osmsosdr sink block you would do RG Gain: 30 (lna) IF Gain : 30 (rf gain) BB Gain : 20 (if gain). To me it would make sense that the IF gain in osmo wouldn't apply because the signal would be at baseband for that same reason it could also make sense that the IF gain used in the config file is the same as the BB gain in osmo. Below is the documentation from the gnuradio source block.

On the gr-osmosdr source block documentation it has:
RF Gain:
Overall RF gain of the device.

IF Gain:
Overall intermediate frequency gain of the device.
This setting is available for RTL-SDR and OsmoSDR devices with E4000 tuners and HackRF in receive and transmit mode. Observations lead to a reasonable gain range from 15 to 30dB.

BB Gain:
Overall baseband gain of the device.
This setting is available for HackRF in receive mode. Observations lead to a reasonable gain range from 15 to 30dB.

AgileEngineeringLLC, 2020-05-31:
Thanks! That makes sense.

Abdob, 2020-06-07:
Ahh for me it was the antenna, below are the results I'm obtaining for with the rtl-sdr and this antenna:
https://www.tersus-gnss.com/product/antenna-ax3703
as you see multiple location and velocity estimates are provided every second. The ublox antenna also worked but not as frequent and with much larger variances in the estimates.

First position fix at 2020-Jun-07 21:28:30.100000 UTC is Lat = 33.4932 [deg], Long = -111.901 [deg], Height= 335.778 [m]
Position at 2020-Jun-07 21:28:30.500000 UTC using 4 observations is Lat = 33.492972657 [deg], Long = -111.901343224 [deg], Height = 323.726 [m]
Velocity: East: 0.019 [m/s], North: -0.009 [m/s], Up = -0.015 [m/s]
Position at 2020-Jun-07 21:28:31.000000 UTC using 4 observations is Lat = 33.492928189 [deg], Long = -111.901407454 [deg], Height = 311.270 [m]
Velocity: East: -0.160 [m/s], North: -0.274 [m/s], Up = -0.177 [m/s]
Current receiver time: 1 min 16 s
Loss of lock in channel 2!
Position at 2020-Jun-07 21:28:31.500000 UTC using 4 observations is Lat = 33.493143300 [deg], Long = -111.901250488 [deg], Height = 325.975 [m]
Velocity: East: -0.137 [m/s], North: -0.262 [m/s], Up = -0.204 [m/s]
Position at 2020-Jun-07 21:28:32.000000 UTC using 4 observations is Lat = 33.492882622 [deg], Long = -111.901437965 [deg], Height = 312.867 [m]
Velocity: East: 0.260 [m/s], North: 0.420 [m/s], Up = 0.353 [m/s]
Current receiver time: 1 min 17 s
Position at 2020-Jun-07 21:28:32.500000 UTC using 4 observations is Lat = 33.492754477 [deg], Long = -111.901515597 [deg], Height = 303.042 [m]
Velocity: East: 0.056 [m/s], North: 0.086 [m/s], Up = 0.100 [m/s]
Loss of lock in channel 5!
Position at 2020-Jun-07 21:28:33.000000 UTC using 4 observations is Lat = 33.492873955 [deg], Long = -111.901417349 [deg], Height = 317.635 [m]
Velocity: East: -0.146 [m/s], North: -0.279 [m/s], Up = -0.297 [m/s]
Current receiver time: 1 min 18 s
Position at 2020-Jun-07 21:28:33.500000 UTC using 4 observations is Lat = 33.492898695 [deg], Long = -111.901420803 [deg], Height = 308.615 [m]
Velocity: East: -0.009 [m/s], North: 0.006 [m/s], Up = -0.172 [m/s]
Position at 2020-Jun-07 21:28:34.000000 UTC using 4 observations is Lat = 33.492763118 [deg], Long = -111.901534764 [deg], Height = 289.685 [m]
Velocity: East: -0.113 [m/s], North: -0.214 [m/s], Up = -0.166 [m/s]
Current receiver time: 1 min 19 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 01 (Block IIF)
Position at 2020-Jun-07 21:28:34.500000 UTC using 4 observations is Lat = 33.492769029 [deg], Long = -111.901549834 [deg], Height = 301.402 [m]
Velocity: East: 0.164 [m/s], North: 0.220 [m/s], Up = 0.192 [m/s]
Position at 2020-Jun-07 21:28:35.000000 UTC using 4 observations is Lat = 33.492762633 [deg], Long = -111.901559209 [deg], Height = 303.480 [m]
Velocity: East: -0.041 [m/s], North: -0.093 [m/s], Up = -0.180 [m/s]
Current receiver time: 1 min 20 s
Position at 2020-Jun-07 21:28:35.500000 UTC using 4 observations is Lat = 33.492762038 [deg], Long = -111.901560661 [deg], Height = 291.675 [m]
Velocity: East: -0.164 [m/s], North: -0.260 [m/s], Up = -0.242 [m/s]
New GPS NAV message received in channel 4: subframe 4 from satellite GPS PRN 12 (Block IIR-M)
New GPS NAV message received in channel 0: subframe 4 from satellite GPS PRN 05 (Block IIR-M)
New GPS NAV message received in channel 3: subframe 4 from satellite GPS PRN 25 (Block IIF)
New GPS NAV message received in channel 7: subframe 4 from satellite GPS PRN 29 (Block IIR-M)
Position at 2020-Jun-07 21:28:36.000000 UTC using 4 observations is Lat = 33.492893764 [deg], Long = -111.901471985 [deg], Height = 300.273 [m]
Velocity: East: -0.288 [m/s], North: -0.546 [m/s], Up = -0.452 [m/s]
Current receiver time: 1 min 21 s
Loss of lock in channel 1!
Position at 2020-Jun-07 21:28:36.500000 UTC using 4 observations is Lat = 33.493170679 [deg], Long = -111.901252689 [deg], Height = 345.440 [m]
Velocity: East: 0.005 [m/s], North: 0.071 [m/s], Up = 0.017 [m/s]
Position at 2020-Jun-07 21:28:37.000000 UTC using 4 observations is Lat = 33.492930981 [deg], Long = -111.901431459 [deg], Height = 318.693 [m]
Velocity: East: 0.044 [m/s], North: 0.076 [m/s], Up = 0.000 [m/s]
Current receiver time: 1 min 22 s
Position at 2020-Jun-07 21:28:37.500000 UTC using 4 observations is Lat = 33.492945078 [deg], Long = -111.901410141 [deg], Height = 310.442 [m]
Velocity: East: 0.014 [m/s], North: 0.044 [m/s], Up = -0.067 [m/s]
Position at 2020-Jun-07 21:28:38.000000 UTC using 4 observations is Lat = 33.492685327 [deg], Long = -111.901601419 [deg], Height = 276.741 [m]
Velocity: East: 0.177 [m/s], North: 0.274 [m/s], Up = 0.236 [m/s]
Current receiver time: 1 min 23 s
Position at 2020-Jun-07 21:28:38.500000 UTC using 4 observations is Lat = 33.492664226 [deg], Long = -111.901605926 [deg], Height = 291.993 [m]
Velocity: East: 0.002 [m/s], North: -0.069 [m/s], Up = -0.107 [m/s]
Position at 2020-Jun-07 21:28:39.000000 UTC using 4 observations is Lat = 33.492635828 [deg], Long = -111.901595628 [deg], Height = 292.733 [m]
Velocity: East: 0.022 [m/s], North: 0.078 [m/s], Up = -0.103 [m/s]
Current receiver time: 1 min 24 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Position at 2020-Jun-07 21:28:39.500000 UTC using 4 observations is Lat = 33.492762380 [deg], Long = -111.901514818 [deg], Height = 303.636 [m]
Velocity: East: 0.165 [m/s], North: 0.272 [m/s], Up = 0.247 [m/s]
[... cut ...]
Velocity: East: 0.024 [m/s], North: 0.026 [m/s], Up = -0.032 [m/s]
Position at 2020-Jun-07 21:28:41.000000 UTC using 4 observations is Lat = 33.492775020 [deg], Long = -111.901509179 [deg], Height = 301.006 [m]
Velocity: East: 0.049 [m/s], North: 0.047 [m/s], Up = -0.019 [m/s]
Current receiver time: 1 min 26 s
Position at 2020-Jun-07 21:28:41.500000 UTC using 4 observations is Lat = 33.492901436 [deg], Long = -111.901399995 [deg], Height = 310.869 [m]
Velocity: East: 0.120 [m/s], North: 0.216 [m/s], Up = 0.126 [m/s]

AgileEngineeringLLC, 2020-06-07:
Looks like you're getting a good solution. I experimented around with the HackRf and was able to go up to 8MS/S and 12 sats tracked on my CPU. Have you tried tracking on L2 or E1? I tried both, but GNSS-SDR seemed to see the sats but kept losing them with out getting a PVT solution.

gooroo363, 2020-06-18:
@Abdob would mind sending over your conf file.....i still can't get it to work.....3 different machines and two different antennas.

AgileEngineeringLLC, 2020-06-18:
@gooroo363 I can send you one I use for HackRF and RTL-SDR front ends, but am out of pocket for a week. Please get back to me after Jun 28th and ping me if you still need it. Once HUGE factor is making sure you have a good oscillator. I like the $25 NooElec Tiny TCXO for the HackRF for example (or a high end GPSDO if I want high timing accuracy for test).

gooroo363, 2020-06-18:
sweet thanks @AgileEngineeringLLC. I've got both along with a limesdr usb, so that would help out a lot.

AgileEngineeringLLC, 2020-06-18:
@gooroo363 with accurate oscillator, i also can both L1-only, or E1-only (Galileo). E1 won't track for me unless I have solid time reference. Neither L1 or E1 track for me with the stock HackRF oscillator.

Abdob, 2020-06-21:
@gooroo363 i just got to pushing this onto my repo:
https://github.com/Abdob/gnss-sdr/blob/next/work/config_files/rtlsdr1.conf

I'm not aware I have tried on L2 or E1, I should look at those settings. What I have now is a non-moving antenna as soon as I move around with the car the tracking is lost. Looking at the signal strength through gnuradio, it is very weak compared to the example data from the capture in spain.

AgileEngineeringLLC, 2020-07-01:
Anyone have some hints on how to get the GNSS-SDR Monitor (https://github.com/acebrianjuan/gnss-sdr-monitor <https://github.com/acebrianjuan/gnss-sdr-monitor>) up and running.

Here’s the steps I’ve followed:
1) Built GNSS-SDR next branch (as required) 
2) Built GNSS-SDR Monitor (verified the GUI opens)
3) added PVT.enable_monitor=true to conf file
4) On the GNSS-SDR Monitor GUI, set Address to my internal Ubuntu PC IP address, and set port to 1234 (the default per the GNSS-SDR guide for monitoring)

I’ve also tried other IP addresses and ports but no luck.

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

https://github.com/Abdob/gnss-sdr/blob/next/work/config_files/rtlsdr1.conf <https://github.com/Abdob/gnss-sdr/blob/next/work/config_files/rtlsdr1.conf>

acebrianjuan, 2020-07-01:
Hi Don (@AgileEngineeringLLC),

> Anyone have some hints on how to get the GNSS-SDR Monitor (https://github.com/acebrianjuan/gnss-sdr-monitor <https://github.com/acebrianjuan/gnss-sdr-monitor>) up and running.

What problems are you experiencing?

This [short guide](https://github.com/acebrianjuan/gnss-sdr-monitor/issues/6#issuecomment-581124725) I wrote should help you with the setup.

Regards,
Álvaro

AgileEngineeringLLC, 2020-07-01:
Hi Alvaro,

Thanks for the guide, I’ll try that now!

The problem I’m having is that the GUI doesn’t see any data, so I must have the setup wrong. I will try your guide and let you know what happens. Thanks much.

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

AgileEngineeringLLC, 2020-07-01:
Hi Alvaro,

I followed your setup guide, and it seems like the GUI is connecting, but no data is flowing to the GUI. I took a screenshot below.

Don

Don Kelly

Agile Engineering

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
4403 Orange Leaf Court
Houston, TX   77059

acebrianjuan, 2020-07-01:
Hi Don,

> I followed your setup guide, and it seems like the GUI is connecting, but no data is flowing to the GUI. I took a screenshot below.

I don't see any screenshot attached to your last message.

Regardless, I have made a test run to verify that the GUI works with the latest version of GNSS-SDR (`next` branch), and I can confirm that everything works fine.

[image]

If your problems persist please open a new issue, either on the `gnss-sdr` issue tracker or on the `gnss-sdr-monitor` issue tracker, as the present issue has to do with HackRF.

Regards,
Álvaro

Bilelhm98, 2021-01-28:
Hi Don, 

Did you get to solve this problem? In case you did, could you explain me how?
I have the same problem than you had. I am using a good oscilator and the HackRF One, however it loses the channel constantly. It can keep the same channel for aproximately 20 or 30 seconds but then it losses again and again.

Best Regards,
Bilel.

AgileEngineeringLLC, 2021-01-28:
If you're referring to the gnss-sdr-monitor issues i had, you need to make sure you match up the conf file port settings and the port settings in the monitor menu. 

As far as hackrf settings, i haven't run one in quite a while, so i don't recall the exact settings. The default ones on the gnss-sdr site should be good enough. You might read some of my memos posted on my LinkedIn site for other suggestions. Make sure you have a good gps antenna and that biastee power is on. 

Don

Don Kelly
Technologist | Engineer
Agile Engineering, LLC

LinkedIn: www.linkedin.com/in/kellydak
Cell: 281-221-2853
4403 Orange Leaf Court
Houston, TX 77059

Bilelhm98, 2021-02-09:
I was talking about the configuration file. Thank you for your advice, I am gonna read your posts.