gnss-sdr/gnss-sdr #298: Failing to acquire Supl GPS data from supl.google.com

gooroo363, 2019-07-24:
Evening all, 
Just wanted to see if others have been experiencing an issue with the supl.google.com data pulls, or if it's just me and my firewall rules (which I thought I've properly set just in case its me).

carlesfernandez, 2019-07-30:
Hi,

please check the updated documentation at https://gnss-sdr.org/docs/sp-blocks/global-parameters/#assisted-gnss-via-supl-v10 for getting SUPL data from a local base station.

If supl.google.com does not work for you, maybe you can try supl.vodafone.com

Hope this helps.

gooroo363, 2019-07-30:
Much better! Thank you. I recommend adding a list of the possible supl services users can utilize. 
Open Cellid is a good resource to help identify MCCs, MNCs, LACs, and CIDs 
https://opencellid.org/#zoom=16&lat=37.77889&lon=-122.41942

closed 2019-07-30 by gooroo363