gnss-sdr/gnss-sdr #300: Wrong Position with SUPL and Acquisition Assistance

gooroo363, 2019-07-30:
Good Evening, 
I was able to get the Assisted GNSS via SUPL v1.0 working to download data via the supl.vodafone.com address. However, when utilizing it I get 

"Get visible satellites at 30/07/2019 23:35:09 UTC, assuming RX position 37.1965 [deg], 267.15 [deg], 0 [m]"

even though that is not my location. I can utilize the "GNSS-SDR.AGNSS_ref_location={lat},{lon}" however I want to use the supl data from vodafone, vs a local file. 

I know my supl MCC, MNC, LAC, and CI number are correct because I get the correct location when I run front-end-cal.

Thoughts?

carlesfernandez, 2019-08-02:
Hi @gooroo363,

The reference position is supplied by the "SUPL_gps_acquisition_server" but not by the "SUPL_gps_ephemeris_server". In some base stations, the later works but the former does not. In this case, you need to provide a rough initial location at "GNSS-SDR.AGNSS_ref_location={lat},{lon}".

However, in the message you provided it seems that you effectively got that position from the network. It should be the position of the base station you are connecting to.

This location is used to compute the visible satellites and search them in the first place, so it does not need to be very accurate. It is not used in the PVT solution in any way, it is just used to shorten the time to first fix.

I'm not understanding the part of " I want to use the supl data from vodafone, vs a local file. " AGNSS only makes sense in real-time, since base stations provide data from the present. If you apply AGNSS data retrieved right now when processing a recorded file, the list of visible satellites will be wrong, so it will be of little help. Maybe I didn't get your point.

gooroo363, 2019-08-02:
Thanks @carlesfernandez for the reply. My apologies for not being clearer. 

I'm trying to run gnss-sdr in real-time with a hackrf. The root of the problem is that I'm not getting the correct visible satellites for my location when I start the application, it actually seems to try and lock on random L1 satellites (see issue #295). That led me to utilize the SUPL configurations in the config file to help get an accurate list of the visible sats in my area st startup.

;NH                  
GNSS-SDR.init_latitude_deg=43.45908706
GNSS-SDR.init_longitude_deg=-71.5616262
GNSS-SDR.init_altitude_m=188

;######### SUPL RRLP GPS assistance configuration #####
; Check http://www.mcc-mnc.com/
; On Android: https://play.google.com/store/apps/details?id=net.its_here.cellidinfo&hl=en
GNSS-SDR.SUPL_gps_enabled=true
GNSS-SDR.SUPL_read_gps_assistance_xml=false
GNSS-SDR.SUPL_gps_ephemeris_server=supl.vodafone.com
GNSS-SDR.SUPL_gps_ephemeris_port=7275
GNSS-SDR.SUPL_gps_acquisition_server=supl.vodafone.com
GNSS-SDR.SUPL_gps_acquisition_port=7275
GNSS-SDR.SUPL_MCC=310
GNSS-SDR.SUPL_MNC=410
GNSS-SDR.SUPL_LAC=23999
GNSS-SDR.SUPL_CI=15498

I am able to successfully connect to the supl.vodafone servers. But you can see that the location its trying to get the visible satellites for is not the same location that I set in the "GNSS-SDR.init_lat/lon"
section of the config. 

------
Initializing GNSS-SDR v0.0.10.git-next-cdfe4c43d ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2018.01.1 
Enabled antenna bias voltage
Actual RX Rate: 2000000.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...
Actual RX Gain: 14.000000 dB...
Actual Bandwidth: 6000000.000000 [Hz]...
SUPL RRLP GPS assistance enabled!
SUPL: Try to read GPS ephemeris data from SUPL server...
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
SUPL: Received ephemeris data for satellite GPS PRN 01 (Block IIF)
SUPL: Received ephemeris data for satellite GPS PRN 06 (Block IIF)
SUPL: Received ephemeris data for satellite GPS PRN 07 (Block IIR-M)
SUPL: Received ephemeris data for satellite GPS PRN 11 (Block IIR)
SUPL: Received ephemeris data for satellite GPS PRN 13 (Block IIR)
SUPL: Received ephemeris data for satellite GPS PRN 15 (Block IIR-M)
SUPL: Received ephemeris data for satellite GPS PRN 17 (Block IIR-M)
SUPL: Received ephemeris data for satellite GPS PRN 18 (Block IIR)
SUPL: Received ephemeris data for satellite GPS PRN 19 (Block IIR)
SUPL: Received ephemeris data for satellite GPS PRN 22 (Block IIR)
SUPL: Received ephemeris data for satellite GPS PRN 28 (Block IIR)
SUPL: Received ephemeris data for satellite GPS PRN 30 (Block IIF)
SUPL: XML ephemeris data file created
SUPL: Try to read Almanac, Iono, Utc Model, Ref Time and Ref Location data from SUPL server...
Current receiver time: 6 s
SUPL: Received almanac data for satellite GPS PRN 01 (Block IIF)
SUPL: Received almanac data for satellite GPS PRN 02 (Block IIR)
SUPL: Received almanac data for satellite GPS PRN 03 (Block IIF)
SUPL: Received almanac data for satellite GPS PRN 05 (Block IIR-M)
SUPL: Received almanac data for satellite GPS PRN 06 (Block IIF)
SUPL: Received almanac data for satellite GPS PRN 07 (Block IIR-M)
SUPL: Received almanac data for satellite GPS PRN 08 (Block IIF)
[... 42 lines cut ...]
Using GPS Ephemeris: Sat 17 Az: 255.893 El: 55.5162
Using GPS Ephemeris: Sat 18 Az: 47.6184 El: 25.518
Using GPS Ephemeris: Sat 19 Az: 243.636 El: 36.1668
Using GPS Ephemeris: Sat 22 Az: 94.0608 El: 7.17041
Using GPS Ephemeris: Sat 28 Az: 358.315 El: 66.5763
Using GPS Ephemeris: Sat 30 Az: 175.366 El: 56.1721
Using GPS Almanac:  Sat 3 Az: 113.859 El: 2.51898
Using GPS Almanac:  Sat 8 Az: 56.0987 El: 4.38242
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 17 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 30 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 28 (Block IIR)
Current receiver time: 9 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 22 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 19 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 11 (Block IIR)
Current receiver time: 10 s
Current receiver time: 11 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 06 (Block IIF)
Current receiver time: 12 s
-------
According to Trimble's GNSS planner these are the sats I should be seeing

[image]

and after running it again just now the majority are in view and I'm getting PVT locations (hadn't happend), but there still are a few sats this its trying to find and lock on with that aren't in view, like Satellite 6.