gnss-sdr/gnss-sdr #303: Does someone have experienced this error before?

fabriciocscte, 2019-08-05:
Stopping GNSS-SDR, please wait! **[At this point everything happened as expected.]**
Stopping TCP/IP server on port 2101
Error in client
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  close: Bad file descriptor
Aborted (core dumped)

carlesfernandez, 2019-08-21:
This is a bug that appeared in old versions, but should be gone in v0.0.11. What version of gnss-sdr are you using? What is your OS?

In the meanwhile, if you are not using the RTCM streaming, you can deactivate the server with `PVT.flag_rtcm_server=false` in the configuration file, so this error will disappear.

fabriciocscte, 2019-08-21:
Thank you. I will update to the last version v0.0.11. As a matter of fact, this is not causing any problem to me, but it's always better to understand what's happening.