gnss-sdr/gnss-sdr #311: wrong calulation in beidou_b1i_telemetry_decoder_gs::decode_bch15_11_01

DavidRamati, 2019-09-16:
```
    err = errind[reg[0] + reg[1] * 2 + reg[2] * 4 + reg[3] * 8];

    if (err > 0 and err < 16)
        {
            decbits[err - 1] *= -1;
        }
```

this code is wrong for two reasons:
1) summing the registers should return an index between 0-15, but because in this stage of the code we use -1 to represents 0s this line `reg[0] + reg[1] * 2 + reg[2] * 4 + reg[3] * 8` can actually produce a negative index. the solution is to turn the -1's to 0's before this calculation, and after the loop, because using zeros in the loop will erase the registers.
2) using the index produced by summing the registers as index in errind is wrong, because if the index is 0 then there is no error at all, so if (index > 0) then when should use it -1 as index in errind.
a corrected code should look something like this:
```
    for (int i = 0; i < 4; ++i)
    {
        reg[i] = (reg[i] + 1) / 2;
    }
    err = reg[0] + reg[1] * 2 + reg[2] * 4 + reg[3] * 8;
    if (err > 0 and err < 16)
    {
        decbits[errind[err - 1]] *= -1;
    }
```
P.S i tested the current code and my fixed code. the current code doesn't work while the corrected one does. my test look something like this:
```
    int32_t realData[15] = {1, 1, 1, -1, -1, -1, 1, -1, -1, -1, 1, 1, 1, -1, 1};
    for (int errorPos = 0; errorPos < 16; ++errorPos)
    {
        int32_t realDataWithError[15] = {0};
        for (int i = 0; i < 15; ++i)
        {
            realDataWithError[i] = realData[i];
        }
        if (errorPos < 15)
        {
            realDataWithError[errorPos] *= -1;
        }
        std::array<int32_t, 15> decodedData = {};
        decode_bch15_11_01(realDataWithError, decodedData);
        for (int i = 0; i < 15; ++i)
        {
            ASSERT_EQ(realData[i], decodedData[i]);
        }
    }
```
the realData was taken from a real BeiDou transmittion

carlesfernandez, 2019-09-17:
Hi,

I applied those changes in our CI system (see this [commit](https://gitlab.com/gnss-sdr/gnss-sdr/commit/553df7db6aa47edcf0a06bfa139749a748f6bb67), with these [results](https://gitlab.com/gnss-sdr/gnss-sdr/pipelines/82763130)), and unfortunately the BeiDou tests are failing.

@dmiralles2009 Could you please take a look at it?

dmiralles2009, 2019-09-18:
Hi @carlesfernandez, absolutely. I will look at this soon, probably immediately after ION. I have a couple of talks there and I am getting ready for those.

carlesfernandez, 2019-09-18:
@dmiralles2009 Enjoy the ION Conference!

DavidRamati, 2019-09-18:
i forgot to write that i also changed the initial values of the registers to -1 that represents 0s instead of 1's. according to the beidou icd.

> The decoding logic of BCH(15,11,1) is shown in Figure 5-4. The initial
> states of the four register stages are all zeros

@carlesfernandez do you mind changing it and run the tests again? (i don't no where those tests are located...)

carlesfernandez, 2019-09-18:
@DavidRamati yes, that did the trick. Now the test are [passing](https://gitlab.com/gnss-sdr/gnss-sdr/pipelines/82984515)

I'm wondering why the current code was working with our test signals (not sure about how they were generated, that's why I was asking @dmiralles2009 ) but not in your real-life captured Beidou data, and this change works with both. Anyway, your proposal makes sense to me, so I'm merging that into the `next` branch.

Thank you very much for your contribution.

DavidRamati, 2019-09-18:
no problem.
regarding your wondering, i'll try to explain.

first of all the last version worked just as well with my real-life captured data, if by worked we mean that the code was able to provide correct position.

the reason for this is that the bch encoding does not touch the data bits, so if the data that we receive is not corrupted (which i assume is the most common case) even if we don't use the bch decoder at all, we will still get the correct position.

assuming that most of the times the data we receive is not corrupted, then the error index we will get from the registers will be 0, and in the last version because of the bugs the error index will be -15. when we take the data of index -15 in the errind array and stores it in err, we get an undefined value. but we dont use this undefined value unless it satisf is the (err > 0 and err < 16) condition, and the chances for that are really low.

the problem with the last version starts when we get corrupted data. in this case half of the times the code will "fix" the wrong bit resulting in 2 wrong bits instead of 1, and in the other half the code will just ignore the wrong bit, still resulting in wrong data bit.

in the unit test i conducted, i took the real data and changed 1 of it's bits to check if the decoder fixes it, this is why the last version didn't pass my test while it passed the general test of finding position.

carlesfernandez, 2019-09-18:
Fully understood. Thanks again.

commit 3b86a8f0 (2019-09-18): Merge branch 'fix-beidou' into next (Fixes: #311)
closed 2019-12-18 by DavidRamati