gnss-sdr/gnss-sdr #314: Does GNSS-SNR estimate the fractional part of carrier phase?

xzbill, 2019-09-29:
I found in Tracking Block that the initial accumulated carrier phase is 0 (https://github.com/gnss-sdr/gnss-sdr/blob/v0.0.11/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L712). But according to GPS thesis, there is a fractional part in the initial accumulated phase.

carlesfernandez, 2019-10-01:
The accumulated phase in the Tracking block needs to start at some given value when no prior information about it is available. You need to start accumulating the phase evolution from some value, and 0.0 seems a fair place to start. Please note that the variable you are pointing at is a double-precision floating point value, in units of radians, so we have pretty nice precision there (52 bits for the fractional part!).

Could you please provide a reference of the GPS thesis you mention, and where that initial fractional part comes from?

ValZapod, 2020-02-29:
@xzbill What thesis?