gnss-sdr/gnss-sdr #318: Wrong Galileo time in PVT solution

sergey-nik, 2019-10-09:
Hi,
When processing Galileo E1 signal data, the PVT block gives wrong time, however the Observables time is correct.
I'm now testing with revision 9f8dbb14 (08/10/2019 04:41 PM) and the same effect was present for the revision cdd490e6 (23/09/2019 06:19 PM) that I previously used.

The issue appears as the following:
For the signal recorded 2019-Sept-27 approximately at 10:10 UTC I can see the same date and time in RINEX output. Hovever in NMEA, GPX, KML and in console messages I see the date-time as 2000-Feb-11 10:10.
Similar situation for the data recorded 2019-Oct-08 12:06 UTC. The RINEX data is timestamped with correct time, but all PVT output is timestamped as 2000-Feb-22 12:06.
Regardless of this time shift, the position computed by PVT is correct.

My PVT configuration is quite simple:

```
PVT.implementation=RTKLIB_PVT
PVT.averaging_depth=100
PVT.flag_averaging=true
PVT.output_rate_ms=10
PVT.display_rate_ms=500
PVT.elevation_mask=10
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
```

Regards,
Sergey.

sergey-nik, 2019-10-10:
I think the cause may be in conversion from GNSS-SDR ephemeris to RTKLIB structure: source files `src/algorithms/libs/rtklib/rtklib_conversions.cc` and `src/algorithms/libs/rtklib/rtklib_rtkcmn.cc`.

carlesfernandez, 2019-11-04:
Thanks @sergey-nik for your report and hints. You were right, there was a bug in the conversion of data to the RTKLIB land, which was causing a wrong Galileo date of the PVT solutions when using signals more recent than the GPS rollover on April 6, 2019. We think it's now fixed in the `next` branch, specifically in  commit 34d203a1383a144b0f8a71d2b54a1c32c5db56e8.

carlesfernandez, 2020-05-16:
Cloisng the issue since it seems fixed.

closed 2020-05-16 by carlesfernandez