gnss-sdr/gnss-sdr #326: Noise level analysis on GPS L1C/A code and phase

valcesch, 2019-11-04:
Dear Developers, 

We are using GNSS-SDR with a USRP X310 operated in common clock and common antenna with a geodetic receiver. Antenna is geodetic grade and external clock is a frequency standard.

We then do some code/phase noise comparisons between the hardware receiver and the software receiver:

Hardware receiver:
[image]

	Multipath STD: 0.07 m, Phase double difference STD: 0.0007 m/s^2

Software receiver:
[image]

	Multipath STD: 5.6 m, Phase double difference STD: 0.7 m/s^2

Accordingly to these results, code noise is around 10x larger on GNSS-SDR than the hardware receiver, and for phase, noise level is around 1000x larger. **These plots are based on RINEX observation data for both receivers**.

Here are some configuration parameters: 

	Sample duration: 30 minutes
	Signal: GPS L1C/A, PRN 26
	Hardware: USRPX310 with UBX daughter board
	gnss-sdr version: 0.0.11 (master branch)
	RINEX sampling rate: 1 second

And here is the configuration file:
	[GNSS-SDR]

	;######### GLOBAL OPTIONS ##################
	GNSS-SDR.internal_fs_sps=2000000

	;######### SIGNAL_SOURCE CONFIG ############
	SignalSource.implementation=File_Signal_Source
	SignalSource.filename=/mnt/local-ssd/N210_L1_2MHz.dat
	SignalSource.item_type=gr_complex
	SignalSource.sampling_frequency=2000000
	SignalSource.enable_throttle_control=false

	;######### SIGNAL_CONDITIONER CONFIG ############
	SignalConditioner.implementation=Pass_Through

	;######### CHANNELS GLOBAL CONFIG ############
	Channels_1C.count=15
	Channels.in_acquisition=1
	Channel.signal=1C

	;######### ACQUISITION GLOBAL CONFIG ############
	Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
	Acquisition_1C.item_type=gr_complex
	Acquisition_1C.threshold=0.015
	Acquisition_1C.doppler_max=5000
	Acquisition_1C.doppler_step=250

	;######### TRACKING GLOBAL CONFIG ############
	Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
	Tracking_1C.item_type=gr_complex
	Tracking_1C.pll_bw_hz=40.0;
	Tracking_1C.dll_bw_hz=4.0;

	;######### TELEMETRY DECODER GPS CONFIG ############
	TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

	;######### OBSERVABLES CONFIG ############
	Observables.implementation=Hybrid_Observables

	;######### PVT CONFIG ############
	PVT.rinex_output_enabled=true
	PVT.rinexobs_rate_ms=1000
	PVT.nmea_output_file_enabled=false
	PVT.output_path=gnss_products_n210_2MHz_3/others
	PVT.rinex_output_path=gnss_products_n210_2MHz_3/rinex
	PVT.rinex_version=3

We have done a similar analysis on phase data extracted from the same raw IQ sample with the following code: https://github.com/pmonta/GNSS-DSP-tools, and the phase noise analysis is provided below (double difference of phase data):

[image]

	Phase double difference STD: ~0.003m/s^2

Have you already noted such noise level on GNSS-SDR observation RINEX ? Is this related to an improper tuning of GNSS-SDR configuration parameters ? Or is there an issue while recording data in the RINEX observation file (clock jitter ? )

Thanks in advance for your answers and best regards

carlesfernandez, 2019-11-26:
While definitely you can improve things playing around with the Tracking block configuration (using longer integration times, a narrow correlator and narrower loop filters after bit synchronization), those results indicate that there might be something wrong in the processing chain. We have confirmed that in other similar double-difference tests and we are tracking down what could be the root cause. Thank you very much for this awesome analysis.

Please check #339 for an explanation of the `PVT.enable_rx_clock_correction` configuration parameter. In your setup, maybe it would make sense to set it to `false`.

valcesch, 2019-11-27:
Thanks for your answer. 

We were also investigating and indeed, with a little bit of tuning, we have managed to obtain some very good tracking results:

[image]

In order to obtain such results, we also had to replace the following line in rtklib_pvt_gs.cc: 

	double Rx_clock_offset_s = d_internal_pvt_solver->get_time_offset_s();

with : 

	double Rx_clock_offset_s  = 0.0;

Nevertheless, in that case we obtain unrealistic pseudoranges which prevent the computation of the PVT solution.

Hope this will help.

Arribas, 2019-11-30:
Dear Valcesch,
Thank you again for doing such detailed debug. I would like to clarify a bit the mechanism of rx clock correction implemented currently in gnss-sdr (next):

1. RX clock offset is computed on every PVT epoch by the "internal" pvt solver, which is nothing more than an RTKLIB solver automatically configured to be as simple as possible, that is, single solution and default parameters. It is possible to see the config parameters on the rtklib_pvt_gs.cc constructor, last lines.

2. Either the rx_clock_correction option is enabled or disabled, the PVT block sends an rx clock correction message to the Observables block if the rx clock offset is greater than a certain threshold (default is 40 ms in Pvt_conf() constructor):

 _double Rx_clock_offset_s = d_internal_pvt_solver->get_time_offset_s();
      if (fabs(Rx_clock_offset_s) * 1000.0 > max_obs_block_rx_clock_offset_ms)
           {
              if (!d_waiting_obs_block_rx_clock_offset_correction_msg)
                   {
                       this->message_port_pub(pmt::mp("pvt_to_observables"), pmt::make_any(Rx_clock_offset_s));
                       d_waiting_obs_block_rx_clock_offset_correction_msg = true;
                        LOG(INFO) << "Sent clock offset correction to observables: " << Rx_clock_offset_s << "[s]";
                    }
             }_

3. If the RX clock correction is enabled in config file (PVT.enable_rx_clock_correction=true, also is default to true), then, in addition to the "gross" correction made by PVT->Observables message, every epoch the RINEX observables are corrected by the computed clock correction value 
_( apply_rx_clock_offset(gnss_observables_map, Rx_clock_offset_s)._

**The noise you discover in the observables could be produced by a noisy estimation of the RX clock from the internal PVT solver.** In principle, if you set PVT.enable_rx_clock_correction=false in your config file, then the receiver, a part from the initial correction, will make only corrections to the RX clock when the Rx_clock_offset_s variable reach 40 ms or greater, which, for instance, using a 0.5 ppm clock, it takes several hours to drift from the initial correction to 40 ms...

I hope this helps!

Best regards.

carlesfernandez, 2020-01-22:
Dear @valcesch,

we have fixed the behavior of the software receiver when setting `PVT.enable_rx_clock_correction=false` in the next branch. You should now obtain usable pseudoranges and get PVT solutions from them when the clock steering is deactivated, which should lead to cleaner (that is, less noisy) observables.

Rango-T10000, 2024-12-14:
Hello, I have the similar setup with you. However, I cannot receive the signal.
Can you give me some advice? This is my issure for the detail: https://github.com/gnss-sdr/gnss-sdr/issues/831