gnss-sdr/gnss-sdr #338: loss of lock even CN0 is larger than 40

yealwang, 2019-11-20:
Hello
I have use GNSSSDR to process the data collected by a stereo front-end.
L1  IF=6.5MHz   int8 I
L2  IF=0.0MHz  int8 IQ (photo)
I process them separate the config as as bellow
The problem is for the L1 and L2 signal, GNSSSDR can not tracking stable, it always loss of lock.
There is no NAV data output.
Is there something wrong with my configs?
Can you give me some help?

Best wilshes

Yeal
[image]

L1 
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=6500000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Nsr_File_Signal_Source
SignalSource.filename=/home/../Data/L1L2082801_L1.dat   ; <- PUT YOUR FILE HERE
SignalSource.item_type=byte
SignalSource.sampling_frequency=26000000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=../data/signal_source.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=float

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat
InputFilter.input_item_type=float
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=4;5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.4
InputFilter.band2_begin=0.6
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=26000000
InputFilter.IF=6500000
InputFilter.decimation_factor=4

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
Resampler.implementation=Pass_Through
Resampler.dump=false
Resampler.dump_filename=../data/resampler.dat
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=10
Channels.in_acquisition=1
Channels_1C.Satellite=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############

Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.threshold=0.005
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.bit_transition_flag=false
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=35.0;
Tracking_1C.dll_bw_hz=2;
Tracking_1C.order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.dump=false
Tracking_1C.dump_filename=../data/epl_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.averaging_depth=100
PVT.flag_averaging=true
PVT.output_rate_ms=10
PVT.display_rate_ms=500

L2

[GNSS-SDR]
;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=26000000

;######### SIGNAL_SOURCE CONFIG ############
# SignalSource.implementation=File_Signal_Source

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/../Data/l1l2081801_l2.dat  ; <- PUT YOUR FILE HERE
SignalSource.item_type=ibyte
SignalSource.sampling_frequency=26000000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=../data/signal_source.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Ibyte_To_Complex

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Pass_Through
# InputFilter.implementation=Freq_Xlating_Fir_Filter
# InputFilter.dump=false
# InputFilter.dump_filename=../data/input_filter.dat
# InputFilter.input_item_type=gr_complex
# InputFilter.output_item_type=gr_complex
# InputFilter.taps_item_type=float
# InputFilter.number_of_taps=4;5
# InputFilter.number_of_bands=2
# InputFilter.band1_begin=0.0
# InputFilter.band1_end=0.4
# InputFilter.band2_begin=0.6
# InputFilter.band2_end=1.0
# InputFilter.ampl1_begin=1.0
# InputFilter.ampl1_end=1.0
# InputFilter.ampl2_begin=0.0
# InputFilter.ampl2_end=0.0
# InputFilter.band1_error=1.0
# InputFilter.band2_error=1.0
# InputFilter.filter_type=bandpass
# InputFilter.grid_density=16
# InputFilter.sampling_frequency=26000000
# InputFilter.IF=6500000
# InputFilter.decimation_factor=4

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
Resampler.dump=false
Resampler.dump_filename=../data/resampler.dat
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_2S.count=10
Channels.in_acquisition=1
Channels_2S.Satellite=1
Channel.signal=2S

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_2S.implementation=GPS_L2_M_PCPS_Acquisition
Acquisition_2S.item_type=gr_complex
Acquisition_2S.coherent_integration_time_ms=4
Acquisition_2S.threshold=0.00074
Acquisition_2S.doppler_max=5000
Acquisition_2S.doppler_step=60
Acquisition_2S.bit_transition_flag=false
Acquisition_2S.max_dwells=1
Acquisition_2S.dump=false
Acquisition_2S.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_2S.implementation=GPS_L2_M_DLL_PLL_Tracking
Tracking_2S.item_type=gr_complex
Tracking_2S.pll_bw_hz=2.0;
Tracking_2S.dll_bw_hz=0.25;
Tracking_2S.order=2;
Tracking_2S.early_late_space_chips=0.5;
Tracking_2S.dump=false
Tracking_2S.dump_filename=../data/epl_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_2S.implementation=GPS_L2C_Telemetry_Decoder
TelemetryDecoder_2S.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.averaging_depth=100
PVT.flag_averaging=true
PVT.output_rate_ms=10
PVT.display_rate_ms=500

shuwang1, 2019-11-26:
How many satellites are you tracking at the same time? And what kind of workstation are you using for the GNSS-SDR?  I guess, it should be a good idea to try a more powerful one instead?

carlesfernandez, 2019-11-26:
Dear @yealwang ,

please note that if your Signal Source is delivering samples in `byte` format, the input data type in the next downstream block (in this case, the DataTypeAdapter) cannot be a `float`. Please check https://gnss-sdr.org/docs/tutorials/understanding-data-types/ for a guide on how to configure the Signal Conditioner appropriately.

yealwang, 2019-11-27:
Hi Carlesfernandez
Thank you for your reply, yeah, the question is the input data type should not be float. I have fix it ,and work out the L1 and L2 dual-frequency rinex data.

closed 2019-11-27 by yealwang