gnss-sdr/gnss-sdr #341: File_Signal_Source W/ data type short (Real- Valued Data )

noelpedro, 2019-11-25:
Hi I'm getting an error when attempting to process a real-only data file.

Here are my configurations.
In Signal Source I have data type: short.
Signal Conditioner: Signal Conditioner
data type adapter Pass_Through,
Input_Filter: Freq_Xlating_Fir_Filter W/input data type short and output data type gr_complex
resampler: Direct_Resampler: data_type gr_complex

Error; "item mismatch: copy0: using 8, freq_xlating_fir_filter_scf0:0 using 2."

If I use cshort instead of gr_complex I get a segfault instead of the std err message above.

Does anyone have experience using gnss-sdr on real valued data files? If so, can you help me get around this error?

Thanks,

noelpedro, 2019-11-26:
Solved:

DataTypeAdapter.item_type=short

closed 2019-11-26 by noelpedro

carlesfernandez, 2019-11-26:
Glad you found the solution. The table at the end of https://gnss-sdr.org/docs/tutorials/understanding-data-types/ can help you on that matter.