gnss-sdr/gnss-sdr #347: Bug in GLONASS time year

DavidRamati, 2019-12-08:
when computing the year of glonass time in the code in the following functions:
`TEST(GlonassGnavEphemerisTest, ComputeGlonassTime)` in glonass_gnav_ephemeris_test.cc,
`Rinex_Printer::compute_UTC_time(const Glonass_Gnav_Ephemeris& eph, const double obs_time)` and `Rinex_Printer::get_leap_second(const Glonass_Gnav_Ephemeris& eph, const double gps_obs_time)` in rinex_printer.cc,
`Glonass_Gnav_Ephemeris::glot_to_utc`, `Glonass_Gnav_Ephemeris::glot_to_gpst` and `Glonass_Gnav_Ephemeris::compute_GLONASS_time` in glonass_gnav_ephemeris.cc, (bad code duplication...)
you use the following code:
```
boost::gregorian::date d1(d_yr, 1, 1);
boost::gregorian::days d2(d_N_T - 1);
boost::posix_time::ptime glonass_time(d1 + d2, t);
```
this code add d_N_T number of days since the beginning of the last leap year, to beginning of d_yr to get current year and day. this code right if d_yr was the last leap year, but d_yr is actually the current year. for example today's date 2019-12-8 the will be calculated as 2022-12-8 instead because the code will add the 3 years since 2016 twice, first time in d_yr, and second time in d_N_T.

the quickest way to fix it is to change the code that initialize d_yr in glonass_gnav_navigation_message.cc:
`gnav_ephemeris.d_yr = 1996 + 4.0 * (gnav_utc_model.d_N_4 - 1.0) + (J - 1.0);`
just remove the  `+ (J - 1.0)` part...

ValZapod, 2020-02-29:
No, better change the other place. I mean http://russianspacesystems.ru/wp-content/uploads/2016/08/ICD_GLONASS_eng_v5.1.pdf on page 58 there is this + (J - 1.0); part, so it is right; please use 
`boost::gregorian::date d1(d_yr- J + 1.0, 1, 1);`

Please note you better use russian version, I already found 2 typos while browsing english doc. ("Current day and month (dd/mm)" should be "hh/mm" and 1006 should be 1996, all on A.3.1.3 about this.)

@carlesfernandez Can you fix this?

carlesfernandez, 2020-03-02:
Hi,

thanks for spotting this bug. I still do not have a clear solution for that. Using  `boost::gregorian::date d1(d_yr - J + 1.0, 1, 1);` implies the recomputation of `J` from `d_N_T`, which is weird since we would be doing that twice. On the other hand, the way it is computed in `glonass_gnav_navigation_message.cc`  is precisely what the ICD says.

Maybe I'm missing the point; could you please submit a pull-request if you can see where exactly the error is?

ValZapod, 2020-03-03:
@carlesfernandez 

> implies the recomputation of J from

Just pass it through... No? Anyway, you are using Boost, so do not be afraid of recomputation, it will not be that important, ha-ha. Or you can change the incorrect year after boost magic... Yeah, to d_yr.

ValZapod, 2020-04-12:
@carlesfernandez So, good news! This doc (with errata) on page 57 has all the decoding so you can deboostify it! Nice! https://github.com/tomojitakasu/RTKLIB/files/3657985/ICD_GLONASS_5.1_eng_201402.pdf

carlesfernandez, 2020-07-20:
This seems to be now fixed. Not the most elegant solution, but provides the correct date according to the tests results. Thanks for your help!

commit 4ab76900 (2020-07-20): Fix bug in GLONASS year computation (Fixes: #347)
closed 2020-07-29 by carlesfernandez