gnss-sdr/gnss-sdr #348: RINEX output bug

yealwang, 2019-12-10:
Hi all
There will be some bug in the RINEX output process, sometimes it can write into the RINEX data sometimes not. 
I have not find it until now.
I use the current next branch, process the L1L2 data, there have the RINEX output file, but they are empty, and sometime they have data.
BW
Yeal

yealwang, 2019-12-10:
And for the GPS L2 data, the output are always not write, empty or with 0.0001.

AmirAllahvirdi, 2021-01-29:
Hi @yealwang,

I have the same issue, it is written [here](https://gnss-sdr.org/docs/sp-blocks/pvt/#kml-geojson-gpx) that if the RINEX file will be deleted if no position fix have been obtained during the execution of the software receiver. 

Dear @carlesfernandez, 
Is there any way to save the RINEX file even if there is no position fix? 

Regards
Amir

carlesfernandez, 2021-01-29:
Hi all,

no, you cannot retrieve RINEX files before getting the first PVT fix. In the case of using a multi-signal receiver (e.g. L1 and L2), you need the first fix AND having at least one locked channel for each of the targeted signals. Annotations on RINEX observation files requires data that cannot be obtained before the computation of the first fix. 

If you are interested in retrieving the collected ephemeris before the first fix, you still can look at the produced .xml files. 

You can also retrieve uncorrected pseudoranges, Dopplers, etc. activating the dump file of the Observables block in your configuration.

Hope this helps,
Carles

AmirAllahvirdi, 2021-01-29:
> Hi all,
>
> no, you cannot retrieve RINEX files before getting the first PVT fix. In the case of using a multi-signal receiver (e.g. L1 and L2), you need the first fix AND having at least one locked [...]

Thank you, @carlesfernandez