gnss-sdr/gnss-sdr #349: Bug in Glonass_Gnav_Navigation_Message::CRC_test

DavidRamati, 2019-12-18:
in some cases the Glonass_Gnav_Navigation_Message::CRC_test returns true even when the hamming bits are wrong.

this is a test that shows how Glonass_Gnav_Navigation_Message::CRC_test returns true on two almost equal strings with the same hamming bits, that differs only in 1 critical bit.
the test also shows that rtklib's implementation for this crc test returns false for the wrong string.
```

#include "gtest/gtest.h"
#include "src/core/system_parameters/GLONASS_L1_L2_CA.h"
#include <bitset>

// test data
std::string string1Real_14_18_00("0000100000111001001001000011101010101100010101001000001001011101010101110011110110101");
std::string string1Real_14_18_30("0000100000111001001011000011101010101100010101001000001001011101010101110011100001010");
std::string string1Wrong_14_18_00("0000100000111001001001000011101010101100010101001000001001011101010101110011100001010");
```
_the only difference between string1Real_14_18_30 and string1Wrong_14_18_00 strings is the last bit of tk that reprsents 30 seconds_

**GnssSdrCRCTest function was copied from src/core/system_parameters/glonass_gnav_navigation_message.cc**
```
bool GnssSdrCRCTest(std::bitset<GLONASS_GNAV_STRING_BITS> bits)
{
    uint32_t sum_bits = 0;
    int32_t sum_hamming = 0;
    int32_t C1 = 0;
    int32_t C2 = 0;
    int32_t C3 = 0;
    int32_t C4 = 0;
    int32_t C5 = 0;
    int32_t C6 = 0;
    int32_t C7 = 0;
    int32_t C_Sigma = 0;
    std::vector<uint32_t> string_bits(GLONASS_GNAV_STRING_BITS);

    // Populate data and hamming code vectors
    for (uint32_t i = 0; i < string_bits.size(); i++)
        {
            string_bits[i] = static_cast<uint32_t>(bits[i]);
        }

    // Compute C1 term
    sum_bits = 0;
    for (int i : GLONASS_GNAV_CRC_I_INDEX)
        {
            sum_bits += string_bits[i - 1];
        }
    C1 = string_bits[0] ^ (sum_bits % 2);

    // Compute C2 term
    sum_bits = 0;
    for (int j : GLONASS_GNAV_CRC_J_INDEX)
        {
            sum_bits += string_bits[j - 1];
        }
    C2 = (string_bits[1]) ^ (sum_bits % 2);

    // Compute C3 term
    sum_bits = 0;
    for (int k : GLONASS_GNAV_CRC_K_INDEX)
        {
            sum_bits += string_bits[k - 1];
        }
    C3 = string_bits[2] ^ (sum_bits % 2);

    // Compute C4 term
    sum_bits = 0;
    for (int l : GLONASS_GNAV_CRC_L_INDEX)
        {
            sum_bits += string_bits[l - 1];
        }
    C4 = string_bits[3] ^ (sum_bits % 2);

    // Compute C5 term
    sum_bits = 0;
    for (int m : GLONASS_GNAV_CRC_M_INDEX)
        {
            sum_bits += string_bits[m - 1];
        }
    C5 = string_bits[4] ^ (sum_bits % 2);

    // Compute C6 term
    sum_bits = 0;
    for (int n : GLONASS_GNAV_CRC_N_INDEX)
        {
            sum_bits += string_bits[n - 1];
        }
    C6 = string_bits[5] ^ (sum_bits % 2);

    // Compute C7 term
    sum_bits = 0;
    for (int p : GLONASS_GNAV_CRC_P_INDEX)
        {
            sum_bits += string_bits[p - 1];
        }
    C7 = string_bits[6] ^ (sum_bits % 2);

    // Compute C_Sigma term
    sum_bits = 0;
    sum_hamming = 0;
    for (int q : GLONASS_GNAV_CRC_Q_INDEX)
        {
            sum_bits += string_bits[q - 1];
        }
    for (int32_t q = 0; q < 8; q++)
        {
            sum_hamming += string_bits[q];
        }
    C_Sigma = (sum_hamming % 2) ^ (sum_bits % 2);

    // Verification of the data
    // (a-i) All checksums (C1,...,C7 and C_Sigma) are equal to zero
    if ((C1 + C2 + C3 + C4 + C5 + C6 + C7 + C_Sigma) == 0)
        {
            return true;
        }
    // (a-ii) Only one of the checksums (C1,...,C7) is equal to zero but C_Sigma = 1
    if (C_Sigma == 1 && C1 + C2 + C3 + C4 + C5 + C6 + C7 == 6)
        {
            return true;
        }

    // All other conditions are assumed errors. TODO: Add correction for case B
    return false;
}

TEST(GlonassCrcTest, GnssSdrCRCTest)
{
    std::bitset<GLONASS_GNAV_STRING_BITS> bits;
    bits = std::bitset<GLONASS_GNAV_STRING_BITS>(string1Real_14_18_00);
    ASSERT_TRUE(GnssSdrCRCTest(bits));
    bits = std::bitset<GLONASS_GNAV_STRING_BITS>(string1Real_14_18_30);
    ASSERT_TRUE(GnssSdrCRCTest(bits));
    bits = std::bitset<GLONASS_GNAV_STRING_BITS>(string1Wrong_14_18_00);
    ASSERT_FALSE(GnssSdrCRCTest(bits));
}
```
**test_glostr function was copied from rtklib repository rcvraw.c file**
```
/* test hamming code of glonass ephemeris string -------------------------------
* test hamming code of glonass ephemeris string (ref [2] 4.7)
* args   : unsigned char *buff I glonass navigation data string bits in frame
*                                with hamming
*                                  buff[ 0]: string bit 85-78
*                                  buff[ 1]: string bit 77-70
*                                  ...
*                                  buff[10]: string bit  5- 1 (0 padded)
* return : status (1:ok,0:error)
*-----------------------------------------------------------------------------*/
bool test_glostr(const unsigned char *buff)
{
    static const unsigned char xor_8bit[256]={ /* xor of 8 bits */
        0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,
        1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,
        1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,
        0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,
        1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,
        0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,
        0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0,1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,
        1,0,0,1,0,1,1,0,0,1,1,0,1,0,0,1,0,1,1,0,1,0,0,1,1,0,0,1,0,1,1,0
    };
    static const unsigned char mask_hamming[][12]={ /* mask of hamming codes */
        {0x55,0x55,0x5A,0xAA,0xAA,0xAA,0xB5,0x55,0x6A,0xD8,0x08},
        {0x66,0x66,0x6C,0xCC,0xCC,0xCC,0xD9,0x99,0xB3,0x68,0x10},
        {0x87,0x87,0x8F,0x0F,0x0F,0x0F,0x1E,0x1E,0x3C,0x70,0x20},
        {0x07,0xF8,0x0F,0xF0,0x0F,0xF0,0x1F,0xE0,0x3F,0x80,0x40},
        {0xF8,0x00,0x0F,0xFF,0xF0,0x00,0x1F,0xFF,0xC0,0x00,0x80},
        {0x00,0x00,0x0F,0xFF,0xFF,0xFF,0xE0,0x00,0x00,0x01,0x00},
        {0xFF,0xFF,0xF0,0x00,0x00,0x00,0x00,0x00,0x00,0x02,0x00},
        {0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xF8}
    };
    unsigned char cs;
    int i,j,n=0;
    
    for (i=0;i<8;i++) {
        for (j=0,cs=0;j<11;j++) {
            cs^=xor_8bit[buff[j]&mask_hamming[i][j]];
        }
        if (cs) n++;
    }
    return n==0||(n==2&&cs);
}

void stringToCharArr(const std::string& src, unsigned char dest[11])
{
    for (int i = 0; i < GLONASS_GNAV_STRING_BITS; ++i)
    {
        if (src[i] == '1')
        {
            dest[i / 8] |= 1 << (7 - (i % 8));
        }
    }
}

TEST(GlonassCrcTest, RtklibCRCTest)
{
    unsigned char buff[11] = {0};
    stringToCharArr(string1Real_14_18_00, buff);
    ASSERT_TRUE(test_glostr(buff));
    memset(buff, 0, 11);
    stringToCharArr(string1Real_14_18_30, buff);
    ASSERT_TRUE(test_glostr(buff));
    memset(buff, 0, 11);
    stringToCharArr(string1Wrong_14_18_00, buff);
    ASSERT_FALSE(test_glostr(buff));
}

```
i don't know for sure where the bug is in Glonass_Gnav_Navigation_Message::CRC_test, but i guess it is in the second condition for success. (the one that marked with "TODO: Add correction for case B"...)
you can always just use rtklib's implementation...

ValZapod, 2020-04-09:
@carlesfernandez
This is a typo in english doc, compare using google translate:

"or if only one of the checksums (C1,...,C7) is equal to zero but CΣ = 1;"

"либо лишь одна из контрольных сумм С1,...,С7 равна единице и при этом СΣ = 1;" (page 47)

From russian can be translated as "or if only one of the checksums (C1,...,C7) is equal to **1 and** CΣ = 1;"

So the second condition for success should be corrected. Smth like (or with ! operators).
```
    if (C_Sigma == 1 && C1 + C2 + C3 + C4 + C5 + C6 + C7 == 1)
        {
            return true;
        }

```

I have a question though. The code here should be self correcting... But is it?? It does not look like it... Strange.

carlesfernandez, 2020-04-10:
Dear @DavidRamati and @ValZapod ,

This is a very good catch! It has been fixed in the `next` branch. Please note that the code correction (Case B) is still not implemented.

ValZapod, 2020-04-10:
Wait a second C1, ..., C7 can be only 1 or 0? Why not use bool then? 1 byte instead of 4.

Also, why not use rtklib code?? Also, even rtklib does not have case B! Why???

ValZapod, 2020-04-12:
Did you see that post @carlesfernandez https://github.com/tomojitakasu/RTKLIB/issues/492#issuecomment-535505519
It has the GLONASS en document WITH ERRATA! And that one typo here is fixed there! Nice, is not it!

ValZapod, 2020-05-31:
> Wait a second C1, ..., C7 can be only 1 or 0?

Ping))

ValZapod, 2020-05-31:
I found the implementation for self correction, BTW. https://github.com/swift-nav/libswiftnav-legacy/pull/337/commits/80d90e75656e2eedfe009534687282fca64d73c1#diff-3ac8d872fc5ef856ac16214211062546R117
Look here for newer lib: https://github.com/swift-nav/libswiftnav/blob/master/src/decode_glo.c#L122 It is used here, BTW https://github.com/cbeighley/peregrine/blob/master/peregrine/iqgen/bits/message_glo.py

ValZapod, 2020-07-20:
@carlesfernandez Did you see the implementation above? It is LGPL-3.0. The only problem would be to get rid of const'ed bitstream and actually invert that 1 bit that needs to be inverted and only then const it.🤔 Nice work on the other issue))

commit ec3c8686 (2020-04-10): Fixes a bug in GLONASS GNAV CRC computation (Fixes: #349)
closed 2020-07-29 by carlesfernandez