gnss-sdr/gnss-sdr #353: Regarding the Acquisition of GPSSIM.bin file by GPSSDR SIM

komalkumar6655, 2020-01-25:
Hello, 
I have generated the gpssim.bin file . But i cant get the acquisition results in GNSS SDR.
Please help me with this 
Thank You.

jonathanmuller, 2021-01-12:
Hi, extremely late reply but may help anyone with the same problem : 

You probably want to use this in your conf file :

```
;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/path/to/your/file/gpssim.bin
SignalSource.item_type=ishort
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
SignalSource.samples=0
```
And make sure to use the conf by using the "-c" option of the command line :
`gnss-sdr --config_file=../conf/from_file.conf`

Depending on how you generated your .bin you should adapt the frequency
For example gps-sdr-sim use  2600000 by default

Full explanation here : https://gnss-sdr.org/docs/sp-blocks/signal-source/

mooction, 2021-12-15:
It seems if sampling frequency is set to 2600000 (defualt value), gnss-sdr failed to decode GPS NAV message. But sampling freq set to 4000000 is OK.

ZiglioUK, 2021-12-15:
Actually I was having a related issue last week, when I tried to replay some simulated data from gps-sdr-sim through a HackRF. I was wondering why the default value wasn't a "round" number. I settled on 2MHz in order to limit the file size.

Unrelated, I thought hackrf_transfer would save files as unsigned 8 bits (like the rtl-sdr) but expects to read them as signed. I'm really confused.