gnss-sdr/gnss-sdr #355: Issue regarding E0128 14:17:04.282202 16982 gnss_block_factory.cc:1112] PVT.RTKLIB_PVT: Undefined implementation for block Segmentation fault (core dumped)

komalkumar6655, 2020-01-28:
Please help me 
E0128 14:17:04.282202 16982 gnss_block_factory.cc:1112] PVT.RTKLIB_PVT: Undefined implementation for block
Segmentation fault (core dumped)
with this issue.

songg12138, 2021-01-11:
Did you find the solution? I am stuck in this problem too...

carlesfernandez, 2021-01-12:
I guess you are using a quite old version of gnss-sdr. What's the output of:

```
$ gnss-sdr --version
```

It should be `gnss-sdr version 0.0.14`, or `gnss-sdr version 0.0.14-git-next-xxxxxx` if you are using the `next` branch.

If you installed gnss-sdr with `sudo apt-get install gnss-sdr`, please uninstall it (`sudo apt-get uninstall gnss-sdr`), and then build and install it from source (following instructions at https://gnss-sdr.org/build-and-install/#clone-build-and-install-gnss-sdr )

Hope this helps,
Carles