gnss-sdr/gnss-sdr #356: Drivers for FPGA

noelpedro, 2020-01-31:
Hi guys,

Can you tell me what drivers you guys used to access dma for the injecting of file signal source to HW accelerators?

I know a while ago you guys used EZDMA, but I know it has some bugs, so I thought I’d ask.

Have you guys looked at github.com/ikwzm/udmabuf driver??

Cheers

mmajoral, 2020-03-16:
Hi noelpedro,
We use the EZDMA driver in order to read files and inject the signal into the HW. You can get it from https://github.com/jeremytrimble/ezdma. It works correctly for our purposes.

Thank you for your suggestion concerning the udmabuf driver, we'll have a look at it.
Regards !

noelpedro, 2020-03-16:
I’m only able to get ezdma to work on the zedboard using jeremeytrimble/Linux-xlnx  Linux kernel.

Is that also the case for you?

For ultrascale+ I’ve been using udmabuf with uio to make dma transactions. It’s not as easy as ezdma’s read write calls but couldn’t get anything else to work on zynqmp ultrascale+.

Thanks man

noelpedro, 2020-03-22:
I cant write more than 4k bytes to the DMA I see you are writing 64K bytes. Im not sure if its ezdma or maybe my axi_dma dosent have correct configurations.

noelpedro, 2020-03-22:
Just had to increase depth of FIFO

closed 2020-03-22 by noelpedro

noelpedro, 2020-03-23:
Hey @mmajoral,

For the File to DMA injection did you use the analogdevices kernel if so which version?