gnss-sdr/gnss-sdr #357: Highest sampling rate for tracking loop FPGA

noelpedro, 2020-01-31:
Hi guys for your closed source HW accelerators, what is the highest sampling rate that has been tested for the tracking loop for both the file signal source and the real time AD9361 signal source?

I know for acquisition it’s best to down sample to at most 2MHz.

Cheers

mmajoral, 2020-03-16:
Hi noelpedro,

The sampling rate in the FPGA is configurable. The highest sampling frequency you can use with the current implementation of the FPGA HW accelerator modules is 16 MSPs.  
However, the length of the buffers between the A/D converter and the HW accelerator modules is fixed. Therefore the number of simultaneous channels that you can process depends on the sampling frequency that you choose.

We usually use a sampling frequency of 12.5 Msps because this was our initial target and because we use a dual band configuration using both the L1 and the L5 frequency bands using a large number of channels. Then we use a downsampling filter in the L1 band for the acquisition. But it is expected that using 16 MSPs and a long tracking integration time then you can probably use a large number of channels too.

noelpedro, 2020-03-16:
Does 4  1C channels at 40msps sounds like a realistic target on a ultra scale+ board.

closed 2020-03-23 by noelpedro