gnss-sdr/gnss-sdr #358: My first position fix

teku45, 2020-02-04:
Hi all, 

I am running the my first position fix example on GNSS-SDR on Linux 18.04 LTS. I am unable to get any position fix. Furthermore, the Current receiver time In the output seems to be advancing around 2-3 times faster than actual real seconds. My result is attached below as well as the configuration file:

```
Initializing GNSS-SDR v0.0.11.git-- ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 s
GPS L1 C/A tracking bit synchronization locked in channel 3 for satellite GPS PRN 11 (Block IIR)
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
Current receiver time: 14 s
GPS L1 C/A tracking bit synchronization locked in channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 15 s
Current receiver time: 16 s
Current receiver time: 17 s
Loss of lock in channel 2!
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Loss of lock in channel 6!
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block IIR)
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
Current receiver time: 21 s
Current receiver time: 22 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
Current receiver time: 27 s
Current receiver time: 28 s
Current receiver time: 29 s
Current receiver time: 30 s
[... 200 lines cut ...]
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2
```

carlesfernandez, 2020-02-05:
Dear @teku45 , thanks for spotting this error.

You are right, the configuration in that page is outdated if you are using the `next`  branch. 

In order to get it run, you can replace the line:
```
Acquisition_1C.threshold=0.008
```
by:
```
Acquisition_1C.pfa=0.01
```

This has been corrected on the website. This new configuration should work both with the `master` and the `next` branches.

teku45, 2020-02-05:
Thank you! This fixed my issue.

closed 2020-02-05 by teku45