gnss-sdr/gnss-sdr #360: while building the gnss-sdr it showing the following error

Rrravi, 2020-02-23:
could not convert ‘std::dynamic_pointer_cast(const std::__shared_ptr<_Tp2, _Lp>&) [with _Tp = PvtInterface; _Tp1 = GNSSBlockInterface; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2]()’ from ‘std::__shared_ptr<PvtInterface, (__gnu_cxx::_Lock_policy)2>’ to ‘std::shared_ptr<PvtInterface>’

carlesfernandez, 2020-02-23:
Could you describe your compiling environment? (OS & version, compiler & version, etc.)

Rrravi, 2020-02-23:
Ubuntu 18.04 lts
The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0

carlesfernandez, 2020-02-24:
Dear @Rrravi ,

both the master and the next branches of the repository build well on Ubuntu 18.04 from a fresh start. There should be something that is not standard in your installation.

How did you install GNU Radio and dependencies? From packages or using PyBOMBS?

Rrravi, 2020-02-24:
i installed gnu radio from dependencies and my problem is resolved by re installing ubuntu and thank you for your faster response

carlesfernandez, 2020-02-24:
Glad you solved your issue!

closed 2020-02-24 by carlesfernandez