gnss-sdr/gnss-sdr #361: GNSS-SDR can not create message queues!

wamlord, 2020-03-09:
Trying the My First Fix Tutorial and got this:
[image]
Could someone tell me what is wrong?
Thanks in advance.

AdityaSanthosh, 2020-03-09:
I think there's something wrong with your installation. I tried the First Fix and everything seems to be fine. Remove the existing Installation and try again
[image]

acebrianjuan, 2020-03-09:
HI @wamlord,

This error indicates that something is preventing GNSS-SDR from creating a System V message queue. The following code fragment is the responsible of creating the message queue:

https://github.com/gnss-sdr/gnss-sdr/blob/4a9dcff5b99087f87d2e17ed86ecf1d256161d12/src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc#L409-L417

According to the [msgget(2)](http://man7.org/linux/man-pages/man2/msgget.2.html) man page, these are the possible causes preventing a message queue from being created:

```
ERRORS

       On failure, errno is set to one of the following values:

       EACCES A message queue exists for key, but the calling process does
              not have permission to access the queue, and does not have the
              CAP_IPC_OWNER capability in the user namespace that governs
              its IPC namespace.

       EEXIST IPC_CREAT and IPC_EXCL were specified in msgflg, but a message
              queue already exists for key.

       ENOENT No message queue exists for key and msgflg did not specify
              IPC_CREAT.

       ENOMEM A message queue has to be created but the system does not have
              enough memory for the new data structure.

       ENOSPC A message queue has to be created but the system limit for the
              maximum number of message queues (MSGMNI) would be exceeded.
```

Therefore, in order to diagnose the cause of your problem you will have to find out the value of the [`errno`](http://man7.org/linux/man-pages/man3/errno.3.html) variable. A simple way to do it would be to print the `errno` value and message string as explained [here](https://www.fayewilliams.com/2011/10/19/using-errno/).

Put simply, follow these steps:

1. Edit the [rtklib_pvt_gs.cc](https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc) file with a text editor.
2. `#include <cerrno>` at the beginning of the file.
3. Add the following lines of code immediately before the `throw std::exception()` in [line 416](https://github.com/gnss-sdr/gnss-sdr/blob/4a9dcff5b99087f87d2e17ed86ecf1d256161d12/src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc#L416):

```
std::cout << "Something went wrong! errno " << errno << ": ";
std::cout << strerror(errno) << std::endl;
```

4. Save the file.
5. Re-build GNSS-SDR.
6. Run the [My first position fix](https://gnss-sdr.org/my-first-fix/) tutorial again and paste the output in this issue.

Regards,
Álvaro

wamlord, 2020-03-09:
@acebrianjuan Thanks for your Reply. I do not have this file. I have a rtklib file, but it looks nothing like the file you are referencing. Should I download this file and put it somewhere? Thanks.

acebrianjuan, 2020-03-10:
Hi @wamlord,

You are looking in the wrong place.
The `rtklib_pvt_gs.cc` file is in the directory: `gnss-sdr/src/algorithms/PVT/gnuradio_blocks/`

Regards,
Álvaro

wamlord, 2020-03-10:
@acebrianjuan, 
First of all, thank you for all your help so far on this. Here is the error message:
[image]

carlesfernandez, 2020-03-10:
Hi guys,

thanks for raising this issue. The Sys V message queue is a mechanism available *almost* everywhere, but we have hit an environnment in which the message queue seems to be not there. This mechanism is only used by the Time-to-First-Fix test, and it does not have any effect on the PVT computation.

Based on  @acebrianjuan's suggestions, I have pushed a commit to the `next` branch (see f233184a315fb082aeed75a923f7e1015e1382ac) that should allow to compute PVT fixes when the Sys V message queue is not available. The TTFF test will not work for you, but all other receiver's functionalities remain the same.

Please @wamlord  let us know if it works for you.

commit f233184a (2020-03-10): Fix receiver if the System V message queue is not available (Fixes: #361)
closed 2020-03-13 by carlesfernandez