gnss-sdr/gnss-sdr #363: VisualStudio support

pietrodcs, 2020-03-11:
Good morning
Is there a support or someone's experience in building with VisualStudio or Eclipse with GCC ?

Thank You
Pietro

carlesfernandez, 2020-03-11:
Hi,

building with Visual Studio is currently not possible. Actually, this is an idea proposed for this year's edition of [Google Summer of Code](https://gnss-sdr.org/google-summer-code-2020-ideas-list/#project-title-porting-gnss-sdr-to-microsoft-windows).

Building with Eclipse and GCC works in Linux and macOS, but not in Windows.

pietrodcs, 2020-03-11:
Hi Carles
Thank You for the reply.
I consider to wait for this step.

Best Regards
Pietro

closed 2020-03-11 by pietrodcs

gvanem, 2020-06-22:
I tried to tweak some sources for building with MSVC or clang-cl. The main hurdle AFAICS is the
use of complex types in plain C (MSVC and GNU-g++ is rather different in this regard).

But I had more success compiling some `volk_gnsssdr/lib/*.c` files as C++. Until this also failed at some `volk_gnsssdr/kernels` files. So I hope something comes out of the GSoC program.

pietrodcs, 2020-06-22:
Hi gvanem
Is the work still in progress or ready to run on the VisualC? 
Also the Eclipse on windows is ok...

Thank You

gvanem, 2020-06-22:
@pietrodcs GNSS is huge! I've only built some parts of it. Besides there is the mess with C++ standards; some stuff  in GNSS requires C++17, but latest GnuRadio has shit like `std::unary_function` which is (will be?) deleted from C++17.

pietrodcs, 2020-06-22:
Ok..
No hurry....

Thank You
Pietro