gnss-sdr/gnss-sdr #366: Multi Source Issue

arielayb, 2020-03-18:
I've been trying to get the gnss-sdr multisource Hybrid ishort  example to work with the latest .dat file, 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat.

The config file I've used is the one provided by the gnss-sdr repo under the next branch. In addition, I've replaced the Galileo signal with GLONASS, 1G.

Please, help me understand what I'm doing wrong.

Thank you!

; This is a GNSS-SDR configuration file
; The configuration API is described at https://gnss-sdr.org/docs/sp-blocks/

; You can define your own receiver and invoke it by doing
; gnss-sdr --config_file=my_GNSS_SDR_configuration.conf
;

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
;internal_fs_sps: Internal signal sampling frequency after the signal conditioning stage [samples per second].
GNSS-SDR.internal_fs_sps=4000000

Receiver.sources_count=2

SignalSource.enable_throttle_control=false
SignalSource.repeat=false

;######### SIGNAL_SOURCE 0 CONFIG ############
SignalSource0.implementation=File_Signal_Source
SignalSource0.filename=../work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat  ; <- PUT YOUR FILE HERE
SignalSource0.item_type=ishort
SignalSource0.sampling_frequency=4000000
SignalSource0.samples=0

;######### SIGNAL_SOURCE 1 CONFIG ############
SignalSource1.implementation=File_Signal_Source
SignalSource1.filename=../work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat ; <- PUT YOUR FILE HERE
SignalSource1.item_type=ishort
SignalSource1.sampling_frequency=4000000
SignalSource1.freq=1575420000
SignalSource1.samples=0

;######### SIGNAL_CONDITIONER 0 CONFIG ############
SignalConditioner0.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER 0 CONFIG ############
DataTypeAdapter0.implementation=Ishort_To_Complex

;######### INPUT_FILTER 0 CONFIG ############
InputFilter0.implementation=Pass_Through
InputFilter0.dump=false
InputFilter0.dump_filename=../data/input_filter.dat
InputFilter0.input_item_type=gr_complex
InputFilter0.output_item_type=gr_complex

;######### RESAMPLER 1 CONFIG ############
Resampler1.implementation=Pass_Through
Resampler1.dump=false
Resampler1.dump_filename=../data/resampler.dat
Resampler1.item_type=gr_complex
Resampler1.sample_freq_in=4000000
Resampler1.sample_freq_out=4000000

;######### SIGNAL_CONDITIONER 1 CONFIG ############
SignalConditioner1.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER 1 CONFIG ############
DataTypeAdapter1.implementation=Ishort_To_Complex

;######### INPUT_FILTER 1 CONFIG ############
InputFilter1.implementation=Pass_Through
InputFilter1.dump=false

;######### RESAMPLER 1 CONFIG ############
Resampler1.implementation=Pass_Through
Resampler1.dump=false
Resampler1.dump_filename=../data/resampler.dat.
Resampler1.item_type=gr_complex
Resampler1.sample_freq_in=4000000
Resampler1.sample_freq_out=4000000

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=2
Channels_1G.count=2
Channels.in_acquisition=1

;# CHANNEL CONNECTION
Channel0.RF_channel_ID=0
Channel1.RF_channel_ID=0
Channel2.RF_channel_ID=1
Channel3.RF_channel_ID=1
;#signal:
;#if the option is disabled by default is assigned "1C" GPS L1 C/A
Channel.signal=1G

;######### GPS ACQUISITION CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.threshold=0.0075
;Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=500
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### GLONASS ACQUISITION CONFIG ############
Acquisition_1G.implementation=GLONASS_L1_CA_PCPS_Acquisition
Acquisition_1G.item_type=gr_complex
Acquisition_1G.coherent_integration_time_ms=4
;Acquisition_1G.threshold=0
Acquisition_1G.pfa=0.0000008
Acquisition_1G.doppler_max=15000
Acquisition_1G.doppler_step=125
Acquisition_1G.dump=false
Acquisition_1G.dump_filename=./acq_dump.dat

;######### TRACKING GPS CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=45.0;
Tracking_1C.dll_bw_hz=4.0;
Tracking_1C.order=3;
Tracking_1C.dump=false
Tracking_1C.dump_filename=../data/epl_tracking_ch_

;######### TRACKING GLONASS CONFIG ############
Tracking_1G.implementation=GLONASS_L1_CA_DLL_PLL_Tracking
Tracking_1G.item_type=gr_complex
Tracking_1G.pll_bw_hz=15.0;
Tracking_1G.dll_bw_hz=2.0;
Tracking_1G.order=3;
Tracking_1G.early_late_space_chips=0.15;
Tracking_1G.very_early_late_space_chips=0.6;
Tracking_1G.dump=false
Tracking_1G.dump_filename=../data/veml_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### TELEMETRY DECODER GLONASS CONFIG ############
TelemetryDecoder_1G.implementation=GLONASS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1G.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.output_rate_ms=100;
PVT.display_rate_ms=500;
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump_filename=./PVT
PVT.dump=false

acebrianjuan, 2020-03-20:
Hi @arielayb,

> Please, help me understand what I'm doing wrong.

Please remember to provide the following pieces of information, so that you can get a better help:

- The gnss-sdr version you are using.
- A detailed description of the problems/difficulties that you are experiencing.

From a cursory look at your configuration file, I can see that both `SignalSource0.filename` and `SignalSource1.filename` are pointing to the same raw signal data file (`2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat`).
If you want to process two files, one for each band (i.e. GPS L1 C/A and Glonass L1 C/A), you will have to point `SignalSource1.filename` to a Glonass raw signal data file instead.

In addition, I would say there is a mistake in the `CHANNEL CONNECTION` settings of your configuration file:

> ;# CHANNEL CONNECTION
> Channel0.RF_channel_ID=0
> Channel1.RF_channel_ID=0
> Channel2.RF_channel_ID=1
> Channel3.RF_channel_ID=1
> ;#signal:
> ;#if the option is disabled by default is assigned "1C" GPS L1 C/A
> Channel.signal=1G

If you want half of the channels for processing `1C` and the other half for `1G`, then try using these settings instead:

```
; # CHANNEL CONNECTION
Channel0.RF_channel_ID=0
Channel1.RF_channel_ID=0
Channel2.RF_channel_ID=1
Channel3.RF_channel_ID=1

Channel0.signal=1C
Channel1.signal=1C
Channel2.signal=1G
Channel3.signal=1G
```

References to the documentation:

- [https://gnss-sdr.org/docs/sp-blocks/signal-source/#multiple-sources](https://gnss-sdr.org/docs/sp-blocks/signal-source/#multiple-sources)

- [https://gnss-sdr.org/docs/sp-blocks/channels/#multi-source-receiver](https://gnss-sdr.org/docs/sp-blocks/channels/#multi-source-receiver)

Let us know if this solved your issue.

Regards,
Álvaro

arielayb, 2020-03-23:
Thank you! My issue is resolved!

closed 2020-03-23 by arielayb