gnss-sdr/gnss-sdr #369: unable to get position for glonass data in gnss-sdr following are configuration file and Result

Rrravi, 2020-04-03:
**CONFIGURATION FILE**
; This is a GNSS-SDR configuration file
; The configuration API is described at https://gnss-sdr.org/docs/sp-blocks/

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=16000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/ravindar/Desktop/gnss-sdr-master-old/data/Glonass/FFF005.DAT ; <- PUT YOUR FILE HERE ; <- PUT YOUR FILE HERE
SignalSource.item_type=ibyte
SignalSource.sampling_frequency=16000000
SignalSource.samples=0
SignalSource.dump=false;
SignalSource.dump_filename=/archive/signal_glonass.bin

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ibyte_To_Complex

InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.filter_type=lowpass
InputFilter.grid_density=14
InputFilter.IF=3500000
InputFilter.bw=7000000
InputFilter.sampling_frequency=16000000

Resampler.implementation=Pass_Through
Resampler.sample_freq_in=16000000
Resampler.sample_freq_out=8000000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channel.signal=1G
Channels.in_acquisition=1
Channels_1G.count=10

Channel0.satellite=2   
Channel1.satellite=16   
Channel2.satellite=17   
Channel3.satellite=18  

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1G.implementation=GLONASS_L1_CA_PCPS_Acquisition
Acquisition_1G.item_type=gr_complex
Acquisition_1G.threshold=0.004
Acquisition_1G.pfa=0.0001
Acquisition_1G.doppler_max=10000
Acquisition_1G.doppler_step=250
Acquisition_1G.coherent_integration_time_ms=2

;######### TRACKING GLOBAL CONFIG ############
Tracking_1G.implementation=GLONASS_L1_CA_DLL_PLL_C_Aid_Tracking
Tracking_1G.item_type=gr_complex
Tracking_1G.early_late_space_chips=0.5
Tracking_1G.pll_bw_hz=40.0;
Tracking_1G.dll_bw_hz=3.0;
Tracking_1G.extend_correlation_ms=1

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1G.implementation=GLONASS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
[... 267 lines cut ...]
New GLONASS L1 GNAV almanac received in channel 5 from satellite Glonass PRN 16 (Block -7)
New GLONASS L1 GNAV almanac received in channel 8 from satellite Glonass PRN 18 (Block 2)
Tracking of GLONASS L1 C/A signal started on channel 2 for satellite Glonass PRN 17 (Block 4)
Current receiver time: 40 s
Loss of lock in channel 1!
Tracking of GLONASS L1 C/A signal started on channel 6 for satellite Glonass PRN 12 (Block -1)
Tracking of GLONASS L1 C/A signal started on channel 1 for satellite Glonass PRN 16 (Block -1)
Loss of lock in channel 2!
Current receiver time: 41 s
Loss of lock in channel 6!
Tracking of GLONASS L1 C/A signal started on channel 4 for satellite Glonass PRN 01 (Block 1)
Loss of lock in channel 1!
Current receiver time: 42 s
Loss of lock in channel 4!
Tracking of GLONASS L1 C/A signal started on channel 4 for satellite Glonass PRN 09 (Block -2)
Current receiver time: 43 s
Tracking of GLONASS L1 C/A signal started on channel 2 for satellite Glonass PRN 17 (Block 4)
New GLONASS L1 GNAV almanac received in channel 3 from satellite Glonass PRN 17 (Block -3)
New GLONASS L1 GNAV almanac received in channel 0 from satellite Glonass PRN 02 (Block -4)
New GLONASS L1 GNAV almanac received in channel 5 from satellite Glonass PRN 16 (Block -7)
New GLONASS L1 GNAV almanac received in channel 8 from satellite Glonass PRN 18 (Block 2)
Loss of lock in channel 4!
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 117.524 [seconds]
GNSS-SDR program ended.

marxwolf, 2020-09-02:
@Rrravi Do you still have this issue? I use USRP x310 and GNSS-SDR v0.0.13, also appears this issue?

YHao29, 2023-05-23:
@marxwolf I have the same issue. do you have any progress? thanks