gnss-sdr/gnss-sdr #373: Build errors on macOS

sdaheng, 2020-05-10:
Hello

I faced some problems when I tried to build with Xcode. 
After configure and generate with CMake. And opened Xcode project, ran "run_tests" . Compiler shows many errors.

**Here are some my environment information.**
OS version: macOS 10.15.4
Xcode version: 11.4.1
CMake version: 3.17.2

**CMake configure log**
The CXX compiler identification is AppleClang 11.0.3.11030032
The C compiler identification is AppleClang 11.0.3.11030032
Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++
Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ - works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang
Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang - works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Configuring GNSS-SDR v0.0.12 to be built on macOS Catalina 10.15
Build type set to Release.
Performing Test CMAKE_HAVE_LIBC_PTHREAD
Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
Checking for module 'gtest'
  No package 'gtest' found
Could NOT find GOOGLETEST (missing: LIBGTEST_DEV_DIR GTEST_INCLUDE_DIRS) 
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_PMT_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-blocks.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: FFT
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-fft.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-filter.dylib;/opt/local/lib/libgnuradio-fft.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-analog.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib;/opt/local/lib/libvolk.dylib
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: UHD
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-uhd.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_UHD_FOUND = TRUE
Found Boost: /usr/local/include (found suitable version "1.68.0", minimum required is "1.53") found components: atomic chrono date_time serialization system thread filesystem 
gflags library found at /opt/local/lib/libgflags.dylib
glog library found at /opt/local/lib/libglog.dylib
PugiXML include = /opt/local/include
PugiXML library = /opt/local/lib/libpugixml.dylib
CMake Warning at /Applications/CMake.app/Contents/share/cmake-3.17/Modules/FindProtobuf.cmake:500 (message):
  Protobuf compiler version 3.5.1 doesn't match library version 3.11.4
Call Stack (most recent call first):
  CMakeLists.txt:2000 (find_package)

[... 204 lines cut ...]
     ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdint:173:8: error: no member named 'int_fast16_t' in the global namespace
using::int_fast16_t;
     ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdint:174:8: error: no member named 'int_fast32_t' in the global namespace
using::int_fast32_t;
     ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdint:175:8: error: no member named 'int_fast64_t' in the global namespace
using::int_fast64_t;
     ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdint:177:8: error: no member named 'uint_fast8_t' in the global namespace
using::uint_fast8_t;
     ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdint:178:8: error: no member named 'uint_fast16_t' in the global namespace
using::uint_fast16_t;
     ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdint:179:8: error: no member named 'uint_fast32_t' in the global namespace
using::uint_fast32_t;
     ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.

**I also tried to set c++ standard library to libstdc++, c++ language dialect to gnu++14. But it shows many others problem, like 'cstddef' not found, 'utility' not found, etc. And I manually add c++ path to system header search path, but it didn`t work.**

Thanks a lot for any help.

carlesfernandez, 2020-05-11:
Dear @sdaheng ,

I've just reproduced the same process, with the same setup, and Xcode can compile the code successfully:

```
$ cmake -GXcode ..
...

-- ***************************************
-- *           SUMMARY REPORT            *
-- ***************************************
-- 
-- Building on macOS Catalina 10.15
-- CMake version: 3.17.2
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The C compiler identification is AppleClang 11.0.3.11030032
```

and then:
```
$ xcodebuild
```

or directly building from Xcode. In both cases, it works well.

I don't know what could be the problem in your setup. Did you have installed Xcode's Command Line Tools?

sdaheng, 2020-05-11:
Thanks very much for your reply.
Yes, I installed Xcode command line tools. I tried your method, and it
works.

I used CMake application not command yesterday. It seems that configured by
CMake application will cause compile errors? Any idea about why it happened?

I also want to know that If I integrate it into my project, I just use
those libs that Xcode built?

carlesfernandez, 2020-05-11:
Hi,

I always use CMake from the command line, so I don't know the peculiarities of the GUI. Probably is something related to the configuration.

Related to including the generated libs in another project, gnss-sdr is not really thought as a library but an application. But if you link your code against the generated static libs, and add to your project the paths of the required .h files, it should work. You probably will need to include into your project the dependencies of those static libs (e.g., GNURADIO, GLOG, GFLAGS, etc.).

Hope this helps!

sdaheng, 2020-05-11:
Yeah, Its really help me. I will think about how to use this for my
research project😁. Thanks again for your help.

carlesfernandez, 2020-05-11:
Great! Good luck with your project!

I'm closing this issue since it seems resolved, feel free to open a new one if other problems arise.

closed 2020-05-11 by carlesfernandez