gnss-sdr/gnss-sdr #376: Question about Loss of lock in channel

sdaheng, 2020-05-13:
Hi

I encounter a problem about lock signal in channel.

I use USPR-2953R(X310) as signal source.

Here is config file.
Its copied from '.../conf/gnss-sdr_GPS_L1_USRP_X300_realtime.conf', and made some changes.
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
;internal_fs_sps: Internal signal sampling frequency after the signal conditioning stage [samples per second].
GNSS-SDR.internal_fs_sps=4000000

;######### SUPL RRLP GPS assistance configuration #####
; Check https://www.mcc-mnc.com/
; On Android: https://play.google.com/store/apps/details?id=net.its_here.cellidinfo&hl=en
GNSS-SDR.SUPL_gps_enabled=false
GNSS-SDR.SUPL_read_gps_assistance_xml=true
GNSS-SDR.SUPL_gps_ephemeris_server=supl.google.com
GNSS-SDR.SUPL_gps_ephemeris_port=7275
GNSS-SDR.SUPL_gps_acquisition_server=supl.google.com
GNSS-SDR.SUPL_gps_acquisition_port=7275
GNSS-SDR.SUPL_MCC=244
GNSS-SDR.SUPL_MNC=5
GNSS-SDR.SUPL_LAC=0x59e2
GNSS-SDR.SUPL_CI=0x31b0

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.device_address=192.168.10.2 ; <- PUT THE IP ADDRESS OF YOUR USRP HERE
SignalSource.item_type=cshort
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.subdevice=A:0
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=../data/signal_source.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=cshort

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Fir_Filter
InputFilter.input_item_type=cshort
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=11
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.48
InputFilter.band2_begin=0.52
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=4000000
[... 400 lines cut ...]
Current receiver time: 1 min 35 s
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 19 (Block IIR)
Current receiver time: 1 min 36 s
Current receiver time: 1 min 37 s
Current receiver time: 1 min 38 s
Current receiver time: 1 min 39 s
Current receiver time: 1 min 40 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 19 (Block IIR)
Current receiver time: 1 min 41 s
Current receiver time: 1 min 42 s
```
I put the GPS antenna in RF0-RX2.
Actually, I am new to USRP...So.. I don't know is it a right place? Or I should put it in GPS Ant.

[image]

[image]

I want to know the configure file is correct? What items should I change that it can lock signal...

Thanks very much!

carlesfernandez, 2020-05-14:
Hi @sdaheng 

The input connector you mention is correct.

This is what I think it's happening: the USRP X310 ships an internal oscillator with poor stability. This use to be ok for most wireless communication applications, but is not enough for GNSS signal processing. If you have access to a better 10 MHz-clock (1 ppm or less is recommended), please plug it into the "REF IN" input and add the following line in your conf file:

```
SignalSource.clock_source=external
```

The "GPS ANT" input is used by an optional component, a GPS-disciplined oscillator (see https://files.ettus.com/manual/page_gpsdo_x3x0.html). This could be another way to get a stable clock, but it's a little bit weird to use a GPS receiver to implement a GPS receiver :-)

sdaheng, 2020-05-17:
Hello, thanks for your reply. I don't have another clock source... 
I plot acquisition dump files in Matlab. They all look like this.
[image]

Is it because there's too much interference that I cannot catch signals?

sdaheng, 2020-05-19:
Hi
I still don't know why it can not lock the signal...

This is histogram of carrier_lock_test that I plot from trachking_ch_0.mat.
[image]

And this is histogram of CN0_SNV_dB_Hz that I plot from tracking_ch_0.mat
[image]

Any suggestions that what can I do?

sdaheng, 2020-05-26:
I found the reason why it can not lock the signal. I use an active antenna and USRP doesn't supply power...
So When it connected a bias-T to supply power and captured  signal at roof of building. I got my position...

closed 2020-05-28 by sdaheng