gnss-sdr/gnss-sdr #381: Acquiring Galileo E1 signals

fgallardo, 2020-05-17:
Hi all,

I hope you are doing well, taking into account these complicated times.

So, I am trying to acquire Galileo signals with an USRP1 and a Tallysman antenna.

So, the GNSS-SDR seems to track some signals, but it loses the track and declares satellites that are not there, as present. In order to discard problems with my HW (I would say it is properly recording signals[1]), I decided to use a Galileo signal generator I coded myself. 

So, at 50dBHz (one single PRN, E31, Delay, and Doppler fixed all the time) it seems to find the satellite right away, now, as soon as I go to 31dBHz or below, it starts to have problems while acquiring the signal. I tried several things, but all resulted in the same: when the PFA is relaxed it starts to find random PRNs, but I would say it does not improve (it does not find the E31 PRN and obviously is not able to track it). I tried:

1. Relaxing the PFA.
2. Trying the two available implementations of GAL E1 ACQ. [2] 
3. Acquiring the E1C pilot signal (acquire_pilot=true)
4. Increasing the integration time (coherent_integration_time_ms) to  8 msec (and acquiring the pilot).
5. Generating the Local replica as CBOC.
6. bit_transition_flag=True
7. Forcing all channels to try to acquire E31

Any suggestion?

[Tracking of GNSS-SDR with E31 and CN0=50.0dBHZ.
](https://drive.google.com/file/d/1GtnZtzXlvGE61y8Xd0Cp3ieUvrlwV9a_/view?usp=sharing) (See the IQ diagram in the picture above, I think it is clear.)

[gnss-sdr INFO log with CN0=50.0dBHZ.](https://drive.google.com/file/d/1hXhMU5-uS3Bh8-DXydw6QlXSlowygOto/view?usp=sharing) 
[gnss-sdr INFO log with CN0=31.0dBHZ.](https://drive.google.com/file/d/12Oyn9QktzWLPlvTCtobNIJx_4c6QaHM1/view?usp=sharing)
[The config file ](https://drive.google.com/file/d/1xI05mz34u87jgCohqJLtWSfSP4uv9sZn/view?usp=sharing)(It has a long amount of tests, I can provide with the important part, better than this dirty version).

Thanks!
Fran

[1] So, fs=8MHz, recorded as gr_complex using the gnu-radio companion. No dropouts, recorded in a RAM disk, and the histogram and spectrum of the data seem to show AWGN, which in principle is a good sign. 
[2] By the way, congrats again for the training part of the web, it is awesome. https://gnss-sdr.org/docs/sp-blocks/acquisition/#galileo-e1-signal-acquisition

carlesfernandez, 2020-07-15:
Dear @fgallardo,

31 dB-Hz is quite on the edge of what we have achieved in acquisition sensitivity. The most effective measure you can take is to increase the `coherent_integration_time_ms` (of course with the pilot signal) and maybe play with `max_dwells`, which is the maximum number of non-coherent integrations. You can set them at the same time (so `coherent_integration_time_ms=8' and `max_dwells=2`will make two non-coherent integrations of 8 ms length coherent integrations). The non-coherent integration stops when the correlation peak goes above the detection threshold.

In our experience, generating the CBOC waveform makes a little difference, if any, apart from consuming more cycles of your CPU.

Forcing all channels to try to acquire the same signal won't be of any help, please force just one of them, and ensure that `Channels.in_acquisition` is equal to the total number of channels, so your "forced" acquisition block will be always trying to acquire E31.

It also helps to activate the dump of .mat files and visually explore the acquisition grid with Matlab/Octave, which can give you a quick intuition about what's happening.

Hope this helps!

fgallardo, 2020-12-29:
Sorry for bringing this topic back again after so long, but I think it may be useful to somebody.

But thanks to @odrisci, now I know what was my problem! The oscillator!! XD 

https://github.com/gnss-sdr/gnss-sdr/issues/458#issuecomment-752101514

It was just a matter of expanding the search space to cope with the uncertanty related to the shitty oscillator of my USRP1 at home.

So the recording was working!!!! :)

It works!

     3.02           OBSERVATION DATA    E                   RINEX VERSION / TYPE
G = GPS  R = GLONASS  E = GALILEO  S = GEO  M = MIXED       COMMENT
GNSS-SDR            fgallardo           20201229 154441 UTC PGM / RUN BY / DATE
GALILEO OBSERVATION DATA FILE GENERATED BY GNSS-SDR         COMMENT
GNSS-SDR VERSION 0.0.11                                     COMMENT
See https://gnss-sdr.org                                    COMMENT
DEFAULT MARKER NAME                                         MARKER NAME
fgallardo           CTTC                                    OBSERVER / AGENCY
GNSS-SDR            Software Receiver   0.0.11              REC # / TYPE / VERS
Antenna number      Antenna type                            ANT # / TYPE
        0.0000        0.0000        0.0000                  APPROX POSITION XYZ
        0.0000        0.0000        0.0000                  ANTENNA: DELTA H/E/N
E    4 C1B L1B D1B S1B                                      SYS / # / OBS TYPES
DBHZ                                                        SIGNAL STRENGTH UNIT
  2020    07    14    19    54   36.2000000     GAL         TIME OF FIRST OBS
    18    18   137     7                                    LEAP SECONDS
                                                            END OF HEADER
> 2020 07 14 19 54 37.0000000  0  4
E05  26000537.330 6  -1371564.728 6     36321.676 6        40.925
E09  21625273.521 6  -1292847.525 6     34479.269 6        41.894
E11  24519712.466 6  -1201957.779 6     31511.872 6        40.012
E36  23862049.848 6  -1287710.839 6     33720.645 6        41.885
   

Thanks @carlesfernandez !

closed 2020-12-30 by fgallardo