gnss-sdr/gnss-sdr #387: Failed to get the first position fix

air-elephant, 2020-06-03:
Hi, thanks for your amazing open source program!

I ran the gnss-sdr on the 18.04.4 Ubuntu (virtual machine) according to the "My first position fix" in the https://gnss-sdr.org/, but I cannot get any position results.
The configuration file is "my-first-GNSS-SDR-receiver.conf" on that web page, and the raw data file is downloaded form "https://sourceforge.net/projects/gnss-sdr/files/data/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.tar.gz".
The gnss-sdr version is v0.0.12 on the master branch, the same issue on the "next" branch. 

Here is the information generated by gnss-sdr while running:

linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.12 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/janson/workspace/gnss-sdr-pro/raw_data_file/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 20 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 32 (Block IIF)
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
Current receiver time: 14 s
Current receiver time: 15 s
Current receiver time: 16 s
**Loss of lock in channel 0!**
Current receiver time: 17 s
**Loss of lock in channel 3!
Loss of lock in channel 6!
Loss of lock in channel 1!**
Current receiver time: 18 s
Current receiver time: 19 s
...[**only just the current receiver time information**]
Current receiver time: 1 min 32 s
Current receiver time: 1 min 33 s
Current receiver time: 1 min 34 s
Current receiver time: 1 min 35 s
Current receiver time: 1 min 36 s
Current receiver time: 1 min 37 s
Current receiver time: 1 min 38 s
Current receiver time: 1 min 39 s
Current receiver time: 1 min 40 s
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 17.8083 [seconds]
GNSS-SDR program ended.

Looking forward to your help.

acebrianjuan, 2020-06-03:
Hi @air-elephant,

I have just tested the _"My first position fix"_ example on Ubuntu 18.04.4 LTS and it works just fine for both the `master` and `next` branches as you can see below.

- Output for `master` branch:

```
$ ~/git/gnss-sdr/install/gnss-sdr gnss-sdr --config_file=./my-first-GNSS-SDR-receiver.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.12 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/alvaro/signals/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 17 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 32 (Block IIF)
Current receiver time: 2 s
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 20 (Block IIR)
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 31 (Block IIR-M)
Current receiver time: 6 s
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
GPS L1 C/A tracking bit synchronization locked in channel 2 for satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 13 s
GPS L1 C/A tracking bit synchronization locked in channel 1 for satellite GPS PRN 32 (Block IIF)
Current receiver time: 14 s
GPS L1 C/A tracking bit synchronization locked in channel 4 for satellite GPS PRN 20 (Block IIR)
GPS L1 C/A tracking bit synchronization locked in channel 0 for satellite GPS PRN 01 (Block IIF)
GPS L1 C/A tracking bit synchronization locked in channel 3 for satellite GPS PRN 11 (Block IIR)
Current receiver time: 15 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 22 (Block IIR)
Current receiver time: 16 s
Current receiver time: 17 s
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
New GPS NAV message received in channel 1: subframe 2 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 2: subframe 2 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 21 s
Loss of lock in channel 6!
Current receiver time: 22 s
Loss of lock in channel 7!
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 2: subframe 3 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 27 s
Current receiver time: 28 s
Current receiver time: 29 s
Current receiver time: 30 s
Current receiver time: 31 s
Loss of lock in channel 5!
Current receiver time: 32 s
[... 679 lines cut ...]
New GPS NAV message received in channel 3: subframe 5 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 4: subframe 5 from satellite GPS PRN 32 (Block IIF)
Position at 2013-Apr-04 06:24:44.000000 UTC using 5 observations is Lat = 41.274854052 [deg], Long = 1.987693376 [deg], Height = 75.247 [m]
Velocity: East: 0.465 [m/s], North: 0.628 [m/s], Up = -0.398 [m/s]
Position at 2013-Apr-04 06:24:44.500000 UTC using 5 observations is Lat = 41.274857702 [deg], Long = 1.987703047 [deg], Height = 82.982 [m]
Velocity: East: -0.280 [m/s], North: 0.719 [m/s], Up = 1.015 [m/s]
Current receiver time: 1 min 39 s
Position at 2013-Apr-04 06:24:45.000000 UTC using 5 observations is Lat = 41.274858995 [deg], Long = 1.987698061 [deg], Height = 69.667 [m]
Velocity: East: -0.487 [m/s], North: -0.561 [m/s], Up = -0.537 [m/s]
Position at 2013-Apr-04 06:24:45.500000 UTC using 5 observations is Lat = 41.274839267 [deg], Long = 1.987689622 [deg], Height = 79.346 [m]
Velocity: East: 0.510 [m/s], North: 0.607 [m/s], Up = -2.052 [m/s]
Current receiver time: 1 min 40 s
Position at 2013-Apr-04 06:24:46.000000 UTC using 5 observations is Lat = 41.274812110 [deg], Long = 1.987626099 [deg], Height = 61.968 [m]
Velocity: East: 0.079 [m/s], North: -0.287 [m/s], Up = 1.224 [m/s]
Position at 2013-Apr-04 06:24:46.500000 UTC using 5 observations is Lat = 41.274865256 [deg], Long = 1.987682428 [deg], Height = 70.179 [m]
Velocity: East: -0.224 [m/s], North: -0.367 [m/s], Up = -0.721 [m/s]
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 4.687418 [seconds]
GNSS-SDR program ended.
```

Regards,

Álvaro

air-elephant, 2020-06-04:
Thank you for your reply.

I mean, there must be something wrong in the Gnss-SDR running on my vritual machine, but I cannot find it.  Do you have any idea about this issue?

That's quite weird.  The same operating system, the same gnss-sdr version, and it seems to be all running well, except for trakcing, decoding NAV message and positioning.
 
Ps: I run thr gnss-sdr on the **ubuntu virtual machine** by the Hyper-v technique of Windows and here is the git-branch information:
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/gui
  remotes/origin/kf-gsoc19
  remotes/origin/master
  remotes/origin/next
  remotes/origin/supl2

air-elephant, 2020-06-04:
Here is the configuration file:
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/janson/workspace/gnss-sdr-pro/raw_data_file/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat
SignalSource.item_type=ishort
SignalSource.sampling_frequency=4000000
S0.0.12Source.freq=1575420000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=4000000
Resampler.sample_freq_out=2000000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

air-elephant, 2020-06-04:
> Hi @air-elephant,
>
> I have just tested the _"My first position fix"_ example on Ubuntu 18.04.4 LTS and it works just fine for both the `master` and `next` branches as you can see below.
>
> * Output for [...]

air-elephant, 2020-06-04:
> Hi @air-elephant,
>
> I have just tested the _"My first position fix"_ example on Ubuntu 18.04.4 LTS and it works just fine for both the `master` and `next` branches as you can see below.
>
> * Output for [...]

Thank you for your reply.

I mean, there must be something wrong in the Gnss-SDR running on my vritual machine, but I cannot find it. Do you have any idea about this issue?

That's quite weird. The same operating system, the same gnss-sdr version, and it seems to be all running well, except for trakcing, decoding NAV message and positioning.

In addition, I notice that your total running time was only 4.68 seconds, but I spent 17+ senconds on it. Does it have something to do with this issue?

Ps: I run thr gnss-sdr on the ubuntu virtual machine by the Hyper-v technique of Windows and here is the git-branch information:

master
remotes/origin/HEAD -> origin/master
remotes/origin/gui
remotes/origin/kf-gsoc19
remotes/origin/master
remotes/origin/next
remotes/origin/supl2

acebrianjuan, 2020-06-04:
> I mean, there must be something wrong in the Gnss-SDR running on my vritual machine, but I cannot find it. Do you have any idea about this issue?

I certainly don't know what the exact cause of your problem might be, if any, so all I can do is speculate. My gut feeling is that this could be related to the performance throughput of your virtual machine.

> In addition, I notice that your total running time was only 4.68 seconds, but I spent 17+ senconds on it. Does it have something to do with this issue?

This might be explained by the reason above. Receiver running time is dependent on the specific environment where the receiver is running on. I am running Ubuntu natively, so this could explain why my running times are ~3.7 times faster than yours.

You can try a couple of things and see if they help:

1. Run `volk_profile` and `volk_gnsssdr_profile`, as explained [here](https://gnss-sdr.org/my-first-fix/#step-1-verify-that-gnss-sdr-is-installed).
2. In your receiver configuration file, set the `SignalSource.enable_throttle_control` to `true` (see list of configuration options [here](https://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-file_signal_source)). To do so, simply add the following line in the `SIGNAL_SOURCE CONFIG` section:

```
;#enable_throttle_control: Enabling this option tells the signal source to keep the
; delay between samples in post processing. It helps to not overload the CPU, but the
; processing time will be longer.

SignalSource.enable_throttle_control=true
```

> Ps: I run thr gnss-sdr on the **ubuntu virtual machine** by the Hyper-v technique of Windows and here is the git-branch information:

The GNSS-SDR [documentation](https://gnss-sdr.org/requirements/#operating-systems) states that virtualization is a valid option for running the receiver on the Windows platform, with a specific mention to Oracle's VirtualBox.

I've just read about [Hyper-V Linux support](https://en.wikipedia.org/wiki/Hyper-V#Linux_support) and, to my understanding, it should outperform any other virtualization tools for running Linux on a Windows host.

By the way, there have been a number of similar issues reported in the past, so this issue is somehow recurrent. You can have a look for [my first position fix](https://github.com/gnss-sdr/gnss-sdr/issues?q=is%3Aissue+my+first+position+fix) and/or [virtual machine](https://github.com/gnss-sdr/gnss-sdr/issues?q=is%3Aissue+virtual+machine) search terms in GNSS-SDR's issue tracker for more help.

Regards,
Álvaro

acebrianjuan, 2020-06-04:
Another thing you can do to find out what might be going on is check the generated log files in the `/tmp` directory. They provide detailed insight of the receiver's internals.

There you should see a file named `gnss-sdr.INFO` which will point you to the log of the latest GNSS-SDR run.

Regards,
Álvaro

air-elephant, 2020-06-04:
no comments

air-elephant, 2020-06-05:
Thanks for your detailed reply, @acebrianjuan .I think, I have found what exactly cause that issue.

I downloaded the raw data file <2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.tar.gz> on the Windows host, and put it into the Ubuntu virtual machine by copy&paste. However, **it was copy&pasting between the host and the virtual machine that broke the compressd raw data file.** That's why the gnss-sdr cannot track any signals. Instead, I copied that data file by the shared document (**local network share**), and it worked!

Ps: Would you mind showing your computer performace? like 
a) number of cpu logical cores/physical cores
b) cpu dominant frequency (Intel Core i7-8700?)
c) RAM size
d) HDD or SSD?
e) GPU parameters
or something like that.
You gnss-sdr really runs much faster than mine! In the near future, I will run more channels, gnss systems on the higher sampling rate (like 20+MHz) in the real time mode, so the real time performance is much of concern to myself.

acebrianjuan, 2020-06-08:
Glad that you discovered the cause of the problem @air-elephant!
That explains why no signals were getting picked up whatsoever.

> Ps: Would you mind showing your computer performance?

- Intel Core i7-8750H (6 cores @ 2,2 GHz - 4,1 GHz)
- 16GB RAM
- 256GB SSD
- NVIDIA GeForce GTX 1060

Regards,
Álvaro

closed 2020-06-09 by air-elephant