gnss-sdr/gnss-sdr #389: Failed to create eclipse project by cmake

air-elephant, 2020-06-08:
Hi,

I use the [cmake -G "Eclipse CDT4 - Unix Makefiles" -DCMAKE_BUILD_TYPE=Debug -DECLIPSE_GENERATE_SOURCE_PROJECT=TRUE -DCMAKE_ECLIPSE_VERSION=4.5 .] command to create Eclipse project in the GNSS-SDR root directory, which generated the ".cproject" and ".project".

But, there was a **warning** showing that 
CMake Warning:
  Manually-specified variables were not used by the project:
    ECLIPSE_GENERATE_SOURCE_PROJECT

And when I imported the this project into Eclipse, there was much “Unresolved Symbols” compiling error.
Even "std::cout" cannot be resolved, as well as many standard include paths, like <string>, <iostream>.

My eclipse cdt version is 4.15.

carlesfernandez, 2020-06-11:
Ok, the Eclipse instructions are outdated in the README file. It seems that Eclipse has issues if the build directory is a subdirectory of the source directory (as is the case of the
`gnss-sdr/build` folder), so it is strongly recommended to use a build directory which is a sibling of the source directory. Try this:

from the `gnss-sdr` root folder:

```
$ cd ..
$ mkdir eclipse && cd eclipse && mkdir gnss-sdr && cd gnss-sdr
$ cmake -G "Eclipse CDT4 - Unix Makefiles" -DCMAKE_ECLIPSE_GENERATE_SOURCE_PROJECT=TRUE -DCMAKE_ECLIPSE_VERSION=4.5 ../../gnss-sdr
```

and then import the created project into Eclipse:

1. Import project using Menu File -> Import.
2. Select General -> Existing projects into workspace.
3. Select your root directory: Browse and select your newly created
   `eclipse/gnss-sdr` directory. Keep "Copy projects into workspace" unchecked.
4. Click on "Finnish" and you should get a fully functional Eclipse project.

Hope this helps!

closed 2020-06-23 by air-elephant

evso9816, 2020-11-24:
Hi Fernandez,
It seems like this issue is closed but repeating the above steps you provided don't work. Can you help how to import gnss-sdr into eclipse? Really appreciate your support. Thank you.

ZiglioUK, 2020-11-24:
Unfortunately this problem has nothing to do with gnss-sdr and all to do with the state of C/C++ development in Eclipse. Basically there's no official support for CMake. It's all based on old plugins that are no longer maintained. It's good enough perhaps for very small projects but not for anything near the complexity of gnss-sdr, with lots of submodules. Perhaps you could try creating a CMake project in Eclipse for a submodule within gnss-sdr. Haven't tried myself. But unfortunately that's the state of things. Eclipse, a once great IDE. now an elephant that's kept on life support.

evso9816, 2020-11-30:
> Unfortunately this problem has nothing to do with gnss-sdr and all to do with the state of C/C++ development in Eclipse. Basically there's no official support for CMake. It's all based on old [...]

Thank you so much for the update! Good to know.

evso9816, 2020-12-12:
@ZiglioUK @carlesfernandez I successfully re-installed all steps, but all manually! It works fine. [cmake -G "Eclipse CDT4 - Unix Makefiles" ...] works well too!
Not sure why it was not the case with "pybombs" installation. 
Thanks!