gnss-sdr/gnss-sdr #390: How to get realtime stream from USRP B210 in ishort format?

Tanggeshi, 2020-06-14:
I'm running GNSS-SDR v0.0.12 by connection with USRP B210.  I can get realtime stream from USRP B210 in gr_complex item_type by use SignalSource=UHD_Signal_Source, but if I want to change the format to ishort,  it always tell me itemsize mismatch.  But if  I record data in ishort format by "uhd_rx_cfile", and then use SignalSource=File_Signal_Source, it work properly. All the setting in SignalConditioner is the same!   Would you please give some advice hwo to solve this problem?

noelpedro, 2020-06-24:
Hi @Tanggeshi.

Looking at uhd signal source it looks like it only can be cshort, cbyte, gr_complex.

Tanggeshi, 2020-06-25:
Oh!  Thank you！I will try cshort tomorrow !

Tanggeshi, 2020-06-25:
Infact, I cannot understand what difference between intervealed I/Q sample and complex sample. Would you please explain it for me? Thank you!

noelpedro, 2020-06-26:
I think Cshort is I and Q concatenated In a 32 bit word.

Check out this tutorial.

https://gnss-sdr.org/docs/tutorials/understanding-data-types/#data-types-in-gnss-sdr