gnss-sdr/gnss-sdr #391: QUESTION FPGA ACQUISITION

noelpedro, 2020-06-24:
How much if any of the acquisition grid is stored in BRAM?

The entitre 2-D grid? only a row at a time? none?

Thanks

Noel

mmajoral, 2020-07-10:
Hi Peter,
There is not enough memory in the FPGA to store the the entire 2-D grid.
The acquisition runs a parallel code-phase search algorithm.
For each possible doppler frequency, the acquisition algorithm performs the following tasks:
- Perform the doppler wipeoff and compute the correlation between the received signal and the local copy of the PRN code. 
- Search for the peak values and peak positions in the correlation results.
During the doppler search, the results of the correlation are not stored, only the peak values and the peak positions  (first peak and second peak values) corresponding to the highest peak values that were found so far. At the end of the doppler search the results for the entire 2D grid are obtained without having to store the entire 2D grid.

Regards
Marc

mmajoral, 2020-07-10:
Another possibility would be to use the external memory (SDRAM) to store the entire 2D grid.

closed 2020-07-10 by noelpedro

noelpedro, 2020-09-06:
Hi Marc,

Is the raw signal stored in bram before performing Doppler wipeoff?

Noel