gnss-sdr/gnss-sdr #395: Overruns in Ubuntu After about 8min

AgileEngineeringLLC, 2020-07-02:
I'm running Ubuntu 20.04 LTS on a new System76 Linux box. I built GNSS-SDR (next option) and am running a HackRF at 4MS/S. After about 8 min I get overruns every time (see attached screenshot). I noticed on the screen shot that there's an RTKLIB error message that always proceeds the overrun start.

Don

[image]

AgileEngineeringLLC, 2020-07-03:
Update: I re-built from next again and this time I'm not seeing the error.

closed 2020-07-03 by AgileEngineeringLLC

carlesfernandez, 2020-07-03:
Nice to know that it is working for you. This smells like an old bug that should be long gone. I was rechecking it when I saw your message.

AgileEngineeringLLC, 2020-07-03:
It actually has started doing it again, latest run took 13 min. I'm running a hackrf at 4MS/S, so seems like it should be fine. I have a B200-mini coming next week, so I'm hoping the USRP interface is maybe more stable than the OSMOSDR interface. Don