gnss-sdr/gnss-sdr #397: How To Ignore Satellite in PVT Solution?

AgileEngineeringLLC, 2020-07-05:
Is there a way to disable channels from being assigned to certain GPS and Galileo satellites?

I'm seeing Galileo Satellite PRN 18 in my E1 solution, with a faulty range. GNSSPlanning.com says that Galileo PRN 18 is out of service, so I'd like to tell GNSS-SDR not to use it. 

Thanks!
Don

carlesfernandez, 2020-07-06:
Yes, there is a way to do that. You can specify the list of satellites to be searched with the configuration parameter `Galileo.prns`. In order to ignore satellite E18, you can add to your conf file:

```
Galileo.prns="1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36"
```

The equivalent for GPS is `GPS.prns`. Satellites not included in those lists will never be assigned to a channel.

I'm afraid this feature is still undocumented in the website.

AgileEngineeringLLC, 2020-07-06:
Great, thanks!!

closed 2020-07-06 by AgileEngineeringLLC

carlesfernandez, 2020-07-06:
Hi @AgileEngineeringLLC 

since this is a useful feature, we implemented new configuration parameters for that, documented at https://gnss-sdr.org/docs/sp-blocks/global-parameters/#banned-satellites

Now, if you pull from `upstream next` and rebuild, you can just set:

```
GNSS-SDR.Galileo_banned_prns=18
```

Parameters `Galileo.prns`, `GPS.prns`, etc. will still work.

Thanks for raising this issue.